serkri/SmartEVSE-3 #11: Help with battery integration

theflexa1, 2022-04-09:
Hi to every one. 

I am trying to include the battery integration. I think I need more info/help.

I have installed the Release v1.2.0-serkri-v3. The firmware.bin and the spinff.bin via http://xx.xx.xx.xx/update. I can see now the new status page.
I am new at this type of proyects and my knowlage in coding is 0.

Can anyone teach me what is my next step to make the sistem to see the battery?
I use a Huawei Luna with a Huawei inversor.

Also I don´t know how can a see the EVSE in my home assistan.

Is anyone as lost as I am? xD

PD: I use the sensorbox to see the L1 details, I am only working on monophase.

Thank you.

k-janssens, 2022-04-11:
As I also use HASS with a Huawei SUN2000 and LUNA2000, I made a wiki page with how I did it:
https://github.com/serkri/SmartEVSE-3/wiki/Home-Assistant---Battery-Integration

You can also do it with a sensor instead of Node-RED, but at least this page should give you some details on how I did it.

theflexa1, 2022-04-16:
Hello there.

I am Using a Synology NAS. I have installed HA, portainer, docker...... But I cannot find info about how to make this work with nodered. My USB is not activated and I understand nothing about coding.... Is there any way to find help to install this by a noob like me? Video? Or a Step by step way?
I am getting crazy :'(

k-janssens, 2022-04-16:
For node red you need a Hass with supervisor, a Hass docker install does not support this.
Node red is just how I did it, if you want to do the same thing, best to deep dive with the help on the Hass forum.
Alternatively you can create a Hass automatision to archieven something similar.

closed 2022-04-27 by k-janssens