serkri/SmartEVSE-3 #13: WiFI disconnects

deqw, 2022-04-11:
Not sure if I need to repeat this here. To prevent esp32 WiFi disconnects:

WiFi.setAutoReconnect(true);
WiFi.persistent(true);

'Right after connecting to Wi-Fi. When the connection is lost, it will automatically reconnect.'

https://randomnerdtutorials.com/solved-reconnect-esp32-to-wifi/

deqw, 2022-04-11:
Great, looking forward to test this out. 
And maybe a great chance to get rid of existing polling/looping code for this WiFi problem.

k-janssens, 2022-04-11:
1.3.4 has the code, also disabled his own reconnect functionality as this should indeed not be needed anymore

fluppie, 2022-04-11:
Also interesting to see how projects like Tasmota and ESP Easy are handling this. Just to be sure not to re-invent the wheel.
I think Tasmota has a pretty high number of users and developers, so I would guess they have a good implementation.

deqw, 2022-04-11:
> 1.3.4 has the code, also disabled his own reconnect functionality as this should indeed not be needed anymore

Thanks for your swift action. Will test this version now.

koen-serneels, 2022-04-11:
> Also interesting to see how projects like Tasmota and ESP Easy are handling this. Just to be sure not to re-invent the wheel. I think Tasmota has a pretty high number of users and developers, so I would guess they have a good implementation.

Good point. The "wifi.h" used on the ESP is rather high level so it's more matter of correct configuration, but if this fix isn't working out we could look at what Tasmota is doing different with it.

Also, wrt reinventing the wheel, interesting that you mention Tasmota, as I was thinking before that it could be a nice platform to start with and implement the EVSE logic as some kind of module or plugin. I have tasmota running on an LED and it comes ootb with everything you need. However, after having a look at it I'm not sure that it outweighs pro/contra as the platform is rather complex and like 90%  of this code base would need to be retained anyhow. It would give ootb WebUI/MQTT and the likes but having the HTTP API is probably enough for most integrations

deqw, 2022-04-16:
No more infinite WiFi disconnects seen in the last days after this update. A sporadic timeout from which is recovered within a minute and two request errors “socket hang up”, thats all.

k-janssens, 2022-04-17:
Sounds good, will close the issue

closed 2022-04-17 by k-janssens