serkri/SmartEVSE-3 #16: new feature request

ce-jaro, 2022-04-19:
I would like to turn this device from being API server to being MQTT IoT client.
So device can connect (subscribe) to remote MQTT topic to receive commands (start/stop charging) and send regularly messages about the energy usage.
I'm looking for a developer who could help me to do this as I'm not very experienced in C++.
If needed I can partially fund this development.
Thank you so much!

koen-serneels, 2022-04-21:
mqtt support directly on the esp should be possible, but more complex. What about a separate adapter app that uses the evse api to periodically poll for state changes and publishes them? It could also react on received events and call the appropriate operation on the api to change stuff. This app should then be deployed separately from somewhere where it can access the evse (could also be a sidecar on the mqtt broker for example)

deqw, 2022-04-23:
Take a look here: https://github.com/emsesp/EMS-ESP
Code for MQTT integration is there. I've got one of these running now for two years against HomeAssistant Mosquitto broker.

koen-serneels, 2022-04-24:
Sure, but the c++ variant will be more complex to implement and maintain. What about the adapter module? Could be deployed alongside your broker for example

deqw, 2022-04-24:
I have no special interest (anymore) in additional MQTT options in SmartEVSE myself, I was just pointing TS in that direction. The API is fine and for phase information I now use the RS485 data from two Eastron SDM630 devices. Before I used the API but found the RS485 bus the more solid option.

ce-jaro, 2022-04-24:
Thank you both for your valuable feedback.
I guess, the adapter module solution will require an additional hardware device (like r.pi ) which would add additional cost to the product which I don't like so much.
The idea of having EVSE as an IoT device with capability to subscribe to remote MQTT cloud service to receive control messages creates a different use-case for this amazing project.
I would also like to see the support for power energy metering by the Smart-EVSE device it self to report current and total power energy used for each charging session.
But II don't know whether the existing hardware device (v.3) has all what is needed for ESP8266 to be able to do an energy metering.
If so, it would be worth to find and pay C++ coder to add this functionalities.

k-janssens, 2022-04-27:
At the moment I personally don't have the need for MQTT.
But you're welcome to open a PR to add it, keep in mind this will also require some refactoring of the API logic as the MQTT client should use the same code - more or less - as the API endpoints.

closed 2022-05-04 by koen-serneels