serkri/SmartEVSE-3 #17: new feature request RFID

hliebscher, 2022-04-24:
hi thanks for the cool API and the new features since your fork!

is it possible to get and set RFID throw API  to make billing for different Cars with power consumtion per RFID possible 
Also a nice feature is 1 and 3 phase switching per rfid 

my next step is integration in IO-BROKER and write a short script to log all data in the influxdb with grafana

i have one SimpleEVSE-WiFi and 3 SmartEVSE-3 running with 30kwP SolarPanels

k-janssens, 2022-04-27:
Most likely that is possible, but as I don't have RFID setup with mine, testing it is quite difficult.
Feel free to open a PR to add it yourself if you're feeling comfortable with doing that.

hliebscher, 2022-05-02:
at first it is fine to get the last read rfid(X) from onewire.cpp
but i have no idea how to integrate in API. 
doc["evse"]["rfid"] = RFID{x} ;

dingo35, 2022-07-02:
Just  add it in the API like you describe, and check it with the "raw data" button in the webserver.

fluppie, 2022-07-24:
This could be done through OCPP, which is meant for that.
See my feature request in the main repo: https://github.com/SmartEVSE/SmartEVSE-3/issues/24

https://github.com/RWTH-i5-IDSG/steve this is a open-source management system.

dingo35, 2022-07-24:
Interesting protocol, when I have some time I will take a look at this protocol; always happy to improve connectivity by open standard protocols!! Thanks @fluppie

k-janssens, 2022-07-24:
OCPP could indeed be very interesting for those who get an EV as a lease car.
I know several people who cannot use smart-evse because their company doesn't allow it. OCPP could help them (if the 3rd party wants to integrate with it)

It ain't rocket science but it'll take quite some time to have it implemented/tested/etc. 
Keep in mind you also need the other side of the OCPP integration to verify the EVSE implementation.

dingo35, 2022-07-24:
I understand the Open Charge Alliance has a test tool available that can take both the role of the Charge Point as of the Central Server [https://www.openchargealliance.org/protocols/test-tool/](url)

fluppie, 2022-07-24:
Yes I had quite some discussion with my employer. Luckily we could agree on a MID certified kWh meter and report every month the kWh reading and then expense these costs. Since I'm the only Belgian employee driving an EV, this is manageable. But for larger companies they need OCPP compliant charging equipment and use companies like Last Mile Solutions for billing.

k-janssens, 2022-07-24:
Keep in mind there are like 4 versions already (1.5/1.6/2.0/2.1).
All of them are maybe not needed but this will also depend on the consuming side of course.

Personally I'm not a fan of building this directly in the smart-evse, I would go for a solution between the OCPP consumer and the EVSE.
My main reasons being:
- keeping the EVSE off the public internet (imagine the /update endpoint being publicly available)
- using a higher level language (e.g. Java) will make it easier to implement/maintain the OCPP versions
- there are already some opensource project for this purpose, meaning you only need to connect this "middleware" with the EVSE API

fluppie, 2022-07-25:
I think that's the goal, make the charge point connected to a management tool in the local network or IOT vlan, not directly with portforwarding and a dyndns to the www for everyone available.

https://github.com/matth-x/ArduinoOcpp#supported-operations
There are functions to request firmware version and also push updates to charging points in the field.
So in my opinion, the SmartEVSE should/could support OCPP but the management software itself should be on something like a Raspberry Pi or some VM running on a local NAS/server. Or maybe you can also run stuff like this in Azure/AWS/whatever. I think those commercial companies like Last Mile Solutions have it set-up this way?

k-janssens, 2022-07-26:
Well there are a few use cases you have to distinguish what OCPP can give.

One use case is for companies, who install charging hardware, to use this to also manage their infrastructure. 
But this is not really useful for us as we only have 1 or 2 to maintain.

The other use case is for billing purposes which makes more sense for most people.
Having an open standard like OCPP is great, as this would allow a 3rd party to integrate quite easily with your home system ... if they are willing to do so.
In this use case, the mgmt component is hosted by the 3rd party, and you need to provide them with an EVSE node they can interact with.

You could implement the OCPP EVSE part directly in the smart-evse, but that is what I'm personally not a fan of (for the reasons above). I would host a OCPP Node service that interacts with your local smart-evse, keeping the smart-evse enclosed from the internet.

This way the OCPP Node service is a proxy between the outside world and your smart-evse, which handles all outside comms and handles security.

An additional benefit is the segregation of concerns.
If one day you upgrade your smart-evse or install something else, you only need to change the integration between your Node service and the EVSE. 
But the integration with the 3rd party will stay AS-IS, not requiring them to change anything, keeping this integration stable.

closed 2022-08-22 by k-janssens