serkri/SmartEVSE-3 #18: 3 Phase only when MODE_NORMAL

deqw, 2022-04-25:
if(Mode == MODE_NORMAL && enable3f)
                CONTACTOR2_ON; 

Can you explain why this MODE_NORMAL is necessary?

k-janssens, 2022-04-27:
As far as I can see, the original code does not support 3 fase solar/smart modes.
E.g. In solar mode and EVSE sees -6A is going to the grid, it'll say the car can load at 6A, but if it charges with 3fases it will load 6A x 3f which is 18

So to avoid any issues this is currently limited for manual mode only as the logic will most likely start flipping on/off otherwise

deqw, 2022-04-28:
I thought that using two contactors gives the option to use only 1 phase; not enabling 3 phase.

In the current original code, where you refer to, C1 and C2, for now, are switched at the same time. This is the same as having just one 4 pole contactor, in which situation 3-phase will always be enabled (with proper wiring of course).

k-janssens, 2022-04-28:
C2 is for enabling 3f, you have to install them in serie, not parallel

koen-serneels, 2022-04-28:
@deqw  that is exactly what it is doing. When following the original EVSE wiring schema it is wired for 3phases via a 4p contactor. The goal of the second contactor is to be able to toggle that for two reasons: a) the original smart/solar code does not seem to take into account 3f when calculating energy b) in most cases when charging solar the minimal energy on 3p will exceed the solar energy available and start to draw power from batt/grid, to lower that you'll have to switch to 1phase charging.

So, if you put a NO 2p (C2) in series with the NO 4p (C1) (as @k-janssens indicates) then in smart/solar mode the EVSE will operate from now on in 1phase mode (as C2 is not closed). If you switch to normal mode it will activate C2 and you'll get 3phase charging.

k-janssens, 2022-04-28:
For safety reasons we decided to go for the serial setup. If for some reason the C1 is OFF and C2 is still ON, there are 2 phase cable with power still connected with the car without an active neutral wire.
Maybe it is overly cautious but better safe than sorry was our opinion.

closed 2022-05-04 by koen-serneels

martijndierckx, 2022-06-30:
Do I understand correctly, I need to wire as follows?
Can you update the documentation to clarify this?

[image]

synegic, 2022-10-09:
@k-janssens can you elaborate why you think the code does not take into account all three phases? If we look at the code we can see at https://github.com/serkri/SmartEVSE-3/blob/master/SmartEVSE-3/src/evse.cpp#L1333 that it will take the highest loaded 'channel' (or phase) to do its calculations. This would in theory mean it will never charge more than the highest loaded phase, which is basically what you want as there is no way to tell the car to load the other phases differently.

Or am I completely off-base here? I do understand Solar charging mode is more difficult for three-phase operation, especially if you have a single phase inverter on a three phase mains supply like me. But at least 'smart mode' with balancing should be safe with enable3f true.

k-janssens, 2022-10-09:
I mainly talk about solar functionality as the goal there is to use 0 power from the grid.There it is an issue as the EVSE does not take into account the amount of phases the car will use.
Meaning if the EVSE tells it to charge with 6A if there is 6A solar available, a 3 phase car will draw 18A resulting in the EVSE to continuously start/stopping the charging process.

Not 100% sure there is an issue for smart mode (never used it).

synegic, 2022-10-09:
Thanks for your reply. It makes perfect sense now. In that case I guess simply excluding MODE_SOLAR in that line where it enables contactor 2 as well should suffice to make three phase smart charging possible.