serkri/SmartEVSE-3 #27: WiFI disconnects (2)

deqw, 2022-06-23:
Follow up on the previous, closed, WiFi topic. The disconnects still occur and then I have to go to the Smart EVSE and fiddle with the menu to get it started again. Completely renewed WiFi network by now, with an access point a few meters away, so see no problem there.  No one else here with disconnecting WiFi? So if someone can take another look at this and make this thing trying to get on WiFi whatever it takes ;o)? Even a scheduled reboot would be better than as it is right now.

jonasf21, 2022-06-26:
Same here, still have disconnects without reconnect. Only thing that helps is power cycle.
My SmartEVSE is quite far from the AP and i usually get around ~86dbm reading which sometimes gets worse depending on various factors. So i am sure it might get dropped out of the wifi due to bad signal. That shouldnt be a thing but it is not reconnecting afterwards which really shouldnt be an issue.

deqw, 2022-06-27:
Normally you can go into the Setup Menu and Disable/Enable WiFi to have it working again, so no power cycle needed, in my case anyway. When I force a reconnect from the WiFI access point (when it is already connected) the WiFI connection is just lost (not restored) and I have to Disable/Enable WiFi to get it working again.

In addition, a fixed IP address setting would be nice to have.

jonasf21, 2022-06-27:
Can confirm - disabling and re-enabling WiFi on the unit does indeed fix it, so a powercycle is not needed. However that doesnt help much, as the breaker is next to the EVSE for me :)
Problem is likely with WiFi.setAutoReconnect(true) or in its used context. I might have a look at it if i have some spare time.

jonasf21, 2022-06-27:
I made a quick and dirty fix, which so far seems to work for me. If i kick the EVSE out of my wifi it now will reconnect after approx. 120 seconds. However this requires further testing - i'll see if its stable the next days. If you want to try it: [https://github.com/jonasf21/SmartEVSE-3/releases/download/0.0.0-dev/firmware-test.bin](url)
I'll publish a pull if its really fixing things.

deqw, 2022-06-27:
Thanks for looking into it. Will upload the bin tomorrow to givebit a try.

jonasf21, 2022-06-29:
After 48hours+ my EVSE is still connected to the SSID, even tho it did reconnect a couple of times due to signal loss. So far i think its fixed. Have you been able to give it a try?

deqw, 2022-06-29:
> After 48hours+ my EVSE is still connected to the SSID, even tho it did reconnect a couple of times due to signal loss. So far i think its fixed. Have you been able to give it a try?

Yep, running since yesterday. I have a Node-Red flow running that checks the availability every minute and notifies when something is wrong. So far looking good, but it has to run a few more days to be sure.

deqw, 2022-06-29:
But to give you already some information: I just did a test, by forcing a reconnect to the access point, after which WiFi was restored in two minutes.

I'm also using an EMS-ESP (for the heater) which also runs WIFi and this device has never failed WiFi in years. Maybe you can get some insight about their WiFi solution at https://github.com/emsesp/EMS-ESP32

jonasf21, 2022-06-29:
I've checked their source, its a bit different because the code also includes support for a ethernet header. Thus they have 2 ways of connecting. But as far as i can see they are basically doing the same thing (checking WL_CONNECTED via interval).

deqw, 2022-07-01:
> Alright, thanks for the feedback. Lets wait till the weekend or alike to be sure.

All OK for me now. Thanks for resolving this.

dingo35, 2022-07-01:
I have exactly the same problem, and exactly the same solution (disabling and reenabling wifi). Cannot test your solution since I have some specialties in my firmware, but looking forward to your PR !!

jonasf21, 2022-07-01:
Just posted the PR. Let me know if it works for you, too. Can probably be cleaner but it works.... :)

dingo35, 2022-07-02:
Great! First have to test RC 1.4.0, but will try after that.

Two remarks:
-you talk about check every 120 seconds, but in the code you check every 5 secs. Is that long enough for the firmware to let it try to reconnect itself (autoreconnect=true)?
-perhaps change status == !WL_CONNECTED to status != WL_CONNECTED, a bit cleaner code?

dingo35, 2022-07-03:
Your PR works fine, it inspired me to make an even faster solution, this one #32 reconnects within 15-20 seconds. Would you be so kind to test this?

deqw, 2022-07-04:
[image]
Just tested this again. When I use the latest binary 1.3.4 WiFi does not return after initiating a reconnect from the UI you see in the screenshot. With the @jonasf21 solution it always reconnects. This has nothing to do with the access point itself (weak or whatever) since the signal strength is 95 to 100% at around -50dB for the SmartEVSE, which devices are only 5m separated from each other.

dingo35, 2022-07-04:
@deqw , you obviously have a different problem than I (I have weak wifi signal problem), but the solution could be in the same PR.

Would you be so kind to test #32 (is not in the binary 1.3.4, that is from April 11th, the PR is from yesterday?

could you test #32 ?

deqw, 2022-07-04:
@dingo35 At the moment I'm back at the @jonasf21 binary. I've no development environment configured for this (on MacOS with xcode and Swift at the moment), but if you provide me with a binary I'm happy to test anything for you.

deqw, 2022-07-04:
Thanks. Tried that binary right away. What I notice is:

- Version: v3serkri-0.00
- when reconnecting via the UI of the AP the SmartEVSE now remains available (cache cleared) on the network, also during the reconnect
- time is 30 minutes off whatever binary I use

dingo35, 2022-07-04:
1. That is correct, I didn't update any version strings
2. Sounds good, so you so don't even have the 120s delay as with the Jonas patch?
 3. That must be entirely different bug, since you have that with all serkri-binaries.
Is it exactly 30 minutes off? It should be synchronizing by built in NTP server, not sure how the timezone is detected...

jonasf21, 2022-07-06:
Running dingo35's fixed version for some days now, just as stable as mine but reconnect is faster, i'd vote to include this into the next release.

deqw, 2022-07-06:
> Running dingo35's fixed version for some days now, just as stable as mine but reconnect is faster, i'd vote to include this into the next release.

Yep, same experience here.

deqw, 2022-07-07:
@dingo35 running your latest bin

koen-serneels, 2022-07-08:
1.4.0 release available (https://github.com/serkri/SmartEVSE-3/releases). Closing this issue.

closed 2022-07-08 by koen-serneels