serkri/SmartEVSE-3 #31: 3 Phase wire scheme

martijndierckx, 2022-07-02:
Can you confirm wiring for 2 contactors in order to switch between 1P and 3P charging should be like this?
(If so, can you update the manual as well?)

[image]

Linked issue: #18

k-janssens, 2022-07-02:
That is correct, it felt for us to be the safest setup. 
For example: with 2x 2P in parallel you could have 2 live fase wires without the neutral if for some reason the contactors aren't working/controlled correctly.

koen-serneels, 2022-07-08:
Below a complete setup when already having an existing installation and the EVSE is placed in a  new cabinet. Of course, many variations are possible depending on local regulations. This setup at least was approved in Belgium. The type B diffs are important as a 6mA DC leak current protection must be in place. Note that when having a type B diff it not allowed/not safe to have upstream type A diffs.

Disclaimer: use at your own risk!

[image]

closed 2022-07-20 by koen-serneels