serkri/SmartEVSE-3 #33: EVCC.io support

fluppie, 2022-07-05:
Pretty cool software, plug-in for Home Assistant is available
https://docs.evcc.io/docs/Home

Don't know if hard to support :)

dingo35, 2022-07-08:
Why would you use yet another piece of software, if you can interface directly with Home Assistant via the REST API:

`sensor:
  - platform: rest
    resource: http://SmartEVSE-51446.lan/settings
    name: smartevse
    scan_interval: 300
    timeout: 20
    json_attributes:
      - mode
#      - mode_id
      - car_connected
      - evse
      - settings
      - ev_meter
      - mains_meter
      - phase_currents
    value_template: "OK"

`

dingo35, 2022-07-09:
> I do have that running for a long time via HTTP REST.
> But this one can communicate in more detail with the vehicle itself. So you could set charging to 80% and even though my EV doesn't support that through their app, I can only limit it in the car. I can now easily change that every day depending where I have to go the next day. It also looks good. The basics of the SmartEVSE are great, like the solar and load-balancing. But you could also for example let it calculate the lowest possible charging current based of the needed SoC by a certain hour. That way in Belgium you could potentially reduce your "capaciteitstarief" which is based on quarter peaks you pull from the grid. Because they want us to spread our consumption as much as possible.

You can do all that in Homeassistant. I have an automation that charges up until a nr of kms or % of range it reaches, because my vehicle doesn't support it (reliably). And you can pit any timer on it you want.

And it can _never_ communicate in more detail with the vehicle itself than SmartEVSE, since it has to get the data either from the EVSE or from the API of the manufacturer.

It positions itself as a  information broker, just like Homeassistant. But Homeassistant has a much bigger user base and much more interfaces to other components. 

But if you like it, build an interface and put it in a PR....

closed 2022-07-24 by k-janssens