serkri/SmartEVSE-3 #34: Request info in interface

rvdgaag, 2022-07-05:
The EVSE is showing the total of the charged kWh's from the EV Eastern unit.

Really would like to have this info displayed in the web interface too.

Thanks for the great work!

dingo35, 2022-07-20:
Sure but I'm a bit busy for the next few weeks, do if there is no hurry you can put it on my list.

fluppie, 2022-07-21:
I added this a while ago in evse.cpp around line 2969-2970

```
        doc["ev_meter"]["EVMeter"] = EMConfig[EVMeter].Desc;
        doc["ev_meter"]["EVMeter_addr"] = EVMeterAddress;
        doc["ev_meter"]["EVMeter_total_kWh"] = EnergyEV;
        doc["ev_meter"]["EVMeter_charged_kWh"] = EnergyCharged;
        doc["ev_meter"]["EVMeter_W"] = PowerMeasured;
```

k-janssens, 2022-07-23:
I updated the API and index page, it should include now the EV meter data (if enabled).
It's in release v1.3.8-serkri-v3

fluppie, 2022-07-23:
Real value is 3148,492 kWh, only have the car for 6 months not 500 years :)
[image]

k-janssens, 2022-07-23:
I don't have a kwh meter connected with my evse so I'm not able to test, what should the value be?

rvdgaag, 2022-07-23:
Same issue here, also Charged kWh is factor 1000 to high

k-janssens, 2022-07-23:
Ah ok, the measurement is in Wh and not kWh

fluppie, 2022-07-23:
Same for the Charged kWh :-)
But the power reading is broken, because the 1Phase Eastron meters have a different register, 0x0C vs 0x34 on the 3Phase variants. I fixed that in my own version, but I used your binaries now and that's why it's broken now.

[image]

    {"1PEastron", ENDIANESS_HBF_HWF, 4, MB_DATATYPE_FLOAT32,    0x0, 0,    0x6, 0,   0x0C, 0,  0x156, 0}, // 1P Eastron SDM230 (V / A / W / kWh) max read count 80
    {"3PEastron", ENDIANESS_HBF_HWF, 4, MB_DATATYPE_FLOAT32,    0x0, 0,    0x6, 0,   0x34, 0,  0x156, 0}, // 3P Eastron SDM630 (V / A / W / kWh) max read count 80

dingo35, 2022-07-23:
Suggest we report in kWh, with accuracy of 1 decimal, like all the other kWh values, to prevent unnecessary space occupation of long term storage databases, if any are used.

k-janssens, 2022-07-24:
I made some time today to finally move myself to the 1.4 release branch. 
My changes with the kWh calculation will be soon in the 1.4 release once we merge it with the master.

dingo35, 2022-07-24:
Great, welcome to the future :-)  !!!

closed 2022-08-12 by koen-serneels