serkri/SmartEVSE-3 #35: Release V1.4.0 overloads Modbus bus

rvdgaag, 2022-07-08:
Just upgraded the SmartEVSE, noticed I could not read the modus (Eastron meters) anymore with this version. Switching back to the 1.3.4 made it possible again.

dingo35, 2022-07-09:
1) Is your modbus terminated with the 120 ohm resistors?
2) Did you appy the modbus-fix for both releases?
3) Did you check whether in 1.4.x the correct meter was still configured? To keep the order of meters somewhat "logical" a new meter config was inserted, so your configuration might now display another meter in the config menu.

Its probably nr. 3 ....

rvdgaag, 2022-07-09:
1. yes, the modes has been running fine for a couple of years.
2. ?? Never heard off, how can I do that ?
3.  Let me check this also.

Really interested on you info how to apply the modbus-fix ?

Thanks

dingo35, 2022-07-09:
Look at:
/SmartEVSE-3/SmartEVSE-3/eModbus-fix/fix.sh

Its two small changes in RTUutils.cpp, that are suggested by Stegen.  
As I understand, some people need this fix if the modbus is not  
terminated properly and/or has interference problems.

Links:
------
[1] https://github.com/serkri/SmartEVSE-3/issues/35#issuecomment-1179506130
[2]  
https://github.com/

rvdgaag, 2022-07-09:
Call me stupid but I can't find the /SmartEVSE-3/SmartEVSE-3/eModbus-fix/fix.sh on Github ?

dingo35, 2022-07-09:
[https://github.com/serkri/SmartEVSE-3/tree/master/SmartEVSE-3/eModbus-fix](url)

rvdgaag, 2022-07-09:
OK this is probably why its working with you 1.3.4 release where the fix is integrated ?

I checked the modbus addresses again with installing 1.4.0 and they are fine, the data is displayed OK on the eve display but you can't request any data anymore from the bus (from the eastron EV and the ALL eastron meter).

dingo35, 2022-07-09:
This is firmware v1.4.0 with PR's 36-39 included, compiled INCLUDING the emodbus-fix, wifi-debugging is on and 
#define LOG_MODBUS LOG_DEBUG                                                     // Default: LOG_WARN

Flash this version, telnet to your smartevse and you should be able to see whats happening on your modbus.

[firmware.zip](https://github.com/serkri/SmartEVSE-3/files/9076199/firmware.zip)

rvdgaag, 2022-07-09:
Yes I did it, the terminal rushes a shit load of data: here is a short copy of it:

ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
*** Remote debug - over telnet - for ESP32 - version 3.0.5
* Host name: SmartEVSE-5646 IP:192.168.1.200 Mac address:C4:5B:BE:8E:69:34
* Free Heap RAM: 187320
* ESP SDK version: v4.4.1-1-gb8050b365e
******************************************************
* Commands:
    ? or help -> display these help of commands
    q -> quit (close this connection)
    m -> display memory available
    v -> set debug level to verbose
    d -> set debug level to debug
    i -> set debug level to info
    w -> set debug level to warning
    e -> set debug level to errors
    s -> set debug silence on/off
    l -> show debug level
    t -> show time (millis)
    profiler:
      p      -> show time between actual and last message (in millis)
      p min  -> show only if time is this minimal
      P time -> set debug level to profiler
    c -> show colors
    filter:
          filter <string> -> show only debugs with this
          nofilter        -> disable the filter

****
* New features available:
* - Now you can debug in web browser too.
*   Please access: http://joaolopesf.net/remotedebugapp
* - Now you can add an simple software debuggger.
*   Please access: https://github.com/JoaoLopesF/RemoteDebugger
****

* Please type the command and press enter to execute.(? or h for this help)
***
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 21 97 fc 40 26 c7 9e 3f c3 96 98 c4 12 3f 8e c3 ee 66 54 c3 85 17 c4 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 0b 04 04 45 10 54 72 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0048
(ModbusDecode)(C0)  Response
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 0b 04 04 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0034
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
[... 453 lines cut ...]
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 22 27 33 40 27 9c 16 3f c6 a7 bd c4 12 b5 a6 c3 ef f0 91 c3 88 ce a7 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 0b 04 04 45 10 54 72 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0048
(ModbusDecode)(C0)  Response
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 0b 04 04 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0034
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

dingo35, 2022-07-09:
Your modbusrequests 3 and 5 seem to go fine, but then for some reason you are sending out modbusrequests 12 ; in my log those don't appear, 

It is not really a congestion on the bus, in my logs I have comparable number of records between the 100ms ticks. 

If this is a setup with multiple smartevse's (master and nodes), please describe your config. 
If this is just a single SmartEVSE: do you have loadbalancing configured to "Disable" or "Master"?

It looks like your mainsmeter sends and receives good messages, what happens if you disable your EV meter in the setup?

rvdgaag, 2022-07-09:
I have a server running Iobroker (yes another Modbus master) , it reeds the values of meters and logs them. I need this info to log the charging of the EV to claim the costs at my employer. It only reeds the bus like every 60 seconds.

dingo35, 2022-07-09:
Ok:
-see my EDIT and EDIT2 remarks above;
-what happens if you disconnect your second modbus master?

In this version you can get the values of the meters from the REST API, so you wouldnt need the second modbus master....

rvdgaag, 2022-07-09:
I updated my answer.

The REST API is fine but it does not offer all the data I need in my home for controlling other stuff

dingo35, 2022-07-09:
So loadbalance is configured master I presume?

rvdgaag, 2022-07-09:
There is no load balancing. I have just 1 EVSE. So the EVSE is master.

I disabled the readout of the EV eastron meter at the EVSE and this seems to improve te readout.

dingo35, 2022-07-09:
Ok and what happens if you temporarily disable you second busmaster and reenable the ev meter?

rvdgaag, 2022-07-09:
Here is a telnet readout with the EV meter off in the EVSE

* Please type the command and press enter to execute.(? or h for this help)
***
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 f7 6b e6 40 6a 27 90 3f bc e4 4f c4 e9 db ba c4 52 5c e1 42 17 55 c2 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 f8 c5 ee 40 67 f2 ef 40 92 c0 92 c4 eb 21 90 c4 4e aa 03 c4 87 df 3a 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 f9 22 61 40 65 ba 55 40 93 39 20 c4 eb 6d 0b c4 4d d2 a0 c4 88 55 0f 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 fa bb ec 40 12 ec 04 3f 9a c4 ef c4 ed 33 c5 c3 ef 39 59 42 8b 8c 3c 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 f9 94 94 40 5e bd 28 40 92 61 46 c4 eb e3 aa c4 47 cb 6c c4 87 97 97 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 13: Request MainsMeter Export Active Energy Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 0c 04 04 45 55 a3 91 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 004a
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 f6 11 a9 40 56 0a 49 40 91 0c 32 c4 e8 b3 bf c4 3e 6f 96 c4 86 55 4c 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 f3 a2 97 40 57 f6 17 40 6d 7c 79 c4 e6 94 08 c4 41 16 9e c4 29 eb 3f 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 f2 20 56 40 59 f0 af 40 8d 01 04 c4 e5 05 62 c4 45 85 56 c4 82 79 42 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 f0 38 fd 40 55 d8 65 40 8c 6c 90 c4 e2 f6 0f c4 40 63 db c4 81 ca b2 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 ee 9d a4 40 50 bd 45 40 8b 20 8f c4 e1 78 c2 c4 38 64 31 c4 80 7e 4e 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 f0 5b 84 40 4f 7e 54 40 8c cb c0 c4 e3 2c 58 c4 38 62 f7 c4 82 4c 6a 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C1) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 f0 62 ce 40 4d 99 0a 40 8c d3 72 c4 e3 44 83 c4 31 c6 fd c4 82 53 26 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
[... 143 lines cut ...]
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 ea e7 63 40 5d 34 33 40 8a dd aa c4 de 1a d1 c4 47 15 6d c4 80 70 19 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 ec 77 ba 40 5d 00 28 40 8b 74 60 c4 df a4 9d c4 47 11 95 c4 80 f0 99 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
[... cut ...]
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 eb 87 46 40 5b ea 83 40 8b 40 03 c4 de dd 45 c4 44 c3 8a c4 80 a5 1d 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 ea 66 34 40 55 d3 b7 40 8a da aa c4 dd be 34 c4 3c a4 aa c4 80 36 03 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006

rvdgaag, 2022-07-09:
EV meter back on, 2nd modbus master off:

{"version":"11:27:31 @Jul  9 2022","mode":"SMART","mode_id":3,"car_connected":"false","evse":{"temp":31,"connected":"false","access":true,"mode":1,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"solar_max_import":0,"solar_start_current":4,"solar_stop_time":10,"3phases_enabled":true,"mains_meter":"Eastron"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"EVMeter_import_active_energy":2309.2},"mains_meter":{"MainsMeter_import_active_energy":0,"MainsMeter_export_active_energy":0},"phase_currents":{"TOTAL":-33,"L1":-36,"L2":24,"L3":-21,"last_data_update":1657366052,"original_data":{"TOTAL":-33,"L1":-36,"L2":24,"L3":-21}},"backlight":{"timer":29,"status":"ON"}}

Telnet:

* Please type the command and press enter to execute.(? or h for this help)
***
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0c 04 18 40 0a 9a 58 40 6a 06 12 3f c0 14 50 c3 f3 25 35 44 48 fe 55 c2 be 23 73 
(ModbusDecode)(C0)  valid Modbus packet: Address 0c Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 0b 04 04 45 10 54 72 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0048
(ModbusDecode)(C0)  Response
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 0b 04 04 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0034
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
[... 550 lines cut ...]
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
[... cut ...]
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 12: Request EVMeter Current Measurement
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04

It looks like there is huge  amount off data send when also the EV meter is on, more than with only the Manometer

dingo35, 2022-07-09:
Ok it looks like its not the second busmaster thats causing the problem. I'll add some debugging lines to the code, I'll upload a new version.
How many modbus clients are there on the modbus?

rvdgaag, 2022-07-09:
There are 2 Eastron meters ( modbus 11 & 12 ), 1 modbus/TCP converter (Modbus 4) and a EVSE-3. (modbus 1)

The Iobroker server connects via TCP to the converter TCP converter.

dingo35, 2022-07-09:
Ok apart from the problem, you might want to move the modbus/TCP converter from address 4 to an address 10 (0xa) or higher, because the lower addresses are reserved for smartevse nodes when in smartmode ...

rvdgaag, 2022-07-09:
Ok will do that in a minute, no problem

dingo35, 2022-07-09:
[firmware.zip](https://github.com/serkri/SmartEVSE-3/files/9076497/firmware.zip)

Ok try this one in your normal setup (EV meter on, second busmaster on ...).

rvdgaag, 2022-07-09:
This version doesn't improve the modbus.

By the way, solar charging does not work, it pulls 100% from the grid.

* Please type the command and press enter to execute.(? or h for this help)
***
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 0b 04 04 46 2e 98 fe 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0034
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 41 77 cb 6a 41 75 b5 aa 41 78 af 69 45 6a 50 f3 45 66 9c 6c 45 69 31 e8 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 41 77 d3 4d 41 75 9a 05 41 78 99 1a 45 6a 19 66 45 66 77 13 45 69 1d f4 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 41 77 a7 00 41 75 aa 7d 41 78 bf 47 45 6a 1f 41 45 66 86 f1 45 69 45 82 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 41 77 6f 6f 41 75 89 4b 41 78 91 bf 45 69 f1 3b 45 66 6c 4d 45 69 25 a5 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 41 77 99 23 41 75 76 73 41 78 b2 ae 45 6a 1e aa 45 66 55 f2 45 69 4c 6f 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 41 77 93 46 41 75 8a 2a 41 78 cb 39 45 6a 18 a8 45 66 63 30 45 69 60 41 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C1) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 41 77 75 fb 41 75 b0 50 41 78 bf 20 45 69 f9 c1 45 66 89 d9 45 69 50 0a 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 41 77 7b da 41 75 9a b8 41 78 a4 d9 45 69 ff 97 45 66 79 7b 45 69 15 a0 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 41 77 69 bf 41 75 6d e3 41 78 b9 16 45 69 e2 ad 45 66 50 8c 45 69 27 81 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 41 77 51 35 41 75 9d ee 41 78 b4 f8 45 69 97 ad 45 66 7d a2 45 69 20 6c 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
[... 1367 lines cut ...]
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 41 77 ab be 41 75 5d d2 41 78 6a e5 45 6a 2e c5 45 66 1d ef 45 69 07 ed 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
[... cut ...]
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 41 77 e2 5d 41 75 9d 0b 41 78 8c 6a 45 6a 67 91 45 66 48 12 45 69 22 5e 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response
(Timer100ms)(C0) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.

dingo35, 2022-07-09:
It is very strange, I have a comparable setup to yours except for the second busmaster; this is my request for EVMeter current measurement:

(Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement
(ModbusSend8)(C0) Sent packet(ModbusSend8)(C0) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 0b 04 18 00 00 00 00 3d 37 69 18 3f 2b e8 4e 00 00 00 00 40 96 6c 92 42 ba 2f e9 
(ModbusDecode)(C0)  valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0)  Register 0006
(ModbusDecode)(C0)  Response

This is yours:

(Timer100ms)(C0) ModbusRequest 19: Request EVMeter Current Measurement
(ModbusSend8)(C1) Sent packet(ModbusSend8)(C1) address: 0b, function: 04, reg: 0006, data: 000c.
(ModbusDecode)(C0) Received packet(ModbusDecode)(C0) (27 bytes) 0b 04 18 41 77 57 67 41 75 12 70 41 78 14 e8 45 6a 1f 63 45 65 f2 e2 45 69 0b af
(ModbusDecode)(C0) valid Modbus packet: Address 0b Function 04
(ModbusDecode)(C0) Register 0006
(ModbusDecode)(C0) Response

So two strange things:
1) your request is sent as request 19, mine is sent as 21; although it shouldnt make any difference, it is strange.
2) bytes 4-7 and 16-19 are zero in _all_ my 0b/04/0006 responses, in your responses they seem garbled ....

Not sure what is going on here....

As to solar, haven't used that much, but I think if you have max_imported set to zero it will ignore that setting. 
IMHO the solsar mode is rather primitive, it is starting to charge whenever you start exporting energy to the mains by more than "startcurrent" A (sum of three phases), and then it stops after the timer expires, even when 2 seconds before the expire you are exporting more than startcurrent. Lots of room for improvement there, but mostly interesting when solar balanced payment becomes active here in +31 in 2025.

dingo35, 2022-07-09:
Ok I think I can reproduce part of your problem; try running with Loadbalance=Master setting!

rvdgaag, 2022-07-09:
OK Master is on @ load balancing. Second master is also running and updating correctly !

Now it is usable.

dingo35, 2022-07-09:
Ok this is just a workaround, of course it should work with Loadbalance=disabled too, I'm going to try to find that bug; hope you can test it if I find it...

rvdgaag, 2022-07-09:
sure, just drop me a message and I will be glad to help you.

Dankjewel !

dingo35, 2022-07-09:
[firmware.zip](https://github.com/serkri/SmartEVSE-3/files/9077389/firmware.zip)

This should fix your problem, please test both in Loadbalance=disabled as in Loadbalance=Master mode.

rvdgaag, 2022-07-09:
Yes, it works fine now, with both load balance disabled and master mode

dingo35, 2022-07-10:
Fixed in #40

koen-serneels, 2022-07-11:
The modbus fix is applied automatically when a release is made via the actions in this repo (the releases available via https://github.com/serkri/SmartEVSE-3/releases). This issue was more of a regression from #21. If my comment is processed in #40 I'll make a new release together with the other PRs

dingo35, 2022-07-20:
Issue can be closed, since #40 is merged.

closed 2022-07-20 by koen-serneels

dingo35, 2022-10-11:
> of a regression from #21[1]. If my comment is processed in #40[2]

Great Koen, love the speed we are working with now !! Just noticed you  
already brought out 1.4.0 four days ago, really enjoying this trip,  
hope to join you on the way to 1.4.1 or 1.5.0 !!

Hans.

Links:
------
[1] https://github.com/serkri/SmartEVSE-3/pull/21
[2] https://github.com/serkri/SmartEVSE-3/pull/40
[3] https://github.com/serkri/SmartEVSE-3/issues/35#issuecomment-1180857613
[4]  
https://github.com/