serkri/SmartEVSE-3 #43: Unlocking the cable

deqw, 2022-07-11:
Is it possible to unlock the (fixed) cable at the car’s side from the SmartEVSE? Would like to have it unlocked with the switch or even not having it locked at all when charging starts with my inside SmartEVSE.

k-janssens, 2022-07-11:
No, it is the car who does the locking at his end.

There are ways to (permanently) disable the locking capability by making slits in the charger cable where the locking holes are but I would not advise to do so. Also if you unlock the car, this mechanism should also unlock which is the preferred way.

closed 2022-07-11 by k-janssens

dingo35, 2022-07-11:
AFAIK it's the car that decides to unlock the cars side of the cable; SmartEVSE decides whether to unlock the socket side (if not fixed cable).

I solved this problem by defining a "home charge location", where I can specify "always unlock" (VAG ID-4 platform).

deqw, 2022-07-11:
Thanks both for the swift reply. Will look if a “home charge location” option is available for my car.