serkri/SmartEVSE-3 #53: Mainsmet disabled

deqw, 2022-07-27:
I’m still running v 1.4 that was published earlier. Today I noticed mainsmet was disabled when I changed the setting from Smart to Normal, also the mainsmet address was gone. When in Normal mode the Mainsmet setting is not in the menu. 3 phase is activated but it is only charging  on 1 phase although all 3 phases are activated. Then after a while probably something went wrong while charging and power was cut completely. Have no time to fiddle with the firmware at the moment but maybe in meantime someone has an idea what went wrong here.

dingo35, 2022-07-28:
1. Losing mainsmeter setting when switching to normal mode: I can reproduce this, put this on my list
2. Losing mainsmeter address: cannot reproduce this; what are your mainsmeter / mainsmeter address settings when in smart mode, and what settings does it report afyer switching smart->normal->smart?
3. Enable3f is limited to Normal mode only, for some reason; perhaps the author can shine some light on thus?

deqw, 2022-07-28:
The address was set to 101, after setting to Normal the address was the default 10. After setting all back and try again within a short time, the 101 address setting was maintained.

The 3f limiting to normal was discussed before in an other topic. There was sone reason for it. Something with security I think.

k-janssens, 2022-07-28:
**TLDR;**
3F in solar mode doesn't make any sense at the moment as the calculation logic is made for 1F.

**Long Answer:**
If we would enable 3F with solar, this would happen:

EVSE sees there is >= 6A solar power available **-->** start charging at 6A **-->** car will start charging at 18A (3x 6A) **-->** EVSE will stop as suddenly the system goes 12A in the negative

And then the system will repeat this flow as long as there is >=6A, resulting in start/stop charging sessions of a few minutes

To have this working properly, the calculation must make use of the amount of configured fases to properly calculate the required power.
And ideally the logic should also switch between 1F and 3F, otherwise the car will only start charging when there is 18A of solar power available.

I have an idea of how this could be implemented, but I can't charge myself at more than 1F so testing this is very hard for me.

dingo35, 2022-07-28:
@deqw what is your correct, main meter (manufacturer) setting, and what does it change to when switching smart->normal->smart?

E.g my main meter setting is InvEastron, but after switching to Normal it is set to Sensorbox.

deqw, 2022-07-28:
@dingo35 The setting is Eastron at address 101 (by @Stegen advice), the other Eastron is for the EVSE side at address 120. What changes is that the complete Mainsmet setting is unavailable in the menu, so I cant check what settings remain when in Normal mode. However, when going back to Smart mode the address is changed to the default (not 101).

I just came home and need to charge 3F for tomorrow, but with the 1.4 version it refuses3F  only L1 is activated while both relais are active. I'm going to load the latest published firmware now.

deqw, 2022-07-28:
OK, for the 3F problem I can report this is solved: I noticed zero Volt on L2 and visited every screw to tighten, so 3F loading now. For the Mainsmet setting and not appearing in the menu when in Normal mode the situation remains as reported before.

deqw, 2022-07-28:
And some minutes later L2 fails again, that is to say the Eastron reports 0 Volt, then 67 Volt and then again back to 234 Volt. The main Eastron reports a normal Voltage for L2 all the time. So maybe the Eastron at the EVSE is suspect. My multimeter is just today at another location so can’t make a measurement myself now.  Back to charge 1F now for the night.

fluppie, 2022-07-28:
I have/had this to when playing around with the solar/smart/normal/off buttons. Eastron SDM230 as mainsmeter.

deqw, 2022-07-30:
I can confirm that the L2 problem I had is caused by bad contacts in the power distribution rail and is solved now. The Eastron 630 meter is reporting voltage correctly.

@dingo35 Running 1.3.8 at the moment. Mainsmeter setting is lost (Disabled) when changing mode. The EVSE meter setting remains OK.

dingo35, 2022-08-21:
Should be solved by PR #60

deqw, 2022-08-21:
Thanks @dingo35 , can you do me a binary with all the latest code? Ran into some mainsmet problem yesterday when trying to load 3 phase. Tnaks again.

dingo35, 2022-08-21:
[firmware.zip](https://github.com/serkri/SmartEVSE-3/files/9389805/firmware.zip)

deqw, 2022-08-22:
> [firmware.zip](https://github.com/serkri/SmartEVSE-3/files/9389805/firmware.zip)
> 
> Current 1.4.0 branch compiled without debugging.

Thanks @dingo35

k-janssens, 2022-08-22:
@deqw Is this fix working for you? If so we can finally merge everything with the master

deqw, 2022-08-22:
@k-janssens I just installed this version. If I'm correct this concerns the WiFi fix and the Mainsmet fix. Just checked the Mainsmet fix and working fine now. WiFi was already OK. So you can go ahead with it.

closed 2022-08-28 by k-janssens