serkri/SmartEVSE-3 #56: charged_kwh & charged_total are 0

rmeekers, 2022-08-05:
Hi there, I've just installed SmartEVSE and flashed it to version 1.3.8.
Any idea why the `charged_kwh` & `charged_total` are 0? They do not return anything.

Thanks!

fluppie, 2022-08-07:
Do you have a seperate kWh meter to measure EV consumption? If yes, is this by accident a one phase Eastron SDM EV meter? Then it's normal since the one phase and three phase version have different modbus registers.

rmeekers, 2022-08-07:
Hmm. No I didn’t. Is that necessary?
If so, where can I find the compatible meters? I have a three phase
installation.

fluppie, 2022-08-07:
Yes, for example, my car charges with 32A monophase, but the real current is only 30.5A. That's +- 345W difference. Sometimes it's even more. Most of the time it charges 6950-7050W instead of the 7400W that's promised. Tesla model 3 really charges on 7400W on this charger. So that's why a meter is needed. Best is to take a MID certified one for accuracy and if you want/need to get kWh's back from your employer or company if the meter is on a private person.
Myself I like Eastron meters, you could take a cheap option SDM72 v2 or a more expensive SDM630. 
I ordered a time or 5 from this webshop: https://stromzähler.eu/stromzaehler/drehstromzaehler/fuer-hutschiene-geeicht/246/sdm72dm-v2-mid-3-phasen-zweirichtungs-drehstromzaehler-mit-rs485-und-s0

rmeekers, 2022-08-07:
Thanks. I’ll order one of those!

Xnygra, 2022-08-07:
How does ev_meter works when you have it set to API. can you send the data through the API, rather than through RS485? Cant find any documentation on that. 
Thanks for the great work, found this branch today after trying to get the data into Home Assistant from the original mstegen branch, without any success.

fluppie, 2022-08-07:
I have no clue :-) but I don't think that it works for EV Meter? I think only for Mainsmeter, but maybe @k-janssens can tell you that?
https://swagger-ui.serkri.be/#/Operations/updateCurrentsData here is the example for updating Mainsmeter data.

Myself I really needed the separate EV kWh meter to have home charged kWh's reimbursed from my employer. So I did not explore the API variant. But I have to say, the Modbus meters from Eastron are acceptable SDM230 €40 for single phase and SDM72 v2 €66 for 3phase. I would really recommend to pay a little bit extra for MID certified versions.

koen-serneels, 2022-08-07:
The mains_meter was not yet in the openapi yaml (I've added it https://swagger-ui.serkri.be/#/Operations/getSettings) although doing a get /settings would reveal it's there (the yaml file needs to be updated manually, this one was forgotten). Anyway, updating the mains_meter is currently only possible via the LCD. When in solar mode, you get the menu options for MainsMeter, when set to API, it's possible to supply the currents by posting them to /updateCurrentsData (see https://swagger-ui.serkri.be/#/Operations/updateCurrentsData). This indeed bypasses the RS485.

closed 2022-08-09 by koen-serneels