serkri/SmartEVSE-3 #58: MQTT support

ict-one-nl, 2022-08-09:
It would be really nice to have full MQTT support to publish status/measurement messages and to receive updates for the charging/current/etc. It's a very light weight protocol, no polling required, guaranteed delivery options and possibility of tight integration with Home Assistant through the auto discovery topics.

koen-serneels, 2022-08-09:
Afaik it is already possible to integrate hass with an http api, with for example the rest sensor. Maybe this requires a bit more config than mqtt but it will surely work. Did you try that?

I do agree that mqtt would be a nice addition next to the http api, but for now we won't be adding mqtt to the esp. The main reason is that we feel that the esp is really good at the low level signalling stuff, but it gets hairy fast when trying to add more higher level use cases. 90% of the code (excl the lcd in the current hardware) can better be moved out towards a more higher level as it will allow for faster/better expansion (especially towards integrations like protocols such as mqtt or adapters for reading power measurments), easier testing, less regression etc. So, although mqtt is pretty light weight and runs of the esp,  we feel that this a dead end in the long run and better to combine everything like that and move it somewhere else

closed 2022-08-12 by koen-serneels

ict-one-nl, 2022-08-12:
Sorry wasn't able to respond sooner. 

An MQTT implementation is much cleaner than a polling http mechanism and it's also more light weight because it only has to do something when something changes. The MQTT stacks for ESP32/8266 are very mature, almost all of these kinds of application use MQTT with pub/sub rather than http polling like solutions.

Take a look at esphome or the million other frameworks that are used on ESP, many are very stable and heavilly dependent on MQTT. MQTT is the the defacto standard when it comes to IoT on the ESP32 so I don't really understand your reluctance in this matter to be honest :)

Sonusss, 2022-08-27:
I fully agree with @ict-one-nl MQTT is really lightweight. I use it on a bunch of ESP8266 without any problem !
If you consider IOT world philosophy it's not the controller that polls all nodes to get data, nodes sends their data on a regular basis and when something changes, this is a much better approach limiting also the traffic. Imagine large systems with hundreds of nodes to be polled... MQTT library PubSubClient (https://github.com/knolleary/pubsubclient)  is very stable and mature. I can fork your code to implement MQTT but I need than to analize and understand your code which is time consuming while you would probably be able to make it in less than one hour...

synegic, 2022-10-05:
Attempted to implement MQTT in my pull request; https://github.com/serkri/SmartEVSE-3/pull/69. First I tried to use the REST method but it just didn't work and decided to just wing the MQTT support in. That went better than expected and within two days it was working. I actually did not see this issue and the maintainer's stance on MQTT. Luckily I made it possible to switch the functionality on/off in the core code.

I definitely agree with the people in this issue though; MQTT is much more suited for the job, especially if you use Home Assistant (but really, any other automation software with MQTT as first-class citizen profits). Rest API is nice and all, but as already stated much heavier/less elegant/cumbersome, especially if you need to update values with a high'ish frequency.

I think there is a place for both; the ESP32 is a very powerful little microcontroller and should be able to handle this just fine. The real question regarding my pull if it handles my crappy code. :D

thucar, 2023-02-08:
I agree with the above. MQTT is designed from ground up for exactly this type of communications. Sending data frequently from a metering device to the EVSE and getting progress updates from EVSE to the rest of the system. The bandwidth needed by HTTP API is many times higher compared to MQTT.