serkri/SmartEVSE-3 #61: charged_kwh & charged_total disappeared after upgrading to 1.4

rmeekers, 2022-08-25:
I've upgraded earlier this week from 1.3 to 1.4 in order to get rid of the wifi issues. But since the upgrade the `charged_kwh` & `charged_total` are not present anymore in the API (and also not visible on the webpage dashboard).
Do I need to do something after the upgrade or is this a bug?

k-janssens, 2022-08-25:
Did you take the manual build of dingo35? That is not yet the full build, hopefully this week we can finally merge 1.4 with the master and have the final build ready but I need to resolve some merge conflicts first

deqw, 2022-08-25:
Maybe related; I thought in earlier releases I had more info in the raw data, like the Mainsmet address.

With the @dingo35 release I get:

`{"version":"v3serkri-0.00","mode":"SMART","mode_id":3,"car_connected":"false","evse":{"temp":34,"connected":"false","access":true,"mode":1,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":13,"current_main":25,"solar_max_import":0,"solar_start_current":4,"solar_stop_time":10,"3phases_enabled":true,"mains_meter":"Eastron"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"EVMeter_import_active_energy":718},"mains_meter":{"MainsMeter_import_active_energy":1755.9,"MainsMeter_export_active_energy":0},"phase_currents":{"TOTAL":30,"L1":8,"L2":16,"L3":6,"last_data_update":1661411985,"original_data":{"TOTAL":30,"L1":8,"L2":16,"L3":6}},"backlight":{"timer":0,"status":"OFF"}}`

k-janssens, 2022-08-25:
Below you'll see what I get in my last 1.4 build, this seems to contain everything?

`{
"version": "v3serkri-1.4.0",
"mode": "SOLAR",
"mode_id": 2,
"car_connected": true,
"evse": {
"temp": 38,
"temp_max": 70,
"connected": true,
"access": true,
"mode": 2,
"solar_stop_timer": 0,
"state": "Connected to EV",
"state_id": 1,
"error": "None",
"error_id": 0,
"rfid": "Not Installed"
},
"settings": {
"charge_current": 160,
"override_current": 0,
"current_min": 6,
"current_max": 16,
"current_main": 40,
"solar_max_import": 0,
"solar_start_current": 6,
"solar_stop_time": 2,
"3phases_enabled": false,
"mains_meter": "Sensorbox"
},
"home_battery": {
"current": -12,
"last_update": 1661412475
},
"ev_meter": {
"description": "Disabled",
"address": 12,
"import_active_energy": 0,
"total_kwh": 0,
"charged_kwh": 0
},
"mains_meter": {
"import_active_energy": 0,
"export_active_energy": 0
},
"phase_currents": {
"TOTAL": 11,
"L1": 9,
"L2": 6,
"L3": -4,
"last_data_update": 1661412476,
"original_data": {
"TOTAL": -1,
"L1": 5,
"L2": 2,
"L3": -8
}
},
"backlight": {
"timer": 0,
"status": "OFF"
}
}`

deqw, 2022-08-25:
In my output I see at least the Mainsmet address missing (should be 101), also the EVMeter info is incomplete (I have the same Eastron there at address 102.

So your version looks more complete. Will have another look at the raw data when you release 1.4.0

dingo35, 2022-08-25:
@k-janssens everything is in there; AFAIK mainsmeter address has never been in there; the last 1.4.0 build has some extra evmeter info as compared to "my" earlier 1.4.0 build, which didn't have some of your / Koens changes since you did them in some other branch.

So new finally everything is converging into one branch. Again, AFAIK the merge went fine!

rmeekers, 2022-08-25:
> Did you take the manual build of dingo35? That is not yet the full build, hopefully this week we can finally merge 1.4 with the master and have the final build ready but I need to resolve some merge conflicts first

I used this one: https://github.com/serkri/SmartEVSE-3/issues/53#issuecomment-1221605205
So that is probably the issue... thanks :)

This is my output:

```
{
   "version":"SERKRI-1.4.0",
   "mode":"SMART",
   "mode_id":3,
   "car_connected":"true",
   "evse":{
      "temp":37,
      "connected":"true",
      "access":true,
      "mode":1,
      "charge_timer":0,
      "solar_stop_timer":0,
      "state":"Connected to EV",
      "state_id":1,
      "error":"None",
      "error_id":0,
      "rfid":"Not Installed"
   },
   "settings":{
      "charge_current":0,
      "override_current":0,
      "current_min":6,
      "current_max":16,
      "current_main":20,
      "solar_max_import":0,
      "solar_start_current":4,
      "solar_stop_time":10,
      "3phases_enabled":false,
      "mains_meter":"Sensorbox"
   },
   "home_battery":{
      "current":0,
      "last_update":0
   },
   "phase_currents":{
      "TOTAL":-7,
      "L1":22,
      "L2":-18,
      "L3":-11,
      "last_data_update":1661415341,
      "original_data":{
         "TOTAL":-7,
         "L1":22,
         "L2":-18,
         "L3":-11
      }
   },
   "backlight":{
      "timer":0,
      "status":"OFF"
   }
}
```

deqw, 2022-08-25:
> @k-janssens everything is in there; AFAIK mainsmeter address has never been in there; 

Ok, I went back to 1.3.8 to confirm what I saw. Here is the raw data and it does show the Mainsmet address.

`{"version":"SERKRI-1.3.8","mode":"SMART","mode_id":3,"car_connected":"false","evse":{"temp":35,"temp_max":65,"connected":"false","access":true,"mode":1,"charge_timer":0,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":13,"current_main":25,"solar_max_import":0,"solar_start_current":4,"solar_stop_time":10,"3phases_enabled":true,"mains_meter":"Eastron"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron","address":102,"power":0,"total_kwh":718054,"charged_kwh":0},"phase_currents":{"TOTAL":30,"L1":8,"L2":16,"L3":6,"last_data_update":1661415477,"original_data":{"TOTAL":30,"L1":8,"L2":16,"L3":6}},"backlight":{"timer":113,"status":"ON"}}`

deqw, 2022-08-25:
No I did not check correct. It is the other Eastron. The mainsmet has address 101, and is indeed missing.

k-janssens, 2022-08-26:
The new release has been build:
https://github.com/serkri/SmartEVSE-3/releases/tag/v1.4.1-serkri-v3

rmeekers, 2022-08-26:
> The new release has been build: https://github.com/serkri/SmartEVSE-3/releases/tag/v1.4.1-serkri-v3

Great. Issue is resolved!

closed 2022-08-26 by rmeekers