serkri/SmartEVSE-3 #64: Wrong charging current in Smart mode with 1.4.2

fluppie, 2022-08-30:
For some reason the new 1.4.2 limits my charge current in Smart mode. Normal mode works well.
Anything change in the calculation? I have read something about Max Circuit here: https://github.com/serkri/SmartEVSE-3/pull/21/files
At the moment it charges 16A-17A instead of 32A setpoint (=30.6A real current). I think it's using the default value of Max Circuit https://github.com/serkri/SmartEVSE-3/blob/1abdea05e8772e7579914541e59dfc6bf8ea3005/SmartEVSE-3/include/evse.h#L129 
But I don't have a menu item to change Max Circuit to 32A.

<html><body>
<div id="content"><div class="tabs "><nav class="tabs-navigation"><ul class="tabs-menu" role="tablist"><li class="tabs-menu-item json is-active" role="presentation"><span class="devtools-tab-line"></span></li></ul></nav></div></div><span class="devtools-tab-line"></span><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"><div class="devtools-separator"></div><div class="devtools-searchbox"><input class="searchBox devtools-filterinput" placeholder="JSON filteren" value=""></div></div><div class="panelContent">

  |  
-- | --
version | "SERKRI-1.4.2"
mode | "SMART"
mode_id | 3
car_connected | true
wifi |  
status | "WL_CONNECTED"
ssid | "WiFi2020"
rssi | -48
bssid | "F0:9F:C2:F4:01:56"
auto_connect | false
auto_reconnect | true
evse |  
temp | 44
temp_max | 65
connected | true
access | true
mode | 1
solar_stop_timer | 0
state | "Charging"
state_id | 2
error | "None"
error_id | 0
rfid | "Ready to read card"
settings |  
charge_current | 169
override_current | 0
current_min | 6
current_max | 34
current_main | 40
solar_max_import | 1
solar_start_current | 4
solar_stop_time | 10
3phases_enabled | false
mains_meter | "Eastron"
home_battery |  
current | 0
last_update | 0
ev_meter |  
description | "Eastron"
address | 102
import_active_energy | 0
total_kwh | 3613.6
charged_kwh | 0.8
mains_meter |  
import_active_energy | 6810.9
export_active_energy | 2329.7
phase_currents |  
TOTAL | 183
L1 | 183
L2 | 0
L3 | 0
last_data_update | 1661893259
original_data |  
TOTAL | 183
L1 | 183
L2 | 0
L3 | 0
backlight |  
timer | 120

</div></div></div></div>
</body>
</html>

dingo35, 2022-08-30:
Try changing Loadbalance to Master, MaxCircuit will appear in the menu, and see if changing that works...

k-janssens, 2022-08-31:
I've only added data to the endpoint so no changes in the calculations

rvdgaag, 2022-09-02:
I experienced similar behaviour with this version, after a hard reset (FLI off/on) it worked as it should.

fluppie, 2022-09-03:
I was on 1.3.8 before updating to 1.4.2 so I had the old calculation. I'll try to go back on 1.4.2 later this weekend.
@rvdgaag what do you mean with FLI off/on?

rvdgaag, 2022-09-04:
The circuit switch.

k-janssens, 2022-09-12:
Is there still an issue with this?

fluppie, 2022-09-18:
I think so, still on 1.3.8 I'll update again to 1.4.2 and retry. I will also cut 230V to SmartEVSE v3 after the update so a full reboot is the case. Car is currently at 100% so I cannot test right away :).

fluppie, 2022-09-19:
Jup, still 16A limit
[image]

fluppie, 2022-09-19:
Putting loadbalancing on Master so the MaxCircuit appears makes it working again. But that's a workaround for a bug that's not there in 1.3.8. Just after setting 33A as max circuit current, the SmartEVSE is flipping. A hard power cycle of the module solves the issue. I'll see if I can upload a video.
I have a mono phase setup with Eastron SDM230 kWh meters for MainsMeter and EVMeter.

fluppie, 2022-09-19:
https://user-images.githubusercontent.com/8402106/191032819-2af89601-f443-4ad1-87b4-1a44405fa5be.mp4

dingo35, 2022-09-22:
I can look at this when I am back at my SmartEVSE location, which will be end of November...

dingo35, 2022-11-24:
@fluppie, my setup is very much like yours:  Eastron SDM72 kWh meters for MainsMeter and EVMeter, but I have a three phase 25A mains connection.

I cannot reproduce this problem. Is it reproducable on current version 1.4.4?

closed 2023-02-18 by dingo35