serkri/SmartEVSE-3 #66: Solar mode

martijndierckx, 2022-09-07:
I'm wondering on how the solar mode works, because I can see a couple of use cases:

1. Solar-Only mode: The car will only charge when the mains-meter is returning at least 6A. And the charge current will be updated continuously to the amount of energy which would normally be pushed back to the grid.
2. Solar-Surplus mode: Same as above, but on top of a setup minimum charge power. F.e. 6A, this way the car will always charge, but when there is left over PV power, it will charge faster.

Are both modes somehow already supported via some combination of settings?

To make these options a bit more complicated, there are 2 options to meter:
- Physically correct: Checks per phase how much power is available and uses the lowest value to check wether it may charge using PV power.
- Billing optimised: In Belgium (and probably in other countries as well), billing is done across phases. So you only need to check the totals across phases. As long as there is 18A available somewhere over the 3 phases, you can charge on those 3 phases at 6A without your billing meter increasing.

Your thoughts please...

dingo35, 2022-09-07:
Good starting point!

For mode 1, I think it would not wise to stop/start charging when a cloud passes by; so the existing timer I would modify to a timer that starts when there is not enough power to charge on solar, and on end actually stops the charging.

Holland also has "billing optimised", so I see no application/business case for "physically correct".

martijndierckx, 2022-09-08:
You can take a look at the EVCC project. They implemented some thresholds to make sure the charging isn't being started/stopped continuously, nor is the charging power.

dingo35, 2022-09-08:
Which thresholds do you like/use?

martijndierckx, 2022-09-08:
EVCC uses these in the yml config:

```
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 3m # switch charger contactor not more often than this (default 5m)
```

k-janssens, 2022-09-08:
Unless I misread your question, this is already supported by the original fw using this config parameter:
`START set the current on which the EV should start Solar charging -0 -48A (sum of all phases)`
`STOP Stop charging when there is not enough solar power available Disabled - 60 minutes (Disabled = never stop charging)`
`IMPORT Allow additional grid power when solar charging (0-20A))`

Even though this is set in minutes, have a 1 or 2 minutes buffer before charging stops should not be the end of the world and it avoid unnecessary abrupt stops because a short dip in solar.

And if you set it to disabled, it will keep charging at the START value.

You could also play with the IMPORT value, for example: START: 6, STOP: 2, IMPORT 4
This will keep charging the car with 2A solar, but still getting a total of 6A charging current. Only to stop charging if the import grows higher than 4A over more than 2 minutes.

dingo35, 2022-09-08:
The timer is not Reset after the cloud has disappeared; so cloud comes by, passes, timer keeps running, switches off after timer expires, and then immediately starts charging again (assuming no clouds).

I experienced this behavior with the original firmware, but AFAIK nothing in solar has changed in the serkri distribution.

k-janssens, 2022-09-08:
That is not my experience, I run almost only in solar mode and I have seen the countdown being stopped.
It is however not immediate, it requires some time of  sufficient solar available before it stops the countdown.

dingo35, 2022-09-08:
Ok I stand corrected, I only tested once and gave up, thanks for the correction.

Another point needed is driving the contactor2, so solar mode can profit from scaling down to 1 phase charging when less then 18A solar power is available.

k-janssens, 2022-09-08:
Yes, solar is not compatible with 3 phases, I have an idea of making it so but I can't test it myself as my car is 1 phase only.
So have not yet put in the time to make the change.

rvdgaag, 2022-09-10:
Solar also works fine with 3 phases in my case, you just need a bigger solar installation on the roof to provide enough current.  I'm hoping the second connector for the Smart EVSE becomes available so you can choose to select relay 1 or 2 depending on your needs.

k-janssens, 2022-09-12:
@rvdgaag When I looked at the code, it appeared for me the logic does not take into account 3 phases.

What I mean with that: 
If there is for example 6A running to the grid, the EVSE will tell the car to start charging at 6A.
But if the car can use 3 phases, it will charge not with 6A but with 18A, which will result in the EVSE to stop charging.
After it has stopped, it will see again 6A of current available and repeat this cycle.

This is of course an assumption based on the logic of the code and how I think the car will react.

Could you chime on this with your experience? And how you have configured your EVSE?

FYI, you can already control a 2nd contactor if you want: https://swagger-ui.serkri.be/#/Operations/updateSettings
Please also read this thread for more details on the wiring: https://github.com/serkri/SmartEVSE-3/issues/31

rvdgaag, 2022-09-12:
Yes with 3 phases it will charge 3x6A so 18 amps. And it works fine. My grid is limited to 3x25A.

I assume that the EVSE only monitors 1 phase in the solar mode maybe ?

k-janssens, 2022-09-12:
No the EVSE can see all phases, but the logic for Solar does not keep in mind the amount of phases.

So if your car starts charging at 18A, and there isn't enough solar, what happens?
Did you configure your EVSE it can also take grid power?

rvdgaag, 2022-09-12:
Yes it can take gridpower ofcourse. as soon as the grid is used it starts to use the timer to shut down.

k-janssens, 2022-09-12:
Ah yes, like I expected. Well that is what I meant with the current logic not being compatible with 3 phases.
At some point I want to make the EVSE aware there are N phases. And thus also use the N for calculating when solar can charge. With 3 phases it should only start when there is 18A available.
This in combination with an automatic enabling/disabling of the C2 contactor, it should optimize the solar usage efficiency.

rvdgaag, 2022-09-12:
I like what I hear from you !

k-janssens, 2022-09-22:
fyi: I've added a new config parameter that can be changed through the /settings endpoint, it is available in 1.4.3.
This will allow you to keep charging at minimum current in solar mode if there is not sufficient sun.

dingo35, 2023-01-11:
The logic for switching second contactor on/off is added in v1.5.0 . Closing issue.

closed 2023-01-11 by dingo35