serkri/SmartEVSE-3 #75: Factory Reset

Schnitzn, 2022-10-24:
Unfortunately my EVSE does not recognise my EV anymore when connecting.
I already checked and temporarily switched cable and car but these seem to be OK communication-wise. Firmware + Spiffs re-upload also did not fix it.

How can I make factory reset to the EVSE to start from scratch?

[image]

dingo35, 2022-10-24:
For a factory reset, just reflash the original firmware and spiffs.

To diagnose, recompile the serkri distribution after modifying include/evse.h : comment
#define DEBUG_DISABLED 1, and modify 
#define LOG_EVSE LOG_INFO to
#define LOG_EVSE LOG_DEBUG;

After reflashing you can telnet to your SmartEVSE, and it will tell you what value is measured at the PP pin; that determines the value of "CONNECTED".

Post your PP pin value here when your car is connected....

DISABLE_DEBUG to false and WARN to DEBI

Schnitzn, 2022-10-25:
Thanks. When reflashing original firmware, my settings still remain stored (WIFI, current limit...).
Any way to make a full wipe?

dingo35, 2022-10-25:
With esptool, flash all zeroes to the esp32, then reflash firmware and spiffs. This will zero out the NVS settings.

Schnitzn, 2022-10-31:
Thanks. "http://EVSE IP-address/erasesettings" also worked out even EV connection issue was finally not related to this. 
Thanks to Michael Stegen

closed 2022-10-31 by Schnitzn