serkri/SmartEVSE-3 #77: Read and display EV meter currents and use them to limit power to MaxCircuit

DariusPL, 2022-11-18:
Hi, I am testing smartevse with Serkri firmware v1.4.1
Thank you for lots of improvements.
A feaure with display of EV meter currents is something that waiting for. Does current values intend to be display on LCD or API as well(?)

Would be grest to get some guidelines shining some light how power limitations function is being performed. The evse LCD shows the currents value that comes from the EV meter (Eastron) however do not experience any power limitations due to these inputs using smart/solar modes 

"Read and display EV meter currents and use **them to limit power to MaxCircuit (#21)**

See that one of the changes vs smartevse firmware incl. add current endpoints. Are these values readable without sensorbox(?)
"Send L1/2/3 current information 
Note: Set MainsMeter to 'API' option in the config menu when sending L1/2/3"

dingo35, 2022-11-18:
It is hard to understand your post; I guess we're both non-native in English.

The current of SmartEVSE will be limited to that the MainsMeter current will not exceed MaxCurrent. This goes for both solar and smart mode.

In addition to that, a very specific situation when SmartEVSE is on a subpanel with other users (e.g. a washer), you can put EVMeter on the level of the subpanel and of Loadbalance=Yes the load on the subpanel will not exceed MaxCircuit.

DariusPL, 2022-11-21:
Thank you ***@***.*** *for your reply.
Indeed, that would be easier if we all could communicate in one language. I
am sorry to make you confused.
Have gathered below my queries.
1. Does the EV meter currents read possibly to display in the API  Charged
kwh & Charged_total appear in the API when using EVMeter.
2. The original SmartEVSE firmware is capable to limit MaxCurrent not
seeing the real EV charging current. Is the ability to read EV meter
currents made any change with regard to the original SmartEVSE (MaxCurrent)
code. As far as I understand the goal for readings of the EV meter currents
are for use to limit power to MaxCircuit only (?)
3. Phases endpoint to replace sensorbox (#7) * Added endpoint to replace
sensorbox. --> Is this for situation where the phase currents reads come
from the external device (e.g home battery) instead of sensorbox(?)

closed 2022-11-24 by dingo35