serkri/SmartEVSE-3 #82: Building Firmware fails due to missing bootloader

ChristianSteffens, 2022-12-25:
Hello,

I just tried building your firmware using `platformio run`. I installed `platform.io` using `brew` on my Mac.
Even though the dependencies seem to be fetched and used, the actual build process fails due to a missing bootloader file.

```
Building in release mode
*** [.pio/build/release/bootloader.bin] ...
 ... Source `/Users/FooBar/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/bin/bootloader__80m.elf' ...
 ... not found, needed by target `.pio/build/release/bootloader.bin'.
```

Am I'm using the wrong release (the `master` branch of your repository).

dingo35, 2022-12-26:
I remember having had that same error message years ago, and I don't remember what solved it; it is a problem in you platformio - core installation.

I just installed a clean Ubuntu OS with a clean platformio 6.1.5 installation and with that everything compiled fine; you might try a pip install -U platformio as normal user, that might fix your installation... I'm sorry but python is not my native language :-)

ChristianSteffens, 2022-12-27:
OK - [I applied these changes](https://github.com/SmartEVSE/SmartEVSE-3/issues/36). Now it's working. Thank you.

closed 2022-12-27 by ChristianSteffens

dingo35, 2022-12-30:
Committed 46f4f6c9c16c638064f074c7a12b6b61361924f7