serkri/SmartEVSE-3 #85: Change serial number when update to 1.5.1

Xnygra, 2023-01-08:
My serial number where changed to a new serial number when update to 1.5.1. Not sure if it really impacts anything, but i had to change the local url to the new serial number. The new serial number is only, to my knowledge, found under its name on the wifi network. Could be an issues if you need to claim a warranty repair I guess.

dingo35, 2023-01-09:
I cannot reproduce this; it is strange because the serial number is calculated from the MAC address, and AFAIK this algorithm hasnt changed from the original firmware.

Two questions:
1. from which firmware did you upgrade to 1.5.1?
2. From what to what changed your serial number?

Xnygra, 2023-01-09:
1. I upgraded from 1.4.4
2. My serial number was 05455, which is also on the physical label on the unit, to 49475.

I noticed, if it matters, that when i had my original serial number, the leading 0 was omitted in the network wifi name and the URL, so it was "smartevse-5455.local"

dingo35, 2023-01-09:
Very interesting! Turns out that when your SmartEVSE is turned on, it calculates a 5 digit serial number from the MAC address, and uses that to put up the AP for wifi configuration. And later on it calculates a 4 digit serial number from a pre-programmed value. To prevent confusion I disabled the second number, but now you say the second one is on the physical label, my conclusion is that I made the wrong decision....

[1.5.2RC.zip](https://github.com/serkri/SmartEVSE-3/files/10372644/1.5.2RC.zip)
Would you be so kind to test the included firmware.bin (spiffs.bin has not changed); it should give you back your "old" serial number, and hopefully it still gets recognized by the HomeAssistant component....

Xnygra, 2023-01-09:
Yes, got my old serial number. perfect, thanks. But still not working with the HA component (ha-SmartEVSEv3) serial number not recognised when typing the 5 digit, with the leading 0, when omitting that and typing the last 4, 5455 it gets recognised but only leds me to a new window with the only option to "Close" and no action is taken.

dingo35, 2023-01-09:
That is strange, it should bring you to the (standard) window where the area for the device can be chosen. I have one other user with the same problem, but I can't reproduce it.
Can you post your logs (grep hass /var/log/syslog or systemctl status home-assistant@homeassistant ) ?

dingo35, 2023-01-09:
Also  you could re-download the custom component files from github, the old version used the OS'es ping command to discover the device, that dependency is removed in the newest version....

closed 2023-01-10 by dingo35