serkri/SmartEVSE-3 #87: communication error in Normal mode met Mains Meter op EM_API

dingo35, 2023-01-10:
via tweakers:
[https://gathering.tweakers.net/forum/list_message/74075026#74075026](url)

Ik heb net serkri naar 1.5.1 geüpdatet in de smartevse en de custom component geinstalleerd. Na wat puzzelen hoe ik achter de serial nr van mijn smartevse kon komen praten ze met elkaar en dat werkt an sich prima.

Enige probleem dat ik heb dat hij nu in ook in normal mode een communication error blijft geven. Mainsmeter staat op api wanneer ik hem op smart zet. Gebruik hem eigenlijk tot nu toe op altijd normal en dat werkte prima zonder L1, L2 en L3 door te geven als ik het goed heb. Nu werkt mijn laadpaal dus niet :'(

komt dat omdat ik deze al eens door heb gegeven en hij oude gegevens gebruikte of is er iets anders veranderd?

dingo35, 2023-01-10:
I cannot reproduce your problem, but what happens if you change your MainsMeter to the default setting (SensorBox)?

dingo35, 2023-01-10:
I just managed to reproduce the problem, and setting your Mainsmeter to Sensorbox (the default setting in the original firmware) solves the problem in my setup...

fluppie, 2023-01-10:
I have the same issue. Setting back on Smart, then Sensorbox as MainsM and then back on normal doesn't solve it. The 1.4 versions don't have this behaviour.

dingo35, 2023-01-10:
Could you press "Raw" on the webserver and post your settings?

fluppie, 2023-01-10:
{"version":"v3serkri-1.5.1","mode":"NORMAL","mode_id":1,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"WiFi2020","rssi":-70,"bssid":"78:8A:20:5A:41:76","auto_connect":false,"auto_reconnect":true},"evse":{"temp":21,"temp_max":65,"connected":true,"access":true,"mode":0,"solar_stop_timer":0,"state":"Charging Stopped","state_id":9,"error":"Communication Error","error_id":2,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":32,"current_main":32,"solar_max_import":0,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Always Off","mains_meter":"Sensorbox"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_energy":0,"total_kwh":0,"charged_kwh":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0,"last_data_update":0,"charging_L1":false,"charging_L2":false,"charging_L3":false,"original_data":{"TOTAL":0}}}

dingo35, 2023-01-11:
Could you please test this version?

[1.5.3RC.zip](https://github.com/serkri/SmartEVSE-3/files/10390797/1.5.3RC.zip)

fluppie, 2023-01-11:
Lijkt te werken :)

Tochstiekumgordon, 2023-01-11:
The 1.5.3rc sort of works.
After the update it defaults to smart mode which then triggers a communication error. Swiching to normal does not clear the error until you send current data. The error in normal mode then clears and you can stop sending data. This is repeatably by switchting to smart or solar and back again. The error in normal mode remains present.

systeembeheerder, 2023-02-07:
Straight upgrade from https://github.com/SmartEVSE/SmartEVSE-3/releases/tag/v3.0.1 No sensorbox, no modbus, no smart, no solar, just normal charging,  to https://github.com/serkri/SmartEVSE-3/releases/download/v1.5.1/v1.5.1-dist.zip also gives "ERROR NO SERIAL COM CHECK WIRING".

Which makes this fork  quite unusable for me.

fluppie, 2023-02-07:
Have you tried [1.5.3RC.zip](https://github.com/serkri/SmartEVSE-3/files/10390797/1.5.3RC.zip) ?

XanderF, 2023-02-12:
I have the same issue: ERROR NO SERIAL COM CHECK WIRING. I started with the default firmware and updated to 1.5.2 yesterday. I'm using a setup without sensor box. Just want to be able to start and stop charging via Home Assistant. I don't need load balancing, solar charging, etc. So I would use normal mode only, single fase at 10A.

Should it be possible to use this custom firmware without a sensor box or is it mandatory?

dingo35, 2023-02-12:
Use 1.5.3rc and you should be OK in Normal mode.

dingo35, 2023-02-12:
> The 1.5.3rc sort of works. After the update it defaults to smart mode 

It defaults to the last mode you set at the LCD display. This is intended behavior.

XanderF, 2023-02-13:
I've installed 1.5.3rc. The error is gone :)

I doesn't seem to default to the last state, as mentioned above. It defaults to 'NORMAL' here. I've set it to 'OFF' and if I press reboot, it changes to 'NORMAL'. Not an issue for me though.

Thank you for the update!

dingo35, 2023-02-13:
It remembers the last state it is set to THROUGH THE LCD DISPLAY.

The mode changes by the webserver are considered temporary changes.

deqw, 2023-02-14:
> It remembers the last state it is set to THROUGH THE LCD DISPLAY.
>
> The mode changes by the webserver are considered temporary changes.

Can you add the OFF mode to the list of options that can be set throught the LCD display? I need it to default to OFF when the SmartEVSE boots.

When in OFF mode the display shows ACCESS DENIED. To what is access denied at that point?

dingo35, 2023-02-14:
It is the mode that the SmartEVSE is in when a RFID reader is connected, but no RFID card has granted access.
Adding "off" mode would make no sense. Try using HA and/or fiddling with RFID settings.

deqw, 2023-02-14:
> It is the mode that the SmartEVSE is in when a RFID reader is connected, but no RFID card has granted access.
> 
> Adding "off" mode would make no sense. Try using HA and/or fiddling with RFID settings.

I have no RFID reader and RFID is Disabled. I do have RCMON enabled. When I disabled RCMON the Access Denied disappeared. After enabling RCMON again Access Denied did not return. Had never changed anything in these settings since first setup.

Adding OFF does make a lot of sense. When the car is connected and the SmartEVSE reboots in NORMAL mode the car starts charging right away. Then HA kicks in and sets mode OFF because the current situation is not within my charging policy, which can be No Charging, Price, Solar, Solar then Price, Charge Now, Timer. This results in starting a charge session when this is not wanted, for maybe a few seconds to one minute, which can be avoided when the SmartEVSE can default to OFF.

dingo35, 2023-02-14:
You can emulate the access bit default OFF by:
-setting RFID reader to EnableOneCard (setting 2)
-setting Switch to  setting 1 or 2 (0:Disable / 1:Access B / 2:Access S / 3:Smart-Solar B / 4:Smart-Solar S)

Can't test this myself currently....

deqw, 2023-02-14:
That OFF emulation does work.

The ACCESS DENIED message is there when mode is OFF with whatever RFID setting. When the above emulation is set it asks for the card, of course.

closed 2023-02-18 by dingo35