serkri/SmartEVSE-3 #90: Default setting to solar after 230V reset

rvdgaag, 2023-01-13:
The current distribution switches on in SOLAR mode when the EVSE had a hardreset after a 230V off/on.

I would prefer the mode to be SMART, so when there is a power outage at night the car will be full instead of empty waiting for the sun to come up.

Thanks

dingo35, 2023-01-13:
Totally agree that this is unwanted behaviour. But I have no clue why this behaviour changed, and currently I'm not near my SmartEVSE so I cannot test. 
It is strange because when you change the mode setting, it should be saved with all the other settings, and at cold boot those settings are read to initialize the variables.

To be clear, with which version did you notice this? And if you are able to compile / run a debug version: do you see a debug message (level Info) "settings saved" , after you change the mode setting?

rvdgaag, 2023-01-15:
It was also with version 1.5.1, now with 1.5.2

I'm not able to compile, a debug version should be ok but with my Mac a telnet session to the evse doesn't work. I have a windows notebook laying around somewhere so I can give it a try.

dingo35, 2023-01-15:
Ok I'm not a Mac user but the internet tells me Mac OS is supposed to have telnet on boars. Why wouldn't it  off of your Mac?

rvdgaag, 2023-01-15:
Ok with a homebrew hack for macos I was able to add telnet

So where should I look ?

dingo35, 2023-01-15:
Ok the 1.5.2 release has debug enabled, so you should be able to telnet into your device; after the telnet welcoming screen, enter "i" followed by enter, so now you have debug level "info".
Now if you change mode, it should print "settings saved"?

rvdgaag, 2023-01-16:
So here is the output when I switch via de webinterface from off to SMART

* Debug: Command received: i
* Debug level set to Info
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(operator())(C1) page / (root) requested and sent
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0^[ A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.9 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.5 A L2: 4.9 A L3: 1.0 A Isum: 5.4 A

rvdgaag, 2023-01-16:
from SMART to off

***
i(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.6 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.6 A L3: 1.0 A Isum: 5.0 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
[1;33m(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
[1;33m(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 4.5 A IsetBalanced: 16.0 A
(I) (UpdateCurrentData)(C0) L1: -0.6 A L2: 4.5 A L3: 1.0 A Isum: 4.9 A

rvdgaag, 2023-01-16:
From off to on via an external relay, it switches after a reset to Solar:

* Debug: Command received: i
* Debug level set to Info
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.7 A L2: 2.3 A L3: 1.0 A Isum: 2.6 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.7 A L2: 2.3 A L3: 1.0 A Isum: 2.6 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.7 A L2: 2.3 A L3: 1.0 A Isum: 2.6 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.7 A L2: 2.3 A L3: 1.0 A Isum: 2.6 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.7 A L2: 2.3 A L3: 1.0 A Isum: 2.6 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.4 A L3: 1.0 A Isum: 2.6 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.4 A L3: 1.0 A Isum: 2.6 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
^[[B(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
(I) (UpdateCurrentData)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (UpdateCurrentData)(C0) L1: -0.8 A L2: 2.3 A L3: 1.0 A Isum: 2.5 A
^C

dingo35, 2023-01-16:
So just to check if I understand you correctly: the mode at boot is determined by the last setting at the lcd display.

It isn't designed on purpose, but it turns out to be a feature instead of a bug: you put smart mode on on the LCD screen, change it with webinterface and or API, but at reboot it will return on smart mode.

Right?

rvdgaag, 2023-01-16:
yes correct