serkri/SmartEVSE-3 #93: Smart mode not always working

daninden, 2023-01-19:
So I just looked at my energy usage and saw my phase which I use for loading was at 28a all the time, car was charging at 25a.
My mode was set at Smart, my main max is 25.
After restarting twice it started to work again. I haven't seen this behaviour before, but found it a little disturbing.
Probably not easy to find out why.

dingo35, 2023-01-19:
You're right, that is not an easy thing to debug.
What version did you notice this on?
Could you flash a debug version on your device, and when you see a similar situation, telnet to it and post the STATE line that is reported ?

Also I remember you are feeding your smartevse the mainmeter information via EM_API; are you sure your feed was up-to-date when this situation occurred?

daninden-priva, 2023-01-19:
the information was indeed up-to-date... haven't seen this happening before... after I updated it to the latest version (around 18:10) it seemed to be stable now. 

Here's the history of the L1 phase pulled from the smartevese api (L2 and L3 were not negative)
[image]

I will monitor it some more.

daninden, 2023-01-20:
i'll close it for now.. if it reoccurs.. i'll let you know

closed 2023-01-20 by daninden