serkri/SmartEVSE-3 #94: Very interesting! Turns out that when your SmartEVSE is turned on, it calculates a 5 digit serial number from the MAC address, and uses that to put up the AP for wifi configuration. And later on it calculates a 4 digit serial number from a pre-programmed value. To prevent confusion I disabled the second number, but now you say the second one is on the physical label, my conclusion is that I made the wrong decision....

deqw, 2023-01-30:
Very interesting! Turns out that when your SmartEVSE is turned on, it calculates a 5 digit serial number from the MAC address, and uses that to put up the AP for wifi configuration. And later on it calculates a 4 digit serial number from a pre-programmed value. To prevent confusion I disabled the second number, but now you say the second one is on the physical label, my conclusion is that I made the wrong decision....

[1.5.2RC.zip](https://github.com/serkri/SmartEVSE-3/files/10372644/1.5.2RC.zip)
Would you be so kind to test the included firmware.bin (spiffs.bin has not changed); it should give you back your "old" serial number, and hopefully it still gets recognized by the HomeAssistant component....

_Originally posted by @dingo35 in https://github.com/serkri/SmartEVSE-3/issues/85#issuecomment-1375477614_

deqw, 2023-01-30:
Seeing the serial number change with 1.5.1 and 1.5.2. Was 3426, now 5264. The router reports smartevse-3426 but can only be reached by smartevse-5264.

dingo35, 2023-01-30:
What does the physical label on your smartevse say as a serial number?

deqw, 2023-01-30:
Already looked for that yesterday: there is no label. But from the first day the number has been: 3426 (had a note of that in my node-red node).

[image]

dingo35, 2023-01-30:
You probably need to have some patience to have mDNS processed the change.
Note that the formal mDNS address would be  <SmartEVSE-3426.local> , noting upper and lowercase.

The fact that your smartevse can be reached by <smartevse-5264[.lan|.local]> is a strange behavior that stims from the original firmware; we just want to stay compatible.

deqw, 2023-01-30:
I which direction is that change going to happen? Will the SmartEVSE be reachable by SmartEVSE-3426.local in the end you expect?

SmartEVSE-3426 has always been the hostname reported by my router, so why does it not resolve? (rhetoric). As fas as I know SmartEVSE-3426.local has never worked before, I've always used the IP address because of that. Until lately...I saw the other hostname in the HA integration.

I'm not attached to any of these numbers, but it is nice to be sure what is (and remains) correct because otherwise some of my flows will break when the hostname does not resolve.

closed 2023-02-13 by deqw