serkri/SmartEVSE-3 #100: Overriding charge current doesn't work in 1.5.2

ict-one-nl, 2023-02-04:
Running the latest from github, it does run and it's connected:

GET http://192.168.5.172/settings
```
{
    "version": "v3serkri-1.5.2",
    "mode": "NORMAL",
    "mode_id": 1,
    "car_connected": true,
    "wifi": {
        "status": "WL_CONNECTED",
        "ssid": "iotwinternet.ict-one.nl",
        "rssi": -70,
        "bssid": "C2:FB:E4:11:2A:7A",
        "auto_connect": false,
        "auto_reconnect": true
    },
    "evse": {
        "temp": 20,
        "temp_max": 65,
        "connected": true,
        "access": true,
        "mode": 0,
        "solar_stop_timer": 0,
        "state": "Charging",
        "state_id": 2,
        "error": "None",
        "error_id": 0,
        "rfid": "Not Installed"
    },
    "settings": {
        "charge_current": 160,
        "override_current": 0,
        "current_min": 6,
        "current_max": 16,
        "current_main": 25,
        "solar_max_import": 0,
        "solar_start_current": 4,
        "solar_stop_time": 10,
        "enable_C2": "Not present",
        "mains_meter": "Sensorbox"
    },
    "home_battery": {
        "current": 0,
        "last_update": 0
    },
    "ev_meter": {
        "description": "Disabled",
        "address": 12,
        "import_active_energy": 0,
        "total_kwh": 0,
        "charged_kwh": 0
    },
    "mains_meter": {
        "import_active_energy": 0,
        "export_active_energy": 0
    },
    "phase_currents": {
        "TOTAL": 0,
        "L1": 0,
        "L2": 0,
        "L3": 0,
        "last_data_update": 0,
        "charging_L1": false,
        "charging_L2": false,
        "charging_L3": false,
        "original_data": {
            "TOTAL": 0
        }
    }
}
```

POST http://192.168.5.172/settings

```
{
    "override_current" : 80
}
```

Returns null

```
{
    "mode" : 0
}
```

null

and so on :)

dingo35, 2023-02-08:
You must be doing something wrong with your POST command, because this works on numerous installation.
For API definition look at https://swagger-ui.serkri.be/#/.
For example how to POST in python see https://github.com/dingo35/ha-SmartEVSEv3

ict-one-nl, 2023-02-08:
Thanks, I figured it out. You need to provide the contents as query parameters which is a bit strange; would have expected that the options should be provided as JSON in the body of the request.

closed 2023-02-08 by dingo35

ict-one-nl, 2023-02-08:
I'm seeing that you closed this, but the structure of the api could be something to think about.