serkri/SmartEVSE-3 #101: EvMeter API endpoints

thucar, 2023-02-08:
Running version 1.5.1 and trying to use all API meters. 
I am able to POST Grid details without a problem but I can not see how to POST EV meter and Solar meter data.
Are these just placeholders for now?

dingo35, 2023-02-08:
The API defintion is right here: https://swagger-ui.serkri.be/#/

thucar, 2023-02-08:
Yes and that is what I have been using. However I only see an option to send Mains meter values (L1, L2, L3) but I do not see any option to send EV meter and Solar Meter values to the controller.

Unless I am missing something obvious here...

dingo35, 2023-02-08:
1. There is no such thing in the SmartEVSE design as a solar meter, there is only MainsMeter and EVMeter.
2. The values of EVMeter are taken off of the modbus, you cannot feed them with the API . Only the MainsMeter values you can feed through the API, and this is because users with a home battery want to prevent the home battery charging the EV battery and/or vice versa. 

Since the API goes through wifi (http) it is not a preferred way to set critical values.

thucar, 2023-02-08:
Ok, gotcha. So the option to select API as meter type for EV Meter is not intentional. I propose to remove that at some point to avoid similar confusion in the future

dingo35, 2023-02-08:
Yup can't test anything right now so let's leave this issue open as a reminder for me.

dingo35, 2023-02-14:
@thucar can you test this version for me? It is the current master version (1.5.2 + some small changes), added to it the change to skip the EM_API option at the EVMeter:
[1.5.2+-em_api.zip](https://github.com/serkri/SmartEVSE-3/files/10733030/1.5.2%2B-em_api.zip)

dingo35, 2023-02-21:
Fixed 8f1e650a1e50fa21451f64a54e896de3b6075543

closed 2023-02-21 by dingo35