serkri/SmartEVSE-3 #105: dsmr 4.2 datagram every 10 seconds from p1

evanes68, 2023-02-21:
The issue is that the EVSE also has a 10 second delay to require new API data for L1/L2/L3.

What happens is that the charging stops and starts randomly because of missing API information.

I would like to extend the delay in EVSE to 12 seconds before charging stops when no API data is received. This provides enough margin for the p1 data to be processed.

evanes68, 2023-02-22:
I will contact the energy provider to see if I can upgrade my meter to dsmr v5.

dingo35, 2023-02-22:
Alternatively you could install one of the supported modbus-kWh meter in your panel. You wouldnt need the Sensorbox in that case.

synegic, 2023-02-22:
I suppose the timeout could be a bit longer than 10 seconds. Most main breakers have C-characteristics and will only trip after being overloaded for minutes at a time. I figured this out while charging in Normal mode by accident and all breakers surviving, even while loading it with 32A for minutes at a time.  I guess making it configurable is even better.

Then again, if you set it through API you could also easily have your automation software send a new value more frequently, even if it's the older value. Just make sure it will stop sending values if there obviously is a communication problem between the P1 meter and your automation software.

evanes68, 2023-03-08:
I currently only send mainmeter values through the api when I actually receive data from p1 signal. I could store the last value and send data more frequently but I think this is bad practice, when p1 meter stops working but I keep sending the data because it is stored there can be unwanted situation. Now when p1 stops I also stop sending and the smartevse will error out correctly. I have been running with a timeout of 12 seconds, it does work. But I have contacted my energy company and they will exchange my meter for a dsmr v5 meter. They say it will take a few months due to shortages.

closed 2023-04-05 by dingo35