serkri/SmartEVSE-3 #106: Is "smart" mode functionnal without an EV meter ?

charlesparvin, 2023-02-22:
On my first charging attempt (with 1.5.1 firmware) I tried using the smart mode to protect Mains.

Charge current kept quickly going up and down between 6 and 32 Amps and SmartEVSE quickly stopped charging and entered the one-minute timeout. After a minute it did the same up and down again and quickly went back in timeout.

Is this happening due to the absence of an EV meter to adjust charging current correctly ? if so should smart mode become unavailable while "EV meter" option is set to disabled ?

dingo35, 2023-02-22:
First of all, please upgrade to the latest release. It is not useful to report possible issues on old releases.

To answer your question: you only need a Mainsmeter configured for Smart Mode, no EV Meter needed.

After upgrading to 1.5.3, surf to http://ipaddress-of-your-smartevse/settings and post the output here, that might give us some insight on what is happening.

charlesparvin, 2023-02-23:
I tried again on 1.5.3 (sorry since changelog wasn't mentionning anything related about charge I naively assumed there would be no change) and it's much better.

I tested with a 40 Amps limit and 32 Amp charge. When I turned on the oven (10 Amps + ~2 Amps for base load) SmartEVSE correctly limited the charge without going in timeout but it kept fluctuating in the 20-28 Amps range with a laggy feeling on the updates causing it to not find the optimal charge rate (which would be around 27-28 Amps in that case) and stay there.

I do suspect my API source (Victron Multiplus-II through Victron's Node Red addon) may not be updating quickly enough and may be causing this so I'll investigate more and I believe this issue can be closed until further notice

closed 2023-02-23 by charlesparvin