serkri/SmartEVSE-3 #107: High initial current on starting the charge

kletteroli, 2023-02-23:
Hello,
in Solar or Smart mode when plugging in the car or toggling the button to charge I notice an initial charge current (1-3s) with the Max current before it actually gets limited (to import + solar contrib). In this situation the charge current will exceed the setting for "MAINS".
I would like to initiate the charging with the MIN current for a few seconds before the charge current then ramps up. (instead of MAX and ramp down)
However as I am not very experienced in coding I find it difficult to figure out where in the code I could introduce such delay and would very much appreciate if somebody could help me out with some hints or a sketch. (I already tried to increase the CHARGE_DETECTION_TIME).

The background is: My off grid setup is a battery inverter in combination with a grid inverter. When the battery is full, the grid inverter is in idle mode and the regulation works rather slow. When this is the case the maximum current is limited to what only the battery inverter can provide whereas when the grid inverter is contributing the maximum current that the system can provide is higher.

dingo35, 2023-02-24:
This is strange behaviour because right at the start of the charging phase the current is set at MinCurrent (when in Smart or Solar mode).

Could you flash the latest debug-enabled firmware, telnet to your smartevse, switch to verbose and post the logs of that initial charging phase, so we can see what is happening?

kletteroli, 2023-02-24:
Please find the log below. (A short charging session in SMART mode with MAX at 16A and MAINS at 10A.)
If I add "IsetBalanced = MIN_CURRENT*10;" at line 621 my problem no longer exists (it's commented out before doing this log). But very likely this is not an "elegant" solution.

[puttySMART.log](https://github.com/serkri/SmartEVSE-3/files/10828758/puttySMART.log)

dingo35, 2023-02-26:
You are right, the secret is probably in setting IsetBalanced; but this is in the heart of the regulation algorythm, so I need to test multiple options myself on my EVSE; I'm travelling at the moment so it will take some weeks before I am at my SmartEVSE location again... lets leave this issue open as a reminder for me.

dingo35, 2023-04-09:
[master-20230409.zip](https://github.com/serkri/SmartEVSE-3/files/11185189/master-20230409.zip)

@kletteroli, could you test the above version in Smart mode?

stevoh6, 2023-04-09:
Can this help? 
https://github.com/kletteroli/SmartEVSE-3/commit/358b08b7d6a0c48b1d319fdcc5998e04cd99d6ed

kletteroli, 2023-04-09:
@dingo35 
I now have the car charging with the above version (I did not flash the Spiffs though).
Current ramp up and down (also ramp up after plugging in the car) in Smart mode seems to work nicely.
Same applies for Solar mode. I will keep an eye on it. Thank you

dingo35, 2023-04-09:
You need to reflash spiffs.bin too, data/index.html has changed with regards to override_current.

kletteroli, 2023-04-10:
new spiffs is running now...
in this log, with current-override set to "no override" I first started in Solar mode, then toggled to Smart mode and observe that the mains setting of 10A gets exceeded. I then toggle back to solar mode where the regulation luckily cuts in again. 
[putty1.log](https://github.com/serkri/SmartEVSE-3/files/11189484/putty1.log)

dingo35, 2023-04-10:
Ok can you post your /settings so I can try to reproduce?

kletteroli, 2023-04-10:
Start: -0A
Import:8A
Mains:10A
Min:6A
Circuito: 23A
Max:23A
Sensorbox
4Wire
EvMeter:Disabled

single phase with a Nissan NV200 (Leaf)

I always have a home battery value present, or 20 or -20 (2A or -2A)

dingo35, 2023-04-10:
@kletteroli :" then toggled to Smart mode and observe that the mains setting of 10A gets exceeded."

When I set my Mains at 10A, my Max at 16A, my EVSE starts charging at 6A and goes up to 13A; at which point my mains usage is 10A (because my solar panels are feeding 3A into my power panel).

Are you sure that when you set your Mains at 10A, that your sensorbox reports an exceeding of your MAINS current(s)?

kletteroli, 2023-04-10:
Interesting: I toggled again to Smart Mode and CAN confirm that the import exceeded the 10A again (checked on the LCD).
Then, for this log that I send you, I unplugged the car, started the log, plugged the car back in being on Solar mode then I toggled twice to Smart but to my surprise NOW mains current was not exceeded anymore.

{"version":"12:04:07 @Apr  9 2023","mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"FRITZ!Box 6820 WR","rssi":-76,"bssid":"38:10:D5:8D:A0:E1","auto_connect":false,"auto_reconnect":true},"evse":{"temp":35,"temp_max":65,"connected":true,"access":true,"mode":2,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":211,"override_current":0,"current_min":6,"current_max":23,"current_main":10,"solar_max_import":8,"solar_start_current":0,"solar_stop_time":10,"enable_C2":"Always Off","mains_meter":"Sensorbox"},"home_battery":{"current":20,"last_update":1681124231},"ev_meter":{"description":"Disabled","address":12,"imp
[putty1.log](https://github.com/serkri/SmartEVSE-3/files/11190111/putty1.log)
ort_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":80,"L1":92,"L2":-6,"L3":-6,"last_data_update":1681124266,"charging_L1":false,"charging_L2":false,"charging_L3":false,"original_data":{"TOTAL":98,"L1":98,"L2":0,"L3":0}},"backlight":{}}

dingo35, 2023-04-10:
Ok the challenge will be to find out how to reproduce this.... You could test with variations like: 
-disconnect the car, put charge in smart mode, then connect the car. 
-disconnect the car, put charge in solar mode, then connect the car, then put it in smart mode
-disconnect the car, put charge in off mode, then connect the car, then put it in smart mode ...

kletteroli, 2023-04-11:
Indeed, it´s a challenge.

yesterday I tried to reproduce the behaviour but no luck, I then left the car plugged in over night, but charging was set to OFF. 
In the log "off but plugged in over night" I start from that condition and start in Solar mode and after a short moment transit to  Smart. In that log we see that Imains of 10A is exceeded.
- In the log I can not see at which point transition in the charging mode (Solar-->Smart) is happening? Can we include those transitions in the log?
When I toggle back to Solar the regulation always becomes stable again.
I attach two more logs, both with the car plugged in and OFF then I start in Solar on toggle to Smart (in the menu I pass from Solar via Normal to Smart). For my tests I never operate from the web interface.
[off but plugged in over night.log](https://github.com/serkri/SmartEVSE-3/files/11198425/off.but.plugged.in.over.night.log)
[Line 375.log](https://github.com/serkri/SmartEVSE-3/files/11198426/Line.375.log)

[Line 537 539.log](https://github.com/serkri/SmartEVSE-3/files/11198428/Line.537.539.log)

dingo35, 2023-04-13:
[debug-messages.zip](https://github.com/serkri/SmartEVSE-3/files/11222244/debug-messages.zip)
Added some debug messages, perhaps we can see what is going on; I still cannot reproduce this.

By the way, you said "I always have a home battery value present, or 20 or -20 (2A or -2A)". Is it possible by taking that out of the equation by disabling it temporarily? It messes up the numbers, and if we notice the problem is gone we have located the problem...

kletteroli, 2023-04-13:
You are not asking for too much!! I am very glad about your commitment and here to help where I can.
I disabled the home battery part and now run the new version. Let´s keep fingers crossed...

kletteroli, 2023-04-14:
Although the log I attach shows a correct behaviour of the regulation part it shows something else: (I don´t think this is the complete explication of what we have seen before). When I toggle the settings (LCD) from Solar to Smart (using the "right hand" button) I briefly pass through "Normal". This is reflected in the log and sets IsetBalanced to 23A which then gets regulated by the following transition to Smart. 
Maybe the desired behaviour would be to "activate" the settings only once that submenu is left with the "central" button or by "exit" of the config menu.
I keep on testing and will avoid passing through "Normal"...
[Line433.log](https://github.com/serkri/SmartEVSE-3/files/11230094/Line433.log)

kletteroli, 2023-04-16:
Good morning, 
I think I have caught a nice one, hope it will help!
Car was plugged in the night before but EVSE in "OFF". This morning the log starts with Solar Mode but with not enough solar available to trigger the charging so on the LCD I switch over to Smart mode. It starts low but keeps rising and the 10A mains gets exceeded (around line 370). I then switch back to Solar and notice that the regulation to 8A import is working correctly. Switching to Smart again results in L1 >10A for a second time.
[putty1.log](https://github.com/serkri/SmartEVSE-3/files/11241524/putty1.log)

dingo35, 2023-04-16:
That is indeed a good one! 
[fix-maxmains.zip](https://github.com/serkri/SmartEVSE-3/files/11241867/fix-maxmains.zip)

It is clear that the problem lies in Imeasured, which should always equal the max of the three phases L1 - L3, and at a certain point it stays at 0.0 .
I found a line in the code that was always suspicious to me, I never understood why it was there but it stems from the original firmware. I'm fairly certain that was the problem, I hope you can confirm it by testing....

This has some new stuff in it, if your test is ok this is going to be RC0 for release 1.6.0, so you will have to reflash spiffs.bin .

Have fun!

kletteroli, 2023-04-17:
I repeated yesterdays conditions.
This time the offgrid inverter couldn´t cope with the >> Imains. So the log becomes interesting at its end...
[putty1.log](https://github.com/serkri/SmartEVSE-3/files/11247635/putty1.log)

dingo35, 2023-04-17:
Ok number of problems here:
1. Log is garbled:
[code](D) (Timer1S)(C0) Detecting_Charging_Phases=1.
(D) (Timer1S)(C0) Detecting_Charging_Phases=0.
(D) (Timer1S)(C1) Trying to detect Charging Phases E(Timer1S)(C1) Error in detecting phases: EnableC2=Always Off and Nr_Of_Phases_Charging=0.
(Timer1S)(C1) Setting Nr_Of_Phases_Charging to 1.[/code]

So you can see in the third line the line is truncated and the (Timer1S) is supposed to start on a new line but it isn't. Also I can see in the code that some important log lines in the phase detection routine are missing.
Assuming you didn't mess up when copying/pasting the log, I think this is a timing problem; I think I solved it by moving a delay to a more proper place .

2 Phases are incorrectly detected. This is probably due to 1.

3 The fallback routine corrects the nr of phases, but fails to set the proper charging phase to true so Imeasure is never updated. I fixed this.

Also I took a more conservative approach in detecting the phases, now all phases are assumed to be charging phases unless probability shows they are not charging. It used to be the other way around...

In my setup it seems to be working ok, I don't have sensorbox but Eastron kWh so I think this might make the difference in our test setups.

Again: have fun!

[master-83ecdeb3c9.zip](https://github.com/serkri/SmartEVSE-3/files/11248639/master-83ecdeb3c9.zip)

kletteroli, 2023-04-19:
I have been testing for two days now and will continue to do so.But I am glad to tell you, that so far it seems stable!! Good fix! And thank you!!

For the logs I use the putty log setting and therefore avoid any copy/paste...

One comment to normal mode. I personally find it problematic that Imax is used as default setting for the charge current because easily  Imains then gets exceeded if not enough solar contribution is present. But as you say, this is very much a personal preference, others might not agree.

I will test a bit more and keep you updated in 2 days.

dingo35, 2023-04-19:
Great to hear that! 

As to MaxCurrent the default for Normal mode, IMHO if you need to worry about MaxMains you should not use Normal mode... but if you still do, it makes sense to reduce MaxCurrent to a safe level.

But using Normal mode is not Smart (pun intended)...

kletteroli, 2023-04-23:
I did some more testing and this looks stable now. Thanks a lot for the fix and all the work you are investing in this project.

kletteroli, 2023-04-26:
**HomeBattery="0"**
{"version":"11:42:02 @Apr 17 2023","mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"FRITZ!Box 6820 WR","rssi":-70,"bssid":"38:10:D5:8D:A0:E1","auto_connect":false,"auto_reconnect":true},"evse":{"temp":36,"temp_max":65,"connected":true,"access":true,"mode":2,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":172,"override_current":0,"current_min":6,"current_max":23,"current_main":10,"solar_max_import":8,"solar_start_current":0,"solar_stop_time":10,"enable_C2":"Always Off","mains_meter":"Sensorbox"},"home_battery":{"current":0,"last_update":1682507440},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":
**{"TOTAL":80,"L1":80,"L2":0,"L3":0,"**
last_data_update":1682507446,"charging_L1":true,"charging_L2":false,"charging_L3":false,"original_data":{"TOTAL":80,"L1":80,"L2":0,"L3":0}},"backlight":{}}

**HomeBattery="40"**
{"version":"11:42:02 @Apr 17 2023","mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"FRITZ!Box 6820 WR","rssi":-72,"bssid":"38:10:D5:8D:A0:E1","auto_connect":false,"auto_reconnect":true},"evse":{"temp":37,"temp_max":65,"connected":true,"access":true,"mode":2,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":203,"override_current":0,"current_min":6,"current_max":23,"current_main":10,"solar_max_import":8,"solar_start_current":0,"solar_stop_time":10,"enable_C2":"Always Off","mains_meter":"Sensorbox"},"home_battery":{"current":40,"last_update":1682507737},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":
**{"TOTAL":70,"L1":96,"L2":-13,"L3":-13,"**
last_data_update":1682507738,"charging_L1":true,"charging_L2":false,"charging_L3":false,"original_data":{"TOTAL":109,"L1":109,"L2":0,"L3":0}},"backlight":{}}

[cycling Isetbalancedup down (HB=40).log](https://github.com/serkri/SmartEVSE-3/files/11332155/cycling.Isetbalancedup.down.HB.40.log)

Hello again,
I reopen the issue because I just "reactivated" my home battery contribution which I had disabled for providing better testing conditions (in this log HomeBatteryCurrent= 40). In the log  you can notice that "IsetBalanced" is regulating down to start the ramp again from MIN to increase until for some reason it thinks mains is exceeded (when it is not (because mains excludes the solar contrib and should not be affected by what is happening to the HomeBattery)).
In the copy/paste above I find interesting what I have marked in bold.  Maybe this can help.

dingo35, 2023-04-26:
" it thinks mains is exceeded (when it is not (because mains excludes the solar contrib and should not be affected by what is happening to the HomeBattery))."

I don't understand what your saying there; when I look at the logs I assume you still have MaxMains = 10.0 A, and you see that it reaches 10.2 A so Maxmains is exceeded, it says "No current" and it immediately reduces the charging to MinCurrent= 6.0 A.
Then it sees there is room and starts regulating the power up....   this is still the old behaviour as designed in the original firmware.
So this is a very conservative approach; I could imagine, since we only superseed MaxMains by 0.2A, that we regulate Isetbalance down with only 1.0A instead of dropping it to MinCurrent; not sure how this would work out in a multi-EVSE setup....

But I dont see any relationship with homebattery or solar current....

Is that  would you mean?

kletteroli, 2023-04-26:
You are right, I got somehow confused. For testing and for first time I set the HomeBatteryCurrent to 40 and this conditioned what we see in the log. Before I always had it running at maximum HomeBatteryCurrent 20 and therefore Import 8A + HB 2A would normally just not trigger the "NO Current" and the following hysteresis. 
And yes, it might be interesting considering your suggestion "...superseed MaxMains by 0.2A, that we regulate Isetbalance down with only 1.0A..."

Responding your "EDIT": depends on the point of view. My mains setting represents what I am comfortable with having as a continuos "base load" for my battery inverter, especially when in SMART mode.  I sometimes need to use SMART to force charging a few % into the car when there is not enough solar contribution. Then there is a second situation when I need to pass through SMART before I can continue with SOLAR. It is when it´s sunny and the batteries are full. Then the PV inverters are in idle mode as the battery inverter regulates the grid frequency up (P(f)). It is when the EVSE status is "waiting for solar" although the sun is there. I then need to get the home battery voltage down so that f decreases. Doing a bit of "SMART" does this and make the PV inverters contribute.
In most cases I can avoid this sequence (that I still do manually) because of the work around with "HomeBattery_Current" set to 20. This setting is higher than the typical base load of our house when we are away and the batteries are sitting full. So I am really glad about the possibility you offer us to tweak on that end. I do this with a simple node red logic.
My setup might sound very individual but...there are simply not too many off gridders having an EV.

But I noticed something else, I hope my observation might help improving. Have a look at below image. 
Home Battery Current is set to 2A. Phase details L2/L3 show both -0.6 A. This is confusing as I am charging in single phase.
If I increase Home Battery Current by 1A I see an additional -0.3 A distributed on each phase in "Phase Details".
Maybe I don´t get the logic behind this. And I also wonder what "Phase details (original)" is telling us?  

[image]

dingo35, 2023-04-27:
Very interesting to learn how you are using your SmartEVSE in your situation! It seems to me there must be a better solution for you to automatically start the PV-inverters when your home battery is full and your EV battery isn't, but thats OT.

The battery_current you are providing is meant to be a "total current", so the total of three phases. What the code does is divide the battery_current by 3 (the nr of phases in the original code), and subtract that from L1 - L3. The original values of L1-L3 are stored in the "Original" phase details, I think the author did this for debugging purposes.

I did not change this part of the code when introducing the single-phase detection logic, since I have no Home Battery to test. It would be quite simple to change this, but I am not sure if it would be better to apply the battery_current only to L1 if we detect single phase charging.... 
What do you think?

As to the abrupt dropping to MinCurrent when in Solar Mode the MaxMains or MaxCircuit are exceeded, I rearranged the code a bit so it is more clear what is happening, and I changed this particular Solar mode / overload behaviour so that it now should drop the charging current to the value that the overload disappears (just like in Smart mode).
I have my EV not here so I cannot test, but if you would like to help, here it is:
[kletteroli.zip](https://github.com/serkri/SmartEVSE-3/files/11340457/kletteroli.zip)

dingo35, 2023-04-28:
When I look at the log, I only see a short overrun of Mains from lines 1040 thru 1089, exactly the moment when you changed HB. So I agree that MaxMains is sacred, and it looks like it works well; of course the regulation mechanism takes time: after measuring an overrun (which happens approx. every second), the CP line is throttled down, and then the car has to respond by lowering that charging current. 

This is fast enough, when you look at the characteristics of Mains breaks small overruns are allowed for long times (>10 minutes), and bigger overruns still for 10-60 seconds.

I added HomeBattery current to the log, as you requested.

I also rethought changing the HomeBatterycurrent to single phase, but there are several problems with that; it would no longer be backwards compatible, also for three phase battery systems this would be disfunctional. 

Now if I understand correctly, for testing you are feeding the SmartEVSE with HB current-information without actually feeding that battery current into your system; that might introduce some problems, I will look into the code whether this is a good testing situation....

kletteroli, 2023-04-28:
The regulation will settle here too high (above mains):
[image]
its not an overshot of the regulation mechanism. (The EVSE+car I notice to be pretty agil.)
And yes for "normal characteristics" of a MAINS circuit breaker. Those that we depend on battery inverters, we have a less forgiving curve, in a way it´s faster.

_"Now if I understand correctly, for testing you are feeding the SmartEVSE with HB current-information without actually feeding that battery current into your system;..."_  Yes, correct, I pretend.

dingo35, 2023-04-28:
I have little time now to look into the code, but am I correct that the overshoot is approx. the same as the HB current you are (not) feeding?

kletteroli, 2023-04-28:
For this session the house consumption was kept constant (except at the end 22:16h when I plug in the laptop). It´s dark outside.
The EVSE settings is still the same with MAINS at 10A and Import at 8A.  Every minute a sample is taken. I keep rising the HB in steps of 1A starting with +2A, the last step is from +5A to +7A. For each step I recorded minimum 3 data points.
[image]
In this graph you can see how EVSE current (from your API) nicely correlates with what the battery inverter measures. And that by changing the battery current setting (all fake) I manage to exceed the 10A MAINS.
Passing the 10A, for every amp that I rise HB I seem to get a +0.3A /0.4A increase in EV charge current.
Below the corresponding screenshots for each step. They were taken once regulation has stabilized.
  
[HB ramp.pdf](https://github.com/serkri/SmartEVSE-3/files/11357290/HB.ramp.pdf)
and a short log with HB + 7A and the corresponding screenshot
[putty1.log](https://github.com/serkri/SmartEVSE-3/files/11357370/putty1.log)
[image]

dingo35, 2023-04-29:
But this is exactly what HB is supposed to do: you offset the Original L1 by 7/3 = 2.3 A so the battery regulating system can do whatever it wants to do.

You test setup is just simply incomplete: if you would _actually_ have your Battery feed 7A into your home grid, like you are telling SmartEVSE you do, your L1 original would become 5.2A and the L1 phase detail (which is an artefact, it doesnt correspond to anything real) would become 3.0A.

Your first post:
"The background is: My off grid setup is a battery inverter in combination with a grid inverter. When the battery is full, the grid inverter is in idle mode and the regulation works rather slow. When this is the case the maximum current is limited to what only the battery inverter can provide whereas when the grid inverter is contributing the maximum current that the system can provide is higher." 

Suppose your battery inverter has a max of 10A, and your grid inverter has a max of 25A. Then I would make a routine in HomeAssistant or whatever tool you are using, so that when your grid inverter is active you feed HB=150 (25A-10A); that way your MaxMains will be artificially increased to 25A when you are on grid, and be 10A when you are off grid...

kletteroli, 2023-04-29:
Ok, I understand. It is correct what you say. If the power "actually" was there it would not be possible to "artificially" increase MAINS the way I do.
Anyway and as I already mentioned, my setup and your SmartEVSE fork - for my particular situation - works great. My intention was simply to contribute to this project by testing. So please let me know if in a future you want me to test any changes especially for single phase...

Maybe just to add for your curiosity. My setup is a typical 100% Offgrid topology, there is no other grid than the one I setup with the battery inverter. It's a often used "AC-coupling" where there is Solar AC contribution via a Grid inverter and also Solar DC contribution via Solar Regulators (it´s about 50/50 where DC is more efficient for charging the batteries and the AC more efficient for directly powering daytime consumers). It is the DC Solar part where it is especially interesting to make use of the SmartEVSE HB settings. It is on that end where I can happily decide about the proportion of current that I want to have for the home batteries or for the battery inverter -->EVSE-->car.
So thanks again for your great contribution and from my side its OK to close this issue.

dingo35, 2023-04-29:
Ok thanks for your efforts!

closed 2023-04-29 by dingo35