serkri/SmartEVSE-3 #109: EV meter connection

camel1cz, 2023-03-04:
Hello,

I started to use SmartEVSE3 with original FW and quickly switched to serkri fork with great results. Basic functions work fine same as HA integration. Thank you!

My question (rather ask for help/explanation) is about connection of EV meter. I have dedicated EV meter in the junction box of the SmartEVSE which can provide data via modbus/RS485.

The question is if it is possible to connect the modbus EV meter to SmartEVSE and read at least energy consumed (kWh) to the car?
And if yes, then how to do it? How to wire the A/B and what to set in the SmartEVSE.

It would be nice to avoid fiddling with ESP just because of this... I saw a lot of notes about possibility to connect EV meter, but there is also mentioned smartbox, smartbox2, no need of smartbox, P1 connector, A/B/GND on the low power connector - sorry buddies, I'm confused and lost.

Could someone help me? Thanks a lot!

dingo35, 2023-03-04:
The connector for modbus RS485 is described in the manual: https://github.com/SmartEVSE/SmartEVSE-3/blob/main/manual/SmartEVSEv3_install_v3.0.pdf

You should wire your EVmeter just like you would wire a sensorboxin the manual, so the lower lwft connectioms connect A, B and ground to the respective connections of your EVmeter.
When you select Smart or Solar mode you can configure a MainsMeter and an EVMeter on the LCD screen.
It is unusual to connect an EVMeter without installing a Mainsmeter.

camel1cz, 2023-03-04:
Thank you! I'll give it a try hopefully tomorrow...

> It is unusual to connect an EVMeter without installing a Mainsmeter.

Hmm, I want to use the SmartEVSE like this:
- be able to control (start/stop) charging from HA,
- be able to change the current dynamically from HA,
- as bonus: know the power consumption and propagate it to HA (thus the EV meter question).

Don't know if it's possible nor anyone using SmartEVSE like this.

dingo35, 2023-03-05:
Yes, only in Normal mode, and Yes. You would run in Normal mode anyways if you have no Mainsmeter installed, just put in in Smart mode to configure your EVmeter and back to Normal mode after that.

evanes68, 2023-03-08:
I am also trying to connect an Eastron Modbus sdm630 but have not been successful. I use the mains meter API so the EV meter is the only Modbus device I am using. I have only connected a and b for Modbus, is the ground necessary?

The mains meter values work perfectly but I cannot get the EV meter readings.

Any thoughts?

royspeed, 2023-03-08:
I too would like to be able to start, stop charging and set the charging current from HA.

My connection is only 1x35A so I use the smart mode with a sensorbox. 
However, it is then no longer possible to set the charging current. 

When I charge at 32A there is a voltage drop of 10V in my house and despite the fact that I have connected everything with 6mm2 cables, quite a bit of heat is noticeable. To limit losses I would like to charge only with 25 or 16 A. However, sometimes it is urgent to load the 32A.

My Eastron SDM120CT-MV still worked well with the original firmwere. With release v1.5.3 only the EV Meter currents are visible, Power, total and charged kwh remain at 0.

Thank you for this project, it has been a very educational process for me

fluppie, 2023-03-08:
SDM120 and SDM230 use other addresses. We should have 2 definitions, Eastron1P and Eastron3P

See https://github.com/SmartEVSE/SmartEVSE-2/issues/4 and https://github.com/reaper7/SDM_Energy_Meter/blob/8bda969f312cfafa5a26506da8ac9b42d4fb38ca/SDM.h#L95

dingo35, 2023-03-09:
You could try to configure an EM_CUSTOM meter; I am not sure what the implications on the total algorithm are, and since I cannot test it in real life Im not going to touch this...

dingo35, 2023-03-10:
Call them. E.g. Liander has a goal to get away from 1 phase connections. They moved me and my 3 neighbours from 1x40A to 3x25A for free... but that was 10 years ago.

dingo35, 2023-03-10:
@fluppie @royspeed 
Ok I was tempted to make a test version anyways; remember this is UNTESTED, so I need you guys to test this thoroughly, by looking at the values presented at the http://your-ip/settings page:
1. check whether the current measurement on L1 is ok, and if L2 and L3 are 0A.
2. check whether the power measurement is ok
3. check whether the import/export energy readings are ok

Have fun!
[UNTESTED-master-20230310+eastron1p.zip](https://github.com/serkri/SmartEVSE-3/files/10945531/UNTESTED-master-20230310%2Beastron1p.zip)

fluppie, 2023-03-11:
I have uploaded it to my SmartEVSE, only the sign is wrong. Current is - at the moment because of solar.
[image]

While charging:
[image]

So power and charged kWh work now. What did you exactly change? Normally changing this: https://github.com/serkri/SmartEVSE-3/blob/7f8410b541434f854651db74237ad0cdadfbe013/SmartEVSE-3/src/evse.cpp#L225 is enough.
The code ignores if L2 and L3 are not received by the modbus read stuff. So normally you should not change that part.

dingo35, 2023-03-11:
@fluppie , I pushed my changes to branch "UNTESTED-eastron-1p" so you can look at the changes I made.

Just to be 100% clear:
1. What is the exact Eastron model you are testing with again?
2. Is it wired "normal" (feeding wires at the top, consumption wires at the bottom) or "inverted"?
3. So your expected value at the first screen shot was L1=-7.0A, and the displayed value was L1=7.0A ?
4. So your expected value at the second screen shot was L1=9.3A, and the displayed value was correct?

fluppie, 2023-03-11:
SDM230
Yes
Yes
Yes :-)

I think you can keep the same code for Eastron3P as for Eastron1P, maybe completing the loop for some reason gets the minus there?

[image]

dingo35, 2023-03-11:
[UNTESTED-master-20230310+eastron1p+testing-current-fix.zip](https://github.com/serkri/SmartEVSE-3/files/10948674/UNTESTED-master-20230310%2Beastron1p%2Btesting-current-fix.zip)

It wouldn't make sense to me, but test this version... I pushed the (possible) fix to the branch so you can keep track of the changes.

fluppie, 2023-03-11:
Hilarious but it works, I also don't really get why.
I'll try charging but battery is 100% atm.

[image]

fluppie, 2023-03-11:
[image]

Triggering the heating can trigger charging :)

dingo35, 2023-03-11:
Ok so currents, power, and energy are all ok now?

fluppie, 2023-03-11:
Yes indeed.
[image]

dingo35, 2023-03-11:
Ok thx I will merge it into master in a few weeks....

royspeed, 2023-03-11:
for my SDM120CT-MV there is no change. Still no power data.
[image]
[image]

fluppie, 2023-03-11:
You have to go into the menu and change EV Meter Eastron3P to the Eastron1P ;-)

royspeed, 2023-03-11:
> Call them. E.g. Liander has a goal to get away from 1 phase connections. They moved me and my 3 neighbours from 1x40A to 3x25A for free... but that was 10 years ago. And they had the same rates on the website then....
> 
> EDIT: I remember this had to do with the fact that me and my 3xneigbours shared the same cable to the network. My house was connected to the grid, the other houses were connected to our house (we all had our own meter). This situation was considered unwanted or even illegal...

Long phone calls with Liander. According to the employee, an upgrade from 1x35 to 1x40A is free. 
So request this one. Have to wait 3 months. 

Maybe I can make a cheaper deal with the mechanic, everything is already connected only 2 extra fuses need to be installed. Strange that they charge 340 euros for that.

royspeed, 2023-03-12:
> @fluppie

fantastic! It seems to be working fine now.
Stupid of me, the smd120m meter was indeed connected incorrectly (L in- L out).
[image]

dingo35, 2023-03-14:
@fluppie @royspeed Guys I need you to test again; I find it a problem that every time I add a new meter configuration, some people will see their configuration messed up when they upgrade to a new firmware version; e.g. your eastron1p would now be the default for everybody who had an EM_CUSTOM meter configured in the past; they would only notice it because of errors and then have to find out themselves they have to reconfigure their mainsmeter and/or evmeter and/or pvmeter.

To prevent this I added some empty slots in the firmware; I also had to add some logic to prevent people choosing those empty slots.

So my question to you guys is, would you please test the attached firmware, ESPECIALLY on your LCD screen, when you step through the possible options for your meter type, that _after_ the option eastron1p NO unused slots are shown but ONLY the EM_CUSTOM option can be selected?
Please test this for MainsMeter, EVMeter and PVMeter.

I didn't change anything in the 1P logic but you might want to check your outputs anyways...

Thanks in advance!
[eastron1p+empty-slots.zip](https://github.com/serkri/SmartEVSE-3/files/10968948/eastron1p%2Bempty-slots.zip)

royspeed, 2023-03-16:
@dingo35, I have on the LCD screen:
 MAINSMET: Sensorbox,  Phoenix C, Finder, Eastron3P, InvEastrn, ABB, SolarEdge, Wago, API, Eastron1P, Custum, Sensorbox.  
EV METER: Eastron1P, Custom, Disabled, Phoenix C, Finder, Eastron3P, InvEastrn, ABB, SolarEdge, Wago, Eastron1P.
I did not have a PVMeter in my setup. I have artificial thatch on my roof. I haven't been able to find an installer yet.
All data looks correct.
[image]

fluppie, 2023-03-19:
I confirm the same order as @royspeed

dingo35, 2023-03-20:
@royspeed @fluppie thank you guys!

dingo35, 2023-04-05:
Merged into release 1.5.5

closed 2023-04-05 by dingo35