serkri/SmartEVSE-3 #110: Feature request - Home Assitant examples

HA-TB303, 2023-03-10:
Hi, congrats on the fantastic changes you made to the firmware. 

I am having some difficulty finding out how this API power meter data feeding stuff works. Could you please share your home assistant configuration and automations regarding the SmartEVSE?

Big thanks!

Michel

dingo35, 2023-03-10:
This will do the job for you:
https://github.com/dingo35/ha-SmartEVSEv3

HA-TB303, 2023-03-11:
Thanks, that helps.

Anybody posting the L1, L2 and L3 data straight from esphome DSMR?

HA-TB303, 2023-03-12:
I could use just a little help since I am not that savvy with json and lambda

esphome yaml:

```
on_..
    then:
      - http_request.post:
          url: http://192.168.207.121/currents
          verify_ssl: false
          json: |-
            root["L1"] = id(current_l1).state;
            root["L2"] = id(current_l2).state;
            root["L3"] = id(current_l3).state;
```
log:
```
[http_request:099]: HTTP Request completed; URL: http://192.168.207.121/currents; Code: 200; Duration: 32 ms
```
but does not work....

this on bash works:

```
### post into the API ###
curl -X 'POST' \
  'http://192.168.207.121/currents?L1=60&L2=70&L3=80' \
  -H 'accept: application/json' \
  -H 'Content-Type: application/json' \
  -d '{}'
```

HA-TB303, 2023-03-12:
Got a little further now, but I am still unable to feed the sensor data to the API:

```
      - http_request.post: 
          url: !lambda |-
              std::string url;
              url.append("http://192.168.207.121");
              url.append("/currents");
              url.append("?");
              url.append("L1=" + id(currentl1).state).c_str();
              url.append("&");
              url.append("L2=77");
              url.append("&");
              url.append("L3=88");
              return url.c_str();
```

```
Compiling /data/slimmelezer/.pioenvs/slimmelezer/src/main.cpp.o
/config/esphome/slimmelezer.yaml: In lambda function:
/config/esphome/slimmelezer.yaml:189:24: error: invalid operands of types 'const char [4]' and 'float' to binary 'operator+'
  189 |               url.append("L1=" + id(currentl1).state).c_str();
      |                  ~~~~~ ^ ~~~~~~~~~~~~~~~~
      |                  |                  |
      |                  const char [4]     float
*** [/data/slimmelezer/.pioenvs/slimmelezer/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.17 seconds ==========================
```

HA-TB303, 2023-03-13:
Solved.

closed 2023-03-13 by HA-TB303

stevoh6, 2023-03-24:
> 

Please share how :-)

dingo35, 2023-03-24:
His solution is uploaded in the integrations directory of the serkri distribution:
https://github.com/serkri/SmartEVSE-3/blob/master/integrations/esphome/ET-SM01%20smart%20meter%20module.yaml