serkri/SmartEVSE-3 #120: Internal clock not shown in web interface when Normal mode

negronoir, 2023-03-27:
I have two SmartEVSE installed together for two cars. One of them, is set in 'Smart mode' with sensorbox. The other is in 'Normal mode' with fixed current set.
Both show right updated time on SmartEVSE screen
Also, I have installed Home Assistant integration
The issue:

-The 'Smart mode' one:
 -In web interface shows date and time updated with local time in my country (UTC + 2) e.g. 22:05:08
 -In HA time is also shown but UTC 20:05:08

-The 'Normal mode' one
 -In web interface doesn't show any data
 -In HA date is stuck 1970-01-01 00:00:00 and never changes

As I have two SmartEVSE units, I've swapped them: Normal -> Smart and Smart -> Normal. Also sensorbox has been disconnected from old Smart unit and reconnected to the new Smart one. 

And the behaviour is the same: the former Smart one which shown date & time, in Normal one doesn't do it. And the former Normal one which didn't show anything, now in Smart mode, shows date and time in a right way.

So the issue clearly is related with 'Normal' mode

dingo35, 2023-03-27:
Which firmware version are both SmartEVSEs running?

negronoir, 2023-03-27:
SERKRI-1.5.3

But I'm sure that an older version which I installed about November, had the same issue.

I'm pretty sure that the issue is related with the absence of Sensorbox: I have just set both SmartEVSE in Normal mode, then reboot, both show on screen current time. But in web interface, only the one that has SensorBox attached shows current date & time. The other doesn't show any data

dingo35, 2023-03-27:
Ok Im going to test if I can reproduce this.

dingo35, 2023-04-05:
Could you test release 1.5.5, and on the EVSE that has no Sensorbox connected, you can now configure MainsMeter to Disabled.

Does that change anything?

negronoir, 2023-04-06:
No. Same result. I attach screenshots of 192.168.1.8 (Mercedes, normal moed), 192.168.1.9 (Tesla, smart mode) and data received by Home Assistant integration

[image]

[image]

[image]

BTW Tesla screen shows "Charge 3x32A". Installation is only one phase but I suppose that Sensorbox Plus is sending raw data from the three theoretical phases: the one with the current sensor and the others with no sensor attached. There's no way to know how many phases has the grid

Thanks!

dingo35, 2023-04-07:
My guess is that your Master device is displaying the time correctly (the UTC display in HA is probably a configuration error in HomeAssistant; did you set your location and/or time settings correct?).

The Slave device (node 1 or your second EVSE) is not supposed to display currents, because those are measured by the master device. So it makes sense that the currents on the webserver of the slave device are never updated, hence the last updated time is epoch 0 which is Jan 1st 1970 .

negronoir, 2023-04-10:
Leaving aside the matter of HA time offset with is out of scope of this thread:

The reality is that my two SmartEVSEs are not configured as master and slave.
Despite being together in the same box, they are isolated. No bus connection between them.
What I've checked some times is: if I connect the sensorbox to the SmartEVSE_1, this unit begins to show date & time at web interface and the SmartEVSE_2 without sensorbox doesn't show anything.
If I disconnect sensorbox from  SmartEVSE_1 and connect it to SmartEVSE_2, the SmartEVSE_2 begins to show date & time at web interface and the SmartEVSE_1 without sensorbox stops to show anything :-)

So the check should be easy:
-SmartEVSE isolated. No sensorbox
-Configured in NORMAL mode
-Serkri 1.5.1 /1.5.3 / 1.5.5 firmware

Time and date are not shown in web interface

[image]

-Power off SmartEVSE
-Connect Sensorbox (mine is "Sensorbox Plus")
-Power on SmartEVSE

Time and date now are shown in web interface

[image]

dingo35, 2023-04-10:
Ok you need to understand that the date & time are NOT the actual date & time of the SmartEVSE, but they are the "last updated" date & time of the current data that is delivered by the MainsMeter.
So it makes sense that if no Sensorbox is connected, that data is never updated, and date/time last updated is not either.

BUT I understand that the display is confusing, so I reworked the html code so now the currents + date/time are only displayed when a MainsMeter is configured.

ALSO I noticed that the webserver, when in OFF mode, and when no MainsMeter and no EVMeter configured, became unresponsive. This should be solved in the latest version too.

Could you please test this version:
[master-20230410-dc1dc320687.zip](https://github.com/serkri/SmartEVSE-3/files/11190012/master-20230410-dc1dc320687.zip)

negronoir, 2023-04-10:
Ok. So no issue :-)

As on both SMmartEVSEs time is right shown on screen, I believed that date & time on Web server was obtained and updated from the same source than screen data.
But, obviously, if data is refreshed from mains meter, makes sense the behaviour.

I attach webserver screenshots. Note that Chrome formatting is not good. MS Edge is ok.

NORMAL Before
[image]

NORMAL After
[image]

SMART Before
[image]

SMART After Google Chrome
[image]

SMART After Miscrosoft Edge
[image]

Many thanks for all your work :-)

closed 2023-04-10 by negronoir