serkri/SmartEVSE-3 #121: Mainmeter address

deekoowee, 2023-03-28:
Hi,
I have mainmeter supplied by inverter manufacturer.
Meter seems to be Eastron OEM. (like SDM630 and the registers are very similar)
Problem is, that inverter have hardcoded mainmeter modbus address 01 and that cannot be changed by user.
Smart EVSE V3 firmware SERKRI-1.5.3 doesn't have mainmeter adresses below 10.
How I can configure address 01 for mainmeter in EVSE ?

Because of two separate modbuses, my goal is to connect these together with EE11's connected to mbusd.

dingo35, 2023-04-05:
Ok my answer through the github mobile app obviously got lost, so here my answer again:

Meter addresses 1 thru 8 are reserved for the SmartEVSE, 1 for the Master and 2-7 for the Nodes/Slaves. This is deeply anchored in the architecture of the SmartEVSE, so we are not changing this.
If you have only 1 SmartEVSE you can tweak the default master address easily to 1, but you will have your private compiled firmware version....

closed 2023-04-05 by dingo35

deekoowee, 2023-04-05:
Thanks for your answer
Now I underestand, why I got some trouble with address 1.
I have only 1 SmartEVSE and I can try private compiled version.
I found MAINS_METER_ADDRESS in evse.h . It is defaulted to 10 and that should changed to 1 ?
I couldn't find the Master device address in code.
If I now underestand right, it is defaulted somewhere to 1 and I should change it to 2 or something else
Where I can find it in code ?

dingo35, 2023-04-05:
Defaulting MAINS_METER_ADDRESS to 1 would do the trick. You should also change MIN_METER_ADDRESS to 1.

Changing the master device address would be the hard part since this is hardcoded, you can look into src/modbus.cpp, see which functions are calling them and work your way back, or modify the address in the modbus read/write functions.

I wouldnt waste my time on this and put a new SDM72 v2 in my power panel, the cost will be much lower than the hours you spend in the software.

deqw, 2023-04-05:
A few years ago it was advised for the original firmware to set the modbus address for the SDM630 to a number higher than 10, eg. 101.