serkri/SmartEVSE-3 #122: Slave EVSE not returning to normal mode after Solar mode was active

djoenez, 2023-03-30:
My skoda does not want to play nice with solar mode if there are alot of clouds (so the charging gets started and stopped a couple of times) So i just want to put it in normal mode but a low charging current to fix this during cloudy days.

unfortunately when i had both my chargers on solar, switch master to Off (only want to charge my slave) and switch slave to normal i  still have the status: Waiting for solar. and if reboot the EVSE, it will default to solar again.

dingo35, 2023-03-30:
Could you post the settings of both EVSEs?

dingo35, 2023-04-05:
What happens if you put both on Solar, switch Slave to Normal, and then switch Master to Off?

djoenez, 2023-04-07:
Hello Dingo, will do asap, unfortunately i will not be home with 2 cars for the next two weeks, will update ASAP (thanks for looking into this!).

Just a question for my understanding: Is it possible to use the EVSE2 (smart or solar) when master is in off mode? or is this prohibited? 

Maybe also a feature request: I ask these questions because sometimes i have two cars connected but i want to charge one car in particular of solar (and only start charging the second car when i have enough solar power, i have enough to charge 1x16A + 1x8A for example). But now i don't have this choice, is there a way to "toggle" the preferred car? Or when the car is connected to have a toggle which states you may charge (so you prevent it to charge even when the selected mode is solar?)

dingo35, 2023-04-07:
The "OFF"mode is actually setting the Access bit off; so this is the same as when you have RFID reader activated and you would not identify yourself with a valid RFID card.

So I would expect the original firmware to be able to charge on EVSE2 when EVSE1 is not used/no RFID card is presented; and I would expect the serkri firmware to do the same, since we did not (intentionally) change that behaviour.

But I only have 1 EVSE so I cannot test it for you....

I understand your feature request, makes a lot of sense; for the near future I would use HomeAssistant and the ha-smartevse integration to script this kind of behaviour. I intend to start using solar mode this summer (also through HomeAssistant) and when I have some practical experience I might either publish my HA automations OR build them into the SmartEVSE firmware....

dingo35, 2023-04-23:
@djoenez could you please test this with newest release 1.6.0 and, if problem still exists, post the verbose telnet/debug log here?

djoenez, 2023-04-23:
@dingo35, will do. Just got back from long work related travel, as soon as i use the second car again i will update this thread with testing. Thanks!

djoenez, 2023-04-28:
Works now as expected. Thanks for the fix!

closed 2023-04-28 by dingo35