serkri/SmartEVSE-3 #123: static ip doesn't stick after  reboot

kiss81, 2023-04-02:
When I set my static ip in the configuration window when I am connected to the AP I can access the webinterface to the static ip.
When I reboot  after a while or when I cut the power it switches back to my dhcp address.
Not sure if it matters, but the IP I set is 10.42.43.9

dingo35, 2023-04-02:
We are using a common used library that handles this; are you sure you are exiting the wifi-setup menu correctly so that all is saved?
Even if this is a bug, it should be solved in the arduino library, not in our firmware....

kiss81, 2023-04-03:
Pretty sure I save it correctly as it does work for a while until the next reboot. I understand this is out of scope for this project. I will have a look at the library!

dingo35, 2023-04-03:
Great!
Make sure you move (arrow) up to the menu-item after you have chosen the right menu-option; because we are missing an "enter" key, this action saves the selected value. Might be the problem, might be not...

dingo35, 2023-04-07:
"Save" means saved in the nonvolatile memory of the ESP32. It is perfectly possible that you enter the right value, but by not exiting correctly the value is not saved in nonvolatile memory.

Mind you, it is good practice to use dhcp so all your networking variables are configured automatically. This also enables the use of mdns.
The ip address that is given is shown on the lcd display in the upper status line in the setup menu. Since your SmartEVSE will be on always, the address should only change when you swap routers...

kiss81, 2023-04-07:
I will try it again doing it in the correct way!

Ok might change my network to work that way! mdns is really nice, but I found it doesn't always work, but maybe that's because I use a a lot of static IP's as well.

dingo35, 2023-04-07:
Install avahi-daemon on all your linux computers, and zeroconf (linux) and bonjour(apple) will discover all the services on all your devices on the network.
Once you got it running it works beautifully, and if you use domotica software (like HomeAssistant) that starts practically autoconfiguring itself!

dingo35, 2023-04-10:
[wifimanager1.15.1.zip](https://github.com/serkri/SmartEVSE-3/files/11190892/wifimanager1.15.1.zip)
Could you test this? Please flash both firmware.bin and spiffs.bin !

kiss81, 2023-04-10:
> [wifimanager1.15.1.zip](https://github.com/serkri/SmartEVSE-3/files/11190892/wifimanager1.15.1.zip) Could you test this? Please flash both firmware.bin and spiffs.bin !

Setting things up with the new wifi manager works really smooth! Feels more smooth compared to the older version.

Except I still have the same issue:

Steps I do:
1) I hold the middle button for two seconds
2) I navigate to Wifi
3) I, press middle button again
4) I press right button to select SetupWifi
5) I see the countdown for the AP getting up. When ready I connect to the AP SmartXxxxx with my pc
6) I get the WIfi configuration browser window
7) I fill in my WiFi settings + static IP
8) I press save
9) With my pc I reconnect to my normal WiFi
10) I can navigate to http://10.42.43.9 (my static ip I just setup)
11) after 5 minutes I turn off the power on my SmartEVSE v3
12) after a few seconds I turn on the power
13) My static IP is gone and the SmartEVSE v3 is connected as 10.42.43.224 (old dhcp setting)

Do I miss something? I don't really get how to save my static IP settings in another way? 
Thank you for your effort!

dingo35, 2023-04-11:
I can reproduce the error (that static ip is not saved), but it should be fixed in the library.... Too bad 1.15.0 still hasn't fixed it, it was fixed in an old version 1.4.......

closed 2023-04-11 by dingo35