serkri/SmartEVSE-3 #129: Charging current indications are wrong when mode for contactor2 is off

Ivesvdf, 2023-04-21:
Take the following scenario:

[image]

Raw data:

```
{
  "version": "SERKRI-1.5.5",
  "mode": "SOLAR",
  "mode_id": 2,
  "car_connected": true,
  "wifi": {
 censored
  },
  "evse": {
    "temp": 28,
    "temp_max": 65,
    "connected": true,
    "access": true,
    "mode": 2,
    "solar_stop_timer": 0,
    "state": "Charging",
    "state_id": 2,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 130,
    "override_current": 0,
    "current_min": 6,
    "current_max": 13,
    "current_main": 16,
    "solar_max_import": 5,
    "solar_start_current": 4,
    "solar_stop_time": 10,
    "enable_C2": "Always Off",
    "mains_meter": "API"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Disabled",
    "address": 12,
    "import_active_energy": 0,
    "total_kwh": 0,
    "charged_kwh": 0,
    "currents": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    }
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": -170,
    "L1": -70,
    "L2": -70,
    "L3": -30,
    "last_data_update": 1682108483,
    "charging_L1": true,
    "charging_L2": true,
    "charging_L3": true,
    "original_data": {
      "TOTAL": -170,
      "L1": -70,
      "L2": -70,
      "L3": -30
    }
  },
  "backlight": {
    "timer": 120,
    "status": "ON"
  }
}
```

as you can see contactor2 is set to always off. 

1. Why does the system indicate we're charging at 3x13A when contactor2 is interrupting 2 of the 3 phases? That is what always off means, right? 
2. The L1, L2, L3 all three have asterixes, which I assume mean "charging on this phase". 
3. charging_Lx fields in the raw data indicate charging on all 3 phases.

dingo35, 2023-04-22:
[master-83ecdeb3c9.zip](https://github.com/serkri/SmartEVSE-3/files/11301165/master-83ecdeb3c9.zip)
You are absolutely right, this is fixed in the upcoming 1.6.0 release; this is release candidate RC0 that we are currently testing.

Could you test this for me? 

NOTE: The phase detection has improved, and now also a sanity check is done so that if ALWAYS_OFF and nr_of_phases != 1 it is corrected to 1. BUT then it assumes that it is charging over the L1 phase. So you make sure your wiring is corresponding to this!!

Ivesvdf, 2023-04-22:
Sure, my wiring is as described. 

Note that I currently have neither a real EV attached nor a mainsmeter, so I'm testing by manually sending currents and an EV simulator box. 

### Test Report

**Note** It's now impossible to change from normal to smart/solar mode from the HW buttons when mainsmeter is not set, but mainsmeter option is now visible even from normal mode -> OK. 

So I understand based on the behavior that there's some kind of slow PID (or similar) loop inside the device which will try to continually adjust the current given to the vehicle based on the mainsmeter data. This makes sense and is the correct approach. 

When continually sending 2A, 2A, 2A as current data the charge current keeps rising until max -> OK. Then sending 0A 0A 0A it still seems to rise -> not sure this is exactly correct but it is still defensible. 

However then I send positive currents then the contactors open within 1 or 2 seconds, even if the EV charge current up until that time is 15A. Should the current not go down in the same manner it went up, until it reaches MIN, and then open only after some amount of time? I assume this is the STOP time. I'm not completely sure but I think with a real system the first time the clouds come out and the PV output current drops a bit contactors will open. 
If you're charging at 15A and consuming 1A from the grid while solar charging the current should be reduced to 14A and charging continued. 

My stop time is set to 10 (minutes according to the LCD).

dingo35, 2023-04-22:
I do not understand your testcase; I suppose you are in Solar mode. If you want to understand the exact behaviour of SmartEVSE I suggest you read the source code, it is too complex to explain here in a few sentences.

Better get yourself an EV and a MainsMeter, thats much more fun!

Ivesvdf, 2023-04-22:
The EV's on order (but will take some months, delivery date close to the end of the year), mains meter is being installed in 3 weeks though that's more because Fluvius, the grid company wants it rather than because I want it .

Yes, The previous test report was in solar mode. I figured out what the problem was by the way, when changing the values that were coming from the MainsMeter there was temporarily no data and this lack of data is why contactors opened. The documentation should probably state somewhere what the timing requirements for the API mainsmeter are, it seems they're fairly short. I'd also include if you're expecting the mainsmeter API readings at all times, or if it's okay to only send them when a vehicle is connected or when in SOLAR/SMART mode. 

This new version behaved as I'd expect:

[image]

So that's good. 

One small thing which I think might be a bad idea is that the charge current displayed on the website/API is sticky, even when no longer charging. So even when the EVSE is not commanding the EV it can charge at x amps, we still display the last charge current. Take this screenshot for example

[image]

This says charge 1x6A while actually, we're not charging at all because we're waiting for solar. I think it would be better to display 0A, as that is the max allowed current the vehicle is currently allowed to draw from the EVSE. The "1x " also has very little meaning, as the C2 setting could have been changed or user could have been using AUTO mode on solar.

closed 2023-04-23 by dingo35