serkri/SmartEVSE-3 #130: In Smart mode, it will not charge on max capacity - Serkri-1.5.5

Xnygra, 2023-04-22:
Hi,

Not sure if this is a bug or wrong configuration from my side, 
Mains of the house: 3x20A
Charger fuse: 3x16A

Charging never goes above 13A, even though there is plenty of capacity, solar panels generating ~3x9A at time if of the snippet below. Any idea why? 

OT: I rarely use SOLAR mode as its more price efficient for me to charge during night and sell excess power from solar during daytime.

{
    "version": "SERKRI-1.5.5",
    "mode": "SMART",
    "mode_id": 3,
    "car_connected": true,
    "wifi": {
        "status": "WL_CONNECTED",
        "ssid": "Useme",
        "rssi": -67,
        "bssid": "E4:5E:1B:FB:86:5A",
        "auto_connect": false,
        "auto_reconnect": true
    },
    "evse": {
        "temp": 31,
        "temp_max": 65,
        "connected": true,
        "access": true,
        "mode": 1,
        "solar_stop_timer": 0,
        "state": "Charging",
        "state_id": 2,
        "error": "None",
        "error_id": 0,
        "rfid": "Not Installed"
    },
    "settings": {
        "charge_current": 130,
        "override_current": 0,
        "current_min": 6,
        "current_max": 17,
        "current_main": 22,
        "solar_max_import": 0,
        "solar_start_current": 5,
        "solar_stop_time": 10,
        "enable_C2": "Not present",
        "mains_meter": "Sensorbox"
    },
    "home_battery": {
        "current": 0,
        "last_update": 0
    },
    "ev_meter": {
        "description": "Disabled",
        "address": 10,
        "import_active_energy": 0,
        "total_kwh": 0,
        "charged_kwh": 0,
        "currents": {
            "TOTAL": 0,
            "L1": 0,
            "L2": 0,
            "L3": 0
        }
    },
    "mains_meter": {
        "import_active_energy": 0,
        "export_active_energy": 0
    },
    "phase_currents": {
        "TOTAL": 48,
        "L1": 46,
        "L2": 0,
        "L3": 2,
        "last_data_update": 1682158080,
        "charging_L1": false,
        "charging_L2": true,
        "charging_L3": true,
        "original_data": {
            "TOTAL": 48,
            "L1": 46,
            "L2": 0,
            "L3": 2
        }
    },
    "backlight": {
        "timer": 120,
        "status": "ON"
    }
}

dingo35, 2023-04-22:
It looks like a problem with the phase detection, it only detects 2 phases charging instead of three; although this would still not explain why it limits itself to 13A. // I see you edited your log, this log has 3 phases detected ok.
[master-83ecdeb3c9.zip](https://github.com/serkri/SmartEVSE-3/files/11301180/master-83ecdeb3c9.zip)

Also, unrelated, you might want to set MaxCurrent to 16A instead of 17A ?

Could you test this version, it is the oncoming 1.6.0 RC0; to see whats going on, after flashing both firmware.bin and spiffs.bin, telnet to your smartevse, press v and enter so you have verbose logging, connect your EV so it starts charging and upload your log from the beginning of the charging on...

dingo35, 2023-04-22:
OT:
"OT: I rarely use SOLAR mode as its more price efficient for me to charge during nighttime and sell excess power from solar during daytime."

So you will probably like the newest branch where we introduce delayed charging (so you no longer are dependent on Home Assistant or other domotica servers/tools).
How do you sell your excess solar power?

Xnygra, 2023-04-23:
> It looks like a problem with the phase detection, it only detects 2 phases charging instead of three; although this would still not explain why it limits itself to 13A. // I see you edited your log, [...]

Yes, It detected the three phases after a reboot, as it probably is not related to this issue, i edit it away for not mixing several fault under one ticket. 

Yes, you are right, I should use 16A instead of 17A, but the fuses are really slow and can cope with some over current for some time, so I use 17A, against all recommendations, to be able to charge the vehicles as fast as possible.

I'll get the logs as soon as I have some EV at home to charge,

Xnygra, 2023-04-23:
> OT: "OT: I rarely use SOLAR mode as its more price efficient for me to charge during nighttime and sell excess power from solar during daytime."
>
> So you will probably like the newest branch where we [...]

In Scandinavia, the grid companies offer a plan that allows you to buy and sell electricity hourly at the price set by the European electricity market,  https://www.nordpoolgroup.com/en/Market-data1/Dayahead/Area-Prices/nl/hourly/?view=table. Electricity is often significantly cheaper at night, sometimes by a factor of ten, so it is advantageous to utilize this time by charging electric vehicles, running dishwashers, heat pumps, etc. During the day, you can sell excess energy from solar panels. By maximizing the use of this plan, you can greatly reduce electricity costs or even earn money when combined with solar power.

Many Scandinavians who use this plan and charge their electric vehicles employ various methods to find the cheapest hours, such as using home assistant and integrations like smart-ev-charging, https://github.com/jonasbkarlsson/ev_smart_charging. However, a delayed charging function would be useful for those who do not use home automation or similar services.

dingo35, 2023-04-23:
Very interesting!!

dingo35, 2023-04-23:
@Xnygra When you are going to produce the log, could you please test on the latest release 1.6.0? This has extended logging on the calculation of the charge current...

closed 2023-05-12 by dingo35