serkri/SmartEVSE-3 #131: Configure Mains, Min & Circuit via API/webserver

mr-leo, 2023-04-24:
I retrofitted a dump/non-smart ev-charger with a SmartEVSE running Serki and i really love it! The only downside is that my SmartEVSE is not easy accessible from outside and i have to unscrew the housing when i want to change certain settings. It would be FANTASTIC when settings like Mains, Min and Circuit could be changed via the webserver or API. Would that be possible?

dingo35, 2023-04-24:
The number of parameters that can be configured through the API is growing fast... but the ones you name I wont make writable through the interfaces, because they are supposed to be fixed; they are determined by your hardware (thickness of wiring, max current of wires and fuses), so IMHO they should be set at installation and never be touched after that.

If you want to influence the charging current in Normal or Smart mode you can use OverrideCurrent.

mr-leo, 2023-04-25:
I get your point! And charge/solar minimum? This is different per car..

dingo35, 2023-04-25:
Why dont you install HomeAssistant and see whats in the smartevse integration? Its all there my friend.

mr-leo, 2023-05-11:
Thank you! I am using the latest version of the SmartEVSE intergration in HA now and can do almost everything i want. Except changing the "SmartEVSE Current Min" value. My EV has different minimum charging currents for 1 and 3 phase charging. I charge on 1 phase in Solar mode (C2 relay) and 3 fase in Normal (and Smart) mode. So at the moment i have to choose between Solar or Charge mode and have to manually change the minimum current on the SmartEVSE when i want to switch. It would be fantastic if the "SmartEVSE Current Min" setting can be changed by the API (in HA) so that i can automate it. Thx!

dingo35, 2023-05-11:
That is very strange, since the minimum current is standardized at 6A, at each and any phase. What car do you have?

mr-leo, 2023-05-11:
Renault ZOE, which is very very picky about charging...

dingo35, 2023-05-11:
A the French with their well known attitude to ignore standards.... 
What are the minimum currents your car needs at 1phase and at 3phase charging?

dingo35, 2023-05-11:
Not sure how this issue got closed, github is having problems....

mr-leo, 2023-05-11:
I don't know the exact numbers, i have the minimum om 12A now but 10A on 1 phase should work. It fails at 6A. The problem is that i have to go outside with a screwdriver and open up the charger to change the minimum current value. (witch can be different per car). That is why asked for the option to change them via the API. I understand why you don't want to change the Mains and Max currents as they are hardware related.

dingo35, 2023-05-13:
[current_min.zip](https://github.com/serkri/SmartEVSE-3/files/11469075/current_min.zip)
With this version you can manipulate MinCurrent between 6A and 16A (per phase), by sending POST commands like this:

 curl -X POST http://10.0.0.76/settings?current_min=6

Please test.

mr-leo, 2023-05-13:
It seems that is does not work, this is the output after changing the min to 6:

{"version":"08:54:33 @May 13 2023","mode":"OFF","mode_id":0,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"Punter 1","rssi":-64,"bssid":"F0:B0:14:57:3E:FF","auto_connect":false,"auto_reconnect":true},"evse":{"temp":27,"temp_max":65,"connected":false,"access":false,"mode":0,"pwm":1024,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":12,"current_max":16,"current_main":25,"solar_max_import":2,"solar_start_current":10,"solar_stop_time":5,"enable_C2":"Always On","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":-150,"L1":-50,"L2":10,"L3":-110,"last_data_update":1683970165,"charging_L1":true,"charging_L2":true,"charging_L3":true,"original_data":{"TOTAL":-150}}}

mr-leo, 2023-05-13:
http://192.168.178.229/settings?current_min=6

dingo35, 2023-05-13:
You have to do a POST, and with your browser you are doing a GET, so that doesnt work.

mr-leo, 2023-05-13:
Thx! That explains a lot! It works perfect using POST.

dingo35, 2023-05-13:
It'll be in the next release!

closed 2023-05-13 by dingo35