serkri/SmartEVSE-3 #132: Feature request, read all modbus meters and also log voltage, cos phi etc

fluppie, 2023-04-30:
Currently I have a ESP32 with modbus shield from [enri.nl ](https://enri.nl/pcb/?page=modbus) to read all parameters / all modbusmeters. This gives 2 modbus masters which is far from ideal. I don't know if it's a lot of work to read out more registers than current and power as well as kWh import / export.

Screenshot of the ESP32 with ESPEasy software:
[image]

The Nan values are because of modbus collisions. I think the SmartEVSE is requesting every 2 seconds data on the modbus. My mainsmeter is at 15s intervals and the PV and EV meter are at 60s intervals. To limit the number of collisions.

List with all available registers:
https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.h

[image]

Just a question, if not I keep using it like this. On the other hand I can save a ESP32, a 5V power supply and modbus collisions :).

Ferforfaen, 2023-04-30:
I would love that! 

I have tried to do readouts with two masters, but failed.

dingo35, 2023-04-30:
I am thinking about a sort of configurable gateway from modbus to REST API....

fluppie, 2023-04-30:
Or an implementation of Modbus TCP, so all Modbus RTU devices can also be read/controlled over Modbus TCP. But I think it would be enough to go full read-out with SmartEVSE v3 and then using the REST API to send Modbus meter data to Home Assistant or other HA systems.
I do like the fact that inter SmartEVSE communication is RTU RS485, so Sensorbox can be far away of the SmartEVSE or for example one electrical cabinet with 4 SmartEVSE's talking to another board with 4 EVSE's 20-30m away. Same story for the sensorbox. In large industrial sites the main board where power from the netdistributor or own local transformator can be quite far away from the electrical cabinets with SmartEVSE's.

dingo35, 2023-05-01:
[modbus-tcp-bridge.zip](https://github.com/serkri/SmartEVSE-3/files/11364568/modbus-tcp-bridge.zip)

In this version your MainsMeter, EVMeter and PVMeter present themselves at port 502 of your SmartEVSE, with the modbus-tcp protocol; so this makes it possible for you to use whatever modbus function you want on whatever modbus register you want.
E.g. my MainsMeter is at slave address 0x0a, so this command reads register 70decimal and following:
mbpoll -a10 -t 3:hex -r 70 -c 10 10.0.0.76 

Please test, have fun!

fluppie, 2023-05-01:
Looks like it's working can it interfere with the SmartEVSE RTU modbus polling? I mean can the TCP requests from HA happen on the same time as the SmartEVSE is polling this data from the RTU modbus?
[image]

Because sometimes I see this:
[image]

Home Assistant yaml config
```
modbus: 
  - type: tcp
    host: 192.168.1.119 # ip van smartevse v3
    port: 502
    name: "smartevsev3modbustcp"
    close_comm_on_error: true
    delay: 3
    timeout: 5
    sensors:
      - name: sdm230_phase_1_line_to_neutral_volts
        slave: 101
        address: 0
        input_type: input
        count: 2
        precision: 2
        data_type: float32
        unit_of_measurement: V
        device_class: voltage
        scan_interval: 10
      - name: sdm230_phase_1_current
        slave: 101
        address: 6
        input_type: input
        count: 2
        precision: 2
        data_type: float32
        unit_of_measurement: A
        device_class: current
        scan_interval: 10
      - name: sdm230_phase_1_power
        slave: 101
        address: 12
        input_type: input
        count: 2
        precision: 3
        data_type: float32
        unit_of_measurement: W
        device_class: power
      - name: sdm230_phase_1_va
        slave: 101
        address: 18
        input_type: input
        count: 2
        precision: 2
        data_type: float32
        unit_of_measurement: VA
        device_class: power
```

dingo35, 2023-05-01:
Thats why I ask you to test it thoroughly: it uses the same client (=modbus server) as the rest of the firmware, so I would expect no collisions. But I enabled modbus lowlevel error routine, so if you have telnet log running you should see no timeout or other modbus errors...

fluppie, 2023-05-01:
Without disconnecting the second ESP32 module with ESPEasy firmware, CRC errors 99% sure coming from the second modbusmaster trying to fetch data:

```
(MBhandleError)(C0) Error response: E2 - CRC check error
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E4 - Server ID mismatch
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E4 - Server ID mismatch
(MBhandleError)(C0) Error response: E4 - Server ID mismatch
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: -9.6 A L2: 0.0 A L3: 0.0 A Isum: -9.6 A
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E2 - CRC check error
(MBhandleError)(C0) Error response: E2 - CRC check error
(MBhandleError)(C0) Error response: E2 - CRC check error
(MBhandleError)(C0) Error response: E2 - CRC check error
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E2 - CRC check error
(MBhandleError)(C0) Error response: E4 - Server ID mismatch
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E4 - Server ID mismatch
(MBhandleError)(C0) Error response: E0 - Timeout
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: -9.6 A L2: 0.0 A L3: 0.0 A Isum: -9.6 A
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E2 - CRC check error
(MBhandleError)(C0) Error response: E4 - Server ID mismatch
(MBhandleError)(C0) Error response: E2 - CRC check error
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E4 - Server ID mismatch
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E4 - Server ID mismatch
(MBhandleError)(C0) Error response: E0 - Timeout
(loop)(C1) 1 clients running.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: -9.6 A L2: 0.0 A L3: 0.0 A Isum: -9.6 A
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E2 - CRC check error
(MBhandleError)(C0) Error response: E4 - Server ID mismatch
(MBhandleError)(C0) Error response: E2 - CRC check error
(MBhandleError)(C0) Error response: E4 - Server ID mismatch
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E4 - Server ID mismatch
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E4 - Server ID mismatch
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: -9.6 A L2: 0.0 A L3: 0.0 A Isum: -9.6 A
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
```

After disconnecting the second ESP32 modbus device:
```
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: -6.5 A L2: 0.0 A L3: 0.0 A Isum: -6.5 A
(MBhandleError)(C0) Error response: E0 - Timeout
(loop)(C1) 0 clients running.
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
[... 21 lines cut ...]
(MBhandleError)(C0) Error response: E0 - Timeout
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: -6.5 A L2: 0.0 A L3: 0.0 A Isum: -6.5 A
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: -6.6 A L2: 0.0 A L3: 0.0 A Isum: -6.6 A
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: -6.5 A L2: 0.0 A L3: 0.0 A Isum: -6.5 A
(MBhandleError)(C0) Error response: E0 - Timeout
```

dingo35, 2023-05-01:
Ok going to try second setup... stay tuned!

dingo35, 2023-05-01:
[modbus-tcp-bridge2.zip](https://github.com/serkri/SmartEVSE-3/files/11365579/modbus-tcp-bridge2.zip)
This any better?

fluppie, 2023-05-01:
No, E0 Time out remains.
I also have the impression I see more "niet beschikbaar" when monitoring the sensors. What parameter(s) did you change? 
[image]

dingo35, 2023-05-01:
I did add another, separate client for the TCP bridge.

Now are you polling the extra registers you requested or are you also polling the registers that are already available through the REST API? Because that would definitely mess up traffic on the bus....

dingo35, 2023-05-02:
When I try to copy your setup, I get this error in HomeAssistant:
`WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.sdm230_phase_1_va (<class 'homeassistant.components.modbus.sensor.ModbusRegisterSensor'>) is using native unit of measurement 'VA' which is not a valid unit for the device class ('power') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+modbus%22`

And also, every 30 seconds I get this:
`May  2 07:28:24 backupserver hass[30207]: 2023-05-02 07:28:24.716 ERROR (SyncWorker_3) [homeassistant.components.modbus.modbus] Pymodbus: smartevsev3modbustcp: Exception Response(132, 4, None)`

When I comment out the device_class: power of this sensor, both messages are gone. I still get an occasional timeout in telnet but not as many as you do .... would this change anything in your setup?

dingo35, 2023-05-02:
I cannot get the timeouts reproduced in my setup; I get an occasional timeout, but this is not often enough to do some decent testing.

I can see the following test scenarios:
1) I can compile a 1.6.1 version (without the modbus TCP bridge) that has the Modbus error messages enabled (for some reason they were disabled until now), to make sure you didn't have timeout messages all the time (without the bridge component)
2) You can add some statements to your configuration.yaml to see if this has any influence:
`  - type: tcp
    host: 10.0.0.76 # ip van smartevse v3
    port: 502
    name: "smartevsev3modbustcp"
    close_comm_on_error: true
    delay: 3
    timeout: 5
    #close_comm_on_error: false
    #retry_on_empty: true
    #retries: 10
    #delay: 10
    #message_wait_milliseconds: 1000
    #timeout: 10
` 
This represents your setup, if you use the commented parameters this might improve the behaviour of HomeAssistant.

Let me know which way you want to go....

dingo35, 2023-05-02:
For test scenario 1 you can use 
https://github.com/serkri/SmartEVSE-3/releases/tag/v1.6.2

since it has no modbus TCP bridge, but it does have the extended log messages for modbus errors.

fluppie, 2023-05-02:
OK great, will look into this. Possible that the current 1.6.2 release has Telnet disabled?

dingo35, 2023-05-03:
You've got to rename firmware.debug.bin to firmware.bin

dingo35, 2023-05-04:
It would not solve the problem that your MainsMeter is fixed as slave at address 1, and SmartEVSE is fixed as master at address 1.

fluppie, 2023-05-04:
Finally had some time to flash the debug version (from the now pulled 1.6.2) , stupid me, not checking the whole folder contents...
```
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarSto                                      pTimer: 0 NoCurrent: 0 Imeasured: 3.4 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.4 A L2: 0.0 A L3: 0.0 A Isum: 3.4 A
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarSto                                      pTimer: 0 NoCurrent: 0 Imeasured: 3.4 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.4 A L2: 0.0 A L3: 0.0 A Isum: 3.4 A
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout
(MBhandleError)(C0) Error response: E0 - Timeout

```

Second ESP32 modbus module is disconnected. So don't really get why I get the timeouts. My meters have 101, 102, 103 and 104 as modbus address.

fluppie, 2023-05-04:
Got it, it's because Load balancing is set to master and no slaves are connected. In the past we had to this to get the Circuit menu option. In newer releases this is not needed anymore, disabling this fixed the timeouts :).

dingo35, 2023-05-05:
Ok thx for discovering this.  So does this also solve your timeout problems for modbus-tcp-bridge.zip?

fluppie, 2023-05-05:
Yes exactly. I used your first tcp build.These are just 7 messages to send to all 7 slaves. And since there are none, these don’t get received/acked.With verbose on my “euro” fell…This weekend I’ll add more meters and see what happens. Good stuff!

dingo35, 2023-05-05:
I have a nice weekend present for you, this version has the problems in 1.6.2 fixed, and also has added detailed logging to the timeout/other modbus error messages; thanks to your trick to put LoadBl = 1 I could test this now!

In normal logging the timeout-messages to the slaves are suppressed, in verbose logging they are still shown... Also a broadcast message that times out will show up if you have loadbalancing enabled without extra EVSE's connected, this one is not suppressed since it is not that annoying...

Let me know what happens if you add more meters!
[modbus-tcp-bridge3.zip](https://github.com/serkri/SmartEVSE-3/files/11405778/modbus-tcp-bridge3.zip)

fluppie, 2023-05-05:
[image]
This is fun, so address 101 and 102 are known by the SmartEVSE. Modbus address 103 (solar kWh meter) and 104 (EV2 kWh meter) are not used/known by SmartEVSE. Is it a pure coincidence that these addresses are not read? Or is the SmartEVSE actively blocking all modbus traffic except messages coming from known addresses? Load balancing is disabled.

dingo35, 2023-05-05:
Currently only traffic for the MainsMeter address, the EVmeter address and the PVmeter address is forwarded.

The forwarding has to be configured per address at the RTU modbus, so if you want more addresses to be read a config routine has to be built...

fluppie, 2023-05-05:
Yes, would be nice if we could add addresses to the list. I also think PV_Meter / PVmeter is not used anymore. I'm not able to find the menu option for it, I can remember with a v2 or v3 with original firmware this menu item was somewhere. Only MainsM and EVM show up.

dingo35, 2023-05-05:
I think it depends on the setting MainsMeterMeasure...

fluppie, 2023-05-05:
Indeed, you can enable it by changing in the menu what the mains meter measures.
[image]

fluppie, 2023-05-06:
Looks like I can't use the mainsmeter setting with PV :). Purple is sensorbox data to my second SmartEVSE (for now a separate modbus network) and blue the data from the first SmartEVSE with several SDM230's.
[image]

Yesterday was with the old working settings, then the data is very similar.

dingo35, 2023-05-06:
I pushed the code to master, Im thinking of detecting TCP requests that are not yet forwarded and opening a forward for them dynamically; not sure if it can be done simply in the emodbus library...

fluppie, 2023-05-06:
Ok cool. So maybe something like could be enough to get it working?
```

MBbridge.attachServer(Extra1MeterAddress, Extra1MeterAddress, ANY_FUNCTION_CODE, &MBclient);
MBbridge.attachServer(Extra2MeterAddress, Extra2MeterAddress, ANY_FUNCTION_CODE, &MBclient);
```

dingo35, 2023-05-06:
Yes that would to it, but I dont want to clutter the user interface with this stuff. Any normal user would not understand, it would generate a lot of unnecessary questions...

fluppie, 2023-05-06:
Yes that works indeed, changed the mainsmeter back to "all" so PV meter doesn't work anymore.

```
uint8_t Extra1MeterAddress = 103;
uint8_t Extra2MeterAddress = 104;
MBbridge.attachServer(Extra1MeterAddress, Extra1MeterAddress, ANY_FUNCTION_CODE, &MBclient);
MBbridge.attachServer(Extra2MeterAddress, Extra2MeterAddress, ANY_FUNCTION_CODE, &MBclient)
```
[image]

Any way we can make this configurable via HTTP commands? Some advanced functionality for users who want more :-D.

dingo35, 2023-05-06:
That might be one way to go. Cannot test myself right now, but if you put logging at verbose, and you request a non-existing slave address through the TCP interface, does that generate an error message on the telnet interface on that address?

dingo35, 2023-05-06:
Or perhaps make an auto-discover routine that discovers all the devices on the modbus...

fluppie, 2023-05-06:
How to build the firmware.debug.bin version? I just did, project tasks, release, build.

fluppie, 2023-05-06:
Got it, DBG 1 in evse.h

I commented one Extrameter1 67/103 out but keep requesting via HomeAssistant. Extrameter2 = 68/104
Looks like the working Extrameter2 is not showing up in the logs. Address 65 = 101 (Mains Meter) and 66 = 102 (EV Meter)
Same for the not working Extrameter1, no signs of this in the logs.

```
(D) (Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(D) (ModbusSend8)(C0) Sent packet(ModbusSend8)(C0) address: 65, function: 04, reg: 0006, data: 000c.
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 65 04 18 40 3f 58 a1 00 00 00 00 00 00 00 00 c3 fd e9 02 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 65 Function 04
(V) (ModbusDecode)(C0)  Register 0006(ModbusDecode)(C0)  Response
(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(D) (ModbusSend8)(C0) Sent packet(ModbusSend8)(C0) address: 66, function: 04, reg: 0048, data: 0002.
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 66 04 04 45 98 0d 3f
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04
(V) (ModbusDecode)(C0)  Register 0048(ModbusDecode)(C0)  Response
(D) (ModbusSend8)(C0) Sent packet(ModbusSend8)(C0) address: 66, function: 04, reg: 000c, data: 0002.
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 66 04 04 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04
(V) (ModbusDecode)(C0)  Register 000c(ModbusDecode)(C0)  Response
(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement
(D) (ModbusSend8)(C0) Sent packet(ModbusSend8)(C0) address: 66, function: 04, reg: 0006, data: 000c.
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 66 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04
(V) (ModbusDecode)(C0)  Register 0006(ModbusDecode)(C0)  Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=41.2 A Imeasured=0.0 A MaxCircuit=33 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=41.2 A, Idifference=33.0, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=41.2 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=41.2 A.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: -2.9 A L2: 0.0 A L3: 0.0 A Isum: -2.9 A
(D) (Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(D) (ModbusSend8)(C0) Sent packet(ModbusSend8)(C0) address: 65, function: 04, reg: 0006, data: 000c.
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 65 04 18 40 3f ef c7 00 00 00 00 00 00 00 00 c3 ff 2a 62 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 65 Function 04
(V) (ModbusDecode)(C0)  Register 0006(ModbusDecode)(C0)  Response
(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(D) (ModbusSend8)(C0) Sent packet(ModbusSend8)(C0) address: 66, function: 04, reg: 0048, data: 0002.
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 66 04 04 45 98 0d 3f
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04
(V) (ModbusDecode)(C0)  Register 0048(ModbusDecode)(C0)  Response
(D) (ModbusSend8)(C0) Sent packet(ModbusSend8)(C0) address: 66, function: 04, reg: 000c, data: 0002.
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (7 bytes) 66 04 04 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04
(V) (ModbusDecode)(C0)  Register 000c(ModbusDecode)(C0)  Response
(D) (Timer100ms)(C0) ModbusRequest 21: Request EVMeter Current Measurement
(D) (ModbusSend8)(C0) Sent packet(ModbusSend8)(C0) address: 66, function: 04, reg: 0006, data: 000c.
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 66 04 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 66 Function 04
(V) (ModbusDecode)(C0)  Register 0006(ModbusDecode)(C0)  Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=41.2 A Imeasured=0.0 A MaxCircuit=33 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=41.2 A, Idifference=33.0, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=41.2 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=41.2 A.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: -2.9 A L2: 0.0 A L3: 0.0 A Isum: -2.9 A
(D) (Timer100ms)(C0) ModbusRequest 3: Request MainsMeter Measurement
(D) (ModbusSend8)(C0) Sent packet(ModbusSend8)(C0) address: 65, function: 04, reg: 0006, data: 000c.
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (27 bytes) 65 04 18 40 48 a8 e0 00 00 00 00 00 00 00 00 c4 0a ab 34 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 65 Function 04
(V) (ModbusDecode)(C0)  Register 0006(ModbusDecode)(C0)  Response
(D) (Timer100ms)(C0) ModbusRequest 5: Request Energy Node 0
(D) (ModbusSend8)(C0) Sent packet(ModbusSend8)(C0) address: 66, function: 04, reg: 0048, data: 0002.
(D) (ModbusSend8)(C0) Sent packet(ModbusSend8)(C0) address: 66, function: 04, reg: 0048, data: 0002.
[... cut ...]
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=41.2 A Imeasured=0.0 A MaxCircuit=33 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=41.2 A, Idifference=33.0, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=41.2 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=41.2 A.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: -3.1 A L2: 0.0 A L3: 0.0 A Isum: -3.1 A
```

dingo35, 2023-05-06:
Ok thx! So that way won't work, will look at the auto discover option....

fluppie, 2023-05-06:
Quite some data in a "stupid" SDM230
[image]

fluppie, 2023-05-10:
Could these weird charging currents (underlined blue section) be related to the scrambled data?
[image]
Was more than enough capacity on the 40A mono with 30

[image]
Towards the end is normal, that's the battery getting to 100%

dingo35, 2023-05-10:
It looks like your EVmeter data is disturbed; why dont you put logging on verbose then you can see exactly whats going on on the modbus...

fluppie, 2023-05-10:
Addr 65 = MainsMeter

Addr 66 = EVMeter
reg 0006 = current
reg 0C = power
reg 48 = import active kwh

So it happens the most with EV meter, as you saw from the display. Disconnecting my second ESP32 with ESPEasy and modbus shield doesn't make any difference in the logs.

```
(loop)(C1) 1 clients running.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(MBhandleError)(C0) Error response: E0 - Timeout, address: 65, function: 04, reg: 0006.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 66, function: 04, reg: 0048.
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 66, function: 04, reg: 000c.
(MBhandleError)(C0) Error response: E2 - CRC check error, address: 66, function: 04, reg: 0006.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.0 A L2: 0.0 A L3: 0.0 A Isum: 3.0 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.0 A L2: 0.0 A L3: 0.0 A Isum: 3.0 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.1 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.1 A L2: 0.0 A L3: 0.0 A Isum: 3.1 A
(loop)(C1) 1 clients running.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.1 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.1 A L2: 0.0 A L3: 0.0 A Isum: 3.1 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.1 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.1 A L2: 0.0 A L3: 0.0 A Isum: 3.1 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.1 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.1 A L2: 0.0 A L3: 0.0 A Isum: 3.1 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(loop)(C1) 1 clients running.
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 66, function: 04, reg: 0048.
(MBhandleError)(C0) Error response: E2 - CRC check error, address: 66, function: 04, reg: 000c.
(MBhandleError)(C0) Error response: E2 - CRC check error, address: 66, function: 04, reg: 0006.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(MBhandleError)(C0) Error response: E2 - CRC check error, address: 66, function: 04, reg: 000c.
(MBhandleError)(C0) Error response: E2 - CRC check error, address: 66, function: 04, reg: 0006.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 66, function: 04, reg: 0048.
(MBhandleError)(C0) Error response: E2 - CRC check error, address: 66, function: 04, reg: 000c.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.1 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.1 A L2: 0.0 A L3: 0.0 A Isum: 3.1 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.1 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.1 A L2: 0.0 A L3: 0.0 A Isum: 3.1 A

```

fluppie, 2023-05-10:
Commenting out MainsM register 0006 and EVM registers 0006, 000C and 0048 heavily reduces the errors.
So looks like it's double fetching it when Home Assistant requests through Modbus TCP?
Also now it are timeouts E0 and with HA requesting we get E2's CRC and E4's Server ID mismatch
```
(loop)(C1) 1 clients running.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.8 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.8 A L2: 0.0 A L3: 0.0 A Isum: 2.8 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(loop)(C1) 1 clients running.
(MBhandleError)(C0) Error response: E0 - Timeout, address: 65, function: 04, reg: 0006.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.8 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.8 A L2: 0.0 A L3: 0.0 A Isum: 2.8 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.8 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.8 A L2: 0.0 A L3: 0.0 A Isum: 2.8 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.1 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.1 A L2: 0.0 A L3: 0.0 A Isum: 3.1 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.2 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.2 A L2: 0.0 A L3: 0.0 A Isum: 3.2 A
(loop)(C1) 1 clients running.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.0 A L2: 0.0 A L3: 0.0 A Isum: 3.0 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.0 A L2: 0.0 A L3: 0.0 A Isum: 3.0 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.0 A L2: 0.0 A L3: 0.0 A Isum: 3.0 A
(MBhandleError)(C0) Error response: E0 - Timeout, address: 66, function: 04, reg: 0006.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.2 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.2 A L2: 0.0 A L3: 0.0 A Isum: 3.2 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.3 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.3 A L2: 0.0 A L3: 0.0 A Isum: 3.3 A
(loop)(C1) 1 clients running.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.2 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.2 A L2: 0.0 A L3: 0.0 A Isum: 3.2 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.1 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.1 A L2: 0.0 A L3: 0.0 A Isum: 3.1 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.1 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.1 A L2: 0.0 A L3: 0.0 A Isum: 3.1 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.0 A L2: 0.0 A L3: 0.0 A Isum: 3.0 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.0 A L2: 0.0 A L3: 0.0 A Isum: 3.0 A
(loop)(C1) 1 clients running.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 3.0 A L2: 0.0 A L3: 0.0 A Isum: 3.0 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.9 A IsetBalanced: 41.2 A
(I) (printStatus)(C0) L1: 2.9 A L2: 0.0 A L3: 0.0 A Isum: 2.9 A
(loop)(C1) 1 clients running.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 41.2 A
```

dingo35, 2023-05-11:
Ok are you running latest v1.6.4 with Loadbalancing off?
If so I can make another version that uses a second client datastruct, that might reduce the collisions....

fluppie, 2023-05-11:
Yes, it's a self compiled version based on the current repo status, only difference are the extra meters. With DBG 1.
[image]

dingo35, 2023-05-11:
[fluppie.zip](https://github.com/serkri/SmartEVSE-3/files/11452757/fluppie.zip)
Try applying this patch, adapter your Extra?MeterAddress - Attachservers too, and see if there are more or less timeouts/crc errors....

fluppie, 2023-05-11:
For some reason it didn't want to:

```
C:\SmartEVSE3serkri\SmartEVSE-3>git apply fluppie.diff
error: patch failed: SmartEVSE-3/src/evse.cpp:2670
error: SmartEVSE-3/src/evse.cpp: patch does not apply
```

Applies it manually but introduce more errors E0 E2 E4 E5. Without the patch only timeouts E0.

```
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 40.0 A
(I) (printStatus)(C0) L1: 3.0 A L2: 0.0 A L3: 0.0 A Isum: 3.0 A
(loop)(C1) 1 clients running.
(MBhandleError)(C0) Error response: E5 - Packet length error, address: 66, function: 04, reg: 000c.
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 40.0 A
(I) (printStatus)(C0) L1: 3.0 A L2: 0.0 A L3: 0.0 A Isum: 3.0 A
(MBhandleError)(C0) Error response: E5 - Packet length error, address: 65, function: 04, reg: 0006.
(MBhandleError)(C0) Error response: E0 - Timeout, address: 66, function: 04, reg: 0048.
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 40.0 A
(I) (printStatus)(C0) L1: 3.0 A L2: 0.0 A L3: 0.0 A Isum: 3.0 A
(MBhandleError)(C0) Error response: E0 - Timeout, address: 66, function: 04, reg: 0006.
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 3.0 A IsetBalanced: 40.0 A
(I) (printStatus)(C0) L1: 3.0 A L2: 0.0 A L3: 0.0 A Isum: 3.0 A

(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 31.1 A IsetBalanced: 32.0 A
(I) (printStatus)(C0) L1: 31.1 A L2: 0.0 A L3: 0.0 A Isum: 31.1 A
(loop)(C1) 1 clients running.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 32.8 A IsetBalanced: 32.0 A
(I) (printStatus)(C0) L1: 32.8 A L2: 0.0 A L3: 0.0 A Isum: 32.8 A
(MBhandleError)(C0) Error response: E0 - Timeout, address: 65, function: 04, reg: 0006.
(MBhandleError)(C0) Error response: E5 - Packet length error, address: 66, function: 04, reg: 0048.
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 66, function: 04, reg: 000c.
(MBhandleError)(C0) Error response: E2 - CRC check error, address: 66, function: 04, reg: 0006.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 32.8 A IsetBalanced: 32.0 A
(I) (printStatus)(C0) L1: 32.8 A L2: 0.0 A L3: 0.0 A Isum: 32.8 A

(I) (printStatus)(C0) L1: 32.7 A L2: 0.0 A L3: 0.0 A Isum: 32.7 A
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 32.8 A IsetBalanced: 32.0 A
(I) (printStatus)(C0) L1: 32.8 A L2: 0.0 A L3: 0.0 A Isum: 32.8 A
(loop)(C1) 1 clients running.
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 65, function: 04, reg: 0006.
(MBhandleError)(C0) Error response: E0 - Timeout, address: 66, function: 04, reg: 000c.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 32.8 A IsetBalanced: 32.0 A
(I) (printStatus)(C0) L1: 32.8 A L2: 0.0 A L3: 0.0 A Isum: 32.8 A

I) (printStatus)(C0) L1: 32.7 A L2: 0.0 A L3: 0.0 A Isum: 32.7 A
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 32.7 A IsetBalanced: 32.0 A
(I) (printStatus)(C0) L1: 32.7 A L2: 0.0 A L3: 0.0 A Isum: 32.7 A
(loop)(C1) 1 clients running.
(MBhandleError)(C0) Error response: E0 - Timeout, address: 65, function: 04, reg: 0006.
(MBhandleError)(C0) Error response: E2 - CRC check error, address: 66, function: 04, reg: 000c.
(MBhandleError)(C0) Error response: E2 - CRC check error, address: 66, function: 04, reg: 0006.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 32.7 A IsetBalanced: 32.0 A
```

fluppie, 2023-05-11:
In the current history I can see that power data is sometimes received as current data :). You can hardly see the regular currents 6 to 32A.
[image]

dingo35, 2023-05-13:
Ok the library should handle all those request async, since I cannot reproduce your timeouts I'm kind of stuck here; not sure if you are overloading the modbus or you hit a bug in the library.

Perhaps you can investigate more by the internal logging of the library:
https://emodbus.github.io/logging

and/or asking around on the library discussion forum:
https://github.com/eModbus/eModbus/discussions

dingo35, 2023-05-20:
@fluppie, perhaps you can test this version, it has one useless task removed and the modbus-handling task moved from low priority to high priority. It might make some difference, it might not....
[master-c232e73d9d2.zip](https://github.com/serkri/SmartEVSE-3/files/11521620/master-c232e73d9d2.zip)

fluppie, 2023-05-20:
Hi @dingo35 I did some test with my second ESP32 module running ESP Easy, without this module and with HA live and shutdown as well as booting HA while logging.

[putty_2ndModbusMaster.log](https://github.com/serkri/SmartEVSE-3/files/11523286/putty_2ndModbusMaster.log) 39 errors E0's, E2's, E4's
[putty_2ndModbusMaster_HAshutdown.log](https://github.com/serkri/SmartEVSE-3/files/11523287/putty_2ndModbusMaster_HAshutdown.log) 37 errors E0's, E2's, E4's
[putty_no2ndModbusMaster.log](https://github.com/serkri/SmartEVSE-3/files/11523288/putty_no2ndModbusMaster.log) 13 E0's
[putty_no2ndModbusMaster_HAshutdown.log](https://github.com/serkri/SmartEVSE-3/files/11523289/putty_no2ndModbusMaster_HAshutdown.log) only one E0 time-out
[putty_no2ndModbusMaster_HAshutdown_HAboot_runningLine4178.log](https://github.com/serkri/SmartEVSE-3/files/11523290/putty_no2ndModbusMaster_HAshutdown_HAboot_runningLine4178.log) Only one E2 before HA live, after HA booting (live from line 4178), 4 E0's

For the moment I only poll modbus registers that are not being polled by the SmartEVSE firmware.
[image]
[image]

Will do some testing with double polling those registers

dingo35, 2023-05-21:
Ok, basically the question is, without the 2nd busmaster, is this version an improvement to v1.6.5 as to modbus errors.

From what I read not much difference?

hp197, 2023-06-12:
👍 

I also have a multi meter setup:

```
[mains] -> [Eastron SDM630] -> [Distribution] -> [Eastron SDM630] -> [SmartEVSE]
                                                                ^
                                                 [Eastron SDM630] <- [solar]
```

The problem I have is that I have to set SmartEVSE mains meter in all mode, to have the correct readings.
But when I do, remote readout of the solar meter stops (after a restart of SmartEVSE).
This is due to the MBbridge thread configured here: https://github.com/serkri/SmartEVSE-3/blob/master/SmartEVSE-3/src/evse.cpp#L2815

My current solution is to build my own patched version, with the 3 if's commented out. So that the MBbridge will always be created for all 3 meters.

* Note for the developer, I believe that if you switch from mains meter mode from mains+evse to all, the MBbridge thread for the solar meter isn't torn down. If you want to do something like auto discovery you'll also need to keep track 'open' bridges.

dingo35, 2023-06-13:
[tcpbridge.zip](https://github.com/serkri/SmartEVSE-3/files/11734091/tcpbridge.zip)
That would be one way to go, I went another way (since with your solution you can just wait until someone has a fixed modbus address meter etc.). 

        //to bridge meter address (decimal)
        //activate tcp Modbus bridge, example:
        //curl -X POST http://smartevse-xxxxx.lan/modbus?bridge_meter_address=101
        //max 5 meters allowed, of which mainsmeter, evmeter and pvmeter might already be taken
        //TODO not sure what happens if you exceed this maximum, since attachServer always returns true....
        //has to be activated after every reboot

Please test thoroughly, especially what happens if you attach the 6th meter: does it simply not work, does it bump out one of the other meters?

dingo35, 2023-06-20:
@fluppie @hp197 Any progress on testing this?

fluppie, 2023-06-21:
I think beginning of July I have some more time to test this version. At the moment I run the latest build with the Modem features.

dingo35, 2023-06-21:
[tcpbridge_with_modem.zip](https://github.com/serkri/SmartEVSE-3/files/11815539/tcpbridge_with_modem.zip)
@fluppie , if it helps you, this is current master (including Arend Jan's modem code) + the patch to test, so you can keep on using the modem stuff during testing....

hp197, 2023-06-21:
Hi sorry, I haven't got time yet to test it.
That's mainly because I need to rework my data collector to send a http request to open up the bridge.

arpiecodes, 2023-07-06:
I also recently 'migrated' my EV meter to modbus because of the addition of the TCP Modbus bridge (before it was pushing in the relevant EV meter details through MQTT on SmartEVSE from HA, which worked very well). Now I see a lot of bogus values appearing. E.g. big jumps in amperage values for the EV meter, big jumps in 'ChargePower' aka the power in watts being transferred into the EV, while nothing is flowing through.. It seems like some sort of corruption of the values or the checksum not being checked properly. Is this a known issue?

dingo35, 2023-07-06:
Thanks for testing, this is new info since the others in this thread didnt have time to test it.

If I understand correctly :
Your sensorbox on modbus 10 gives good values in SmartEVSE and over the tcp bridge;
Your EVmeter on modbus 11 gives strange values in SmartEVSE but correct values over the tcp bridge.
Correct?

arpiecodes, 2023-07-06:
I must not forget to mention that this is with the Modbus Bridge logic incorporated in the current master branch, not the custom build versions that were shared in this pull request thread.

> Your sensorbox on modbus 10 gives good values in SmartEVSE and over the tcp bridge;
> Your EVmeter on modbus 11 gives strange values in SmartEVSE but correct values over the tcp bridge.

The Sensorbox is not read out through the modbus bridge (only by SmartEVSE) but directly connected to HA through ESPHome.

Since the whole reason I moved the EV meter from HA to SmartEVSE was the TCP Bridge functionality, being able to read all values in HA as well, I am actually not sure whether this issue also occurs without using the Modbus bridge functionality.. I may have to do some testing soon to confirm that. I do see some Modbus communication CRC/timeout errors appearing from time to time, in the debug output on telnet. I do not use load balancing btw.

[22 more replies]