serkri/SmartEVSE-3 #133: Feature request: Post EV Meter through API

tomcoonen, 2023-05-02:
Would it be an option to also have the 'API' option also for the EV Meter?
I'm posting Mains meter values to the 'currents' endpoint now, maybe include the EV Meter values there to limit the amount of calls to be handled?

I'm happy to draft a PR if you are open to the idea?

dingo35, 2023-05-02:
The API for the MainsMeter makes sense to me, because a lot of power panels already contain  mainsmeters that have incompatible interfaces (P1 or otherwise), and cheap solutions are for sale to interface that to EM_API.

But if you already bought an extra EVmeter, why would you not feed it through modbus? I cannot imagine a use case , so help me?

dingo35, 2023-05-02:
Wow, for half the money you would have had a modbus-MID meter of the same brand. The EV meter is usually placed next to or inside the SmartEVSE box, so wireless does not add much. Considering the reliability of modbus vs wifi I would always recommend the wired version....

Im sorry but I think your use case is too specific to add to the mainstream firmware...

tomcoonen, 2023-05-02:
I can agree with your view on this, I would also like to get WiFi out of the picture for this.
I'll get me a modbus gateway and get that running. (If you have any recommendations, they are welcome. Probably will get me this one: https://www.amazon.nl/RS485-Ethernet-Converter-Industrial-auto-negotiation/dp/B0BGHVRMPJ/).

Not really half the money ;) https://www.amazon.nl/Elektrische-Energiemeter-achtergrondverlichting-MID-gecertificeerd-gekalibreerd/dp/B08X911LWF

dingo35, 2023-05-02:
Instead of adding another component in the communication chain, I would go thisxway:

https://www.amazon.nl/Draaistroommeter-SDM72-digitale-stroommeter-SDM72DM-V2/dp/B09QK3VBN3/ref=mp_s_a_1_1?crid=2Z5WFIDO60XPW&keywords=sdm72dm-v2&qid=1683015070&sprefix=sdm72%2Caps%2C119&sr=8-1

The HA-smartevse integration will read the values from the SmartEVSE through the REST API.

Then you can use your wifi meter for other purposes or sell it :-)

tomcoonen, 2023-05-02:
Thanks, let's try that :)

Enough options to put an extra meter somewhere 😄

dingo35, 2023-05-02:
I agree, monitoring energy usage will only become more important..
Make sure you have the v2 version, the v1 versions are known not to work over modbus!

dingo35, 2023-05-08:
@tomcoonen Ok Tom, don't know if you noticed but a lot of other people had the same request as you, so we are currently working on finishing off this request. Hope you didn't order your new EV meter already ....

tomcoonen, 2023-05-08:
I ordered it anyway, the 'old' meter will get a better purpose for it's functionality 😃.
Nice to see it will be an option though!

dingo35, 2023-05-08:
#135 committed to 7d9faa7.

closed 2023-05-08 by dingo35