serkri/SmartEVSE-3 #137: Cannot set mode to OFF

daninden, 2023-05-07:
If I set the mode to OFF (using the web ui, homeassistant or direct POST), the charging stops but the mode does not turn to OFF anymore, after a few seconds, charging starts again.

I did some debugging, write_settings() was added at some places, this calls validate_settings(), and in there is this check:
```
if (RFIDReader == 2) Access_bit = 0;
    // Enable access if no access switch used
    else if (Switch != 1 && Switch != 2) Access_bit = 1;
```

the else if flips the Access_bit to 1 again on my setup. I am not entirely sure what needs to be done to fix it in the current flow.

dingo35, 2023-05-08:
Thanks for noticing Dan, I have no RFID reader attached so I didn't notice this behaviour; it forced me to release v1.6.4 early, that has a bugfix in it for this problem. Please confirm if it tests ok @ your setup?

daninden, 2023-05-08:
i don't have an RFID reader attached either (it actually hits it if RFIDReader != 2).. it does this if Switch !=1 or 2.. which both are activation switche types (i have my switch configured as smart/solar switcher).

dingo35, 2023-05-08:
Aha ok it should still be fixed since those lines will no longer be hit when changing modes....

closed 2023-05-12 by dingo35