serkri/SmartEVSE-3 #144: slave EVSE mode selection when master EVSE is on off

djoenez, 2023-05-14:
this issue is similar to the issues that was already solved. I noticed this:

EVSE1 (master, gets data via API) is off
EVSE2 set to solar

EVSE2 then starts charging at max current (while solar was at ~10A), restart EVSE2 and still at max current.
Only way to get EVSE2 on solar is to quickly switch EVSE1 in Solar and then off, After that EVSE2 is responding to my API input on EVSE1.

Another one: 
EVSE1 is on Solar
EVSE2 i want to start charging anyway at maximum, so set EVSE2 at Smart
EVSE2 will not start charging, again have to switch EVSE1 to Smart and then to Solar, and EVSE2 will charge at max input then.

dingo35, 2023-05-16:
Sorry cannot test or reproduce this since my setup has only one SmartEVSE.

But you can do what I would do: first flash latest version v1.6.5, test it, and downgrade to v1.5.0, test it, and so on, until you found the release/commit that causes the problem....

djoenez, 2023-05-22:
Hello, i think this is not something that came with a certain update (atleast i don't know if it was working before, i am only experimenting with two EVSEs since ~2 months. I'll update to the latest and see what happens.

Just a question, is for the slave EVSE the master evse mode leading? (so if i set the master EVSE to a certain mode, does the slave follow?)

dingo35, 2023-05-22:
I have no idea; the communication between the Master and the nodes looks fairly complex, so I keep my hands off of it. Perhaps load the original software and try, or ask the seller?

jaroslawp, 2023-05-27:
@djoenez :  As I can see in my setup slave follows mode set on master.  I have tried to run master in different mode than slave but this usually resulted in one or both reporting errors requiring power-cycle to recover - while changing modes via API or using a hardware button  (last year I have asked upstream maintainer and at the time he confirmed that mode changes are to be done on master only: https://github.com/SmartEVSE/SmartEVSE-3/issues/20#issuecomment-1148774070 .. it seems this is still the case with current firmware versions)

dingo35, 2023-06-07:
This is a duplicate of #152

closed 2023-06-07 by dingo35