serkri/SmartEVSE-3 #145: Feature request: Provide kWh consumed through API

snoekieboe, 2023-05-21:
It might exceed the processing power of the ESP32, but would it be possible to provide a (calculated) value for "EV charged kWh" and "EV total kWh"?

In my setup (without a seperate kWh meter with modbus) these values now stay empty.

dingo35, 2023-05-21:
If you dont have a kWh meter that measures the power, how would SmartEVSE know how much power is charged? 
The charging current that is shown is the MAXimum current the EV us allowed to draw, the EV can easily slow charging down or even stop it without letting the EVSE know.

That kind of info should be given by the EV, not the EVSE.

snoekieboe, 2023-05-21:
I think you might have a point there ;-)  I did not think about the fact the EVSE does not know the power actually used to charge. It is just aware of the status of MAINS instead which is the sum of all things connected and not just the EV.

Guess you can close this one :-)

closed 2023-05-21 by dingo35