serkri/SmartEVSE-3 #147: Delayed charging for solar mode

hmmbob, 2023-05-22:
The docs read the following:

> the SmartEVSE will stay on indefinitely. Delayed Charging in Solar mode is not supported, since in Solar mode the sun will decide when charging starts...

I would still like to use delayed start for solar mode 😄  The reason is financial: I have plenty of PV panels and am on a dynamic contract (hourly prices). Because of the dynamic prices, it is sometimes more favorable for me to wait to use my own energy until the prices are extremely low, because then returning to grid is completely not interesting. Like yesterday, "use yourself" is very interesting from 12h on, whilst returning to grid was more beneficial in the hours before that:

[image]

I would have loved to delay solar charging until after 12:00. Is there a possibility to also add "delayed start" in solar mode?

dingo35, 2023-05-22:
Have some patience my friend!

mr-leo, 2023-05-22:
When you have a dynamic contract and the price is low, it makes more sense to charge as fast as you can (normal/smart mode) instead of solar mode and also take energy from the grid, doesn't it? So that you can make more use of the low prices.

hmmbob, 2023-05-22:
"Low price" vs "paying nothing at all because I am using my own solar production"

option 2 is better 😉

mr-leo, 2023-05-22:
No because you get the exact same amount as you pay with dynamic prices. So you pay 6ct of get 6ct..

hmmbob, 2023-05-22:
I know how dynamic pricing works - I don't see why this discussion is relevant for the request?

dingo35, 2023-05-22:
IMHO any discussion on the use case is relevant, to prevent function creep.....

dingo35, 2023-05-22:
But how does it look if you do Smart charging with delay until 12:00 ?

hmmbob, 2023-05-22:
> But how does it look if you do Smart charging with delay until 12:00 ?

Gimme a sec.

dingo35, 2023-05-22:
And so we all learn a bit; thanks for your work! 
I still have enabling delayed solar charging on my list, because I believe in maximum user freedom as long as it doesn't ask for to much code. In fact it might even reduce some code :-) . But I always like to understand my users cases..... and you are right, if dynamic pricing stops paying the same amount as charging, it will be a different ball game.

mr-leo, 2023-05-22:
If you already use HA, you can set the mode from OFF to SOLAR at the desired time, and don't need the timer on the SmartEVSE at all. You can even automate it at the cheapest moment.

deqw, 2023-05-22:
> If you already use HA, you can set the mode from OFF to SOLAR at the desired time, and don't need the timer on the SmartEVSE at all. You can even automate it at the cheapest moment.

I did so and used Node Red to make some charge policies. I’m not sure but when I use Solar mode it looks to me that the SmartEVSE tries to prevent using power from the net, to a certain level. 
[image]

hmmbob, 2023-05-22:
> I did so and used Node Red to make some charge policies. I’m not sure but when I use Solar mode it looks to me that the SmartEVSE tries to prevent using power from the net, to a certain level. [image]

Nice! Got your code avail somewhere?

deqw, 2023-05-22:
Not right now, but no problem to share. I will document it and share it here if that is possible.

deqw, 2023-05-25:
@hmmbob That involves a lot of screenshots and additional work for you. Just install Node-Red on HA and import the json and you will be on your way in a few minutes. Besides that, I'm using Safari on MacOS which has a bug with forms resulting in not showing all the forms properties, so a lot of screenshots will have incomplete information (empty properties).

Let me know what you are going to do?

Anyway, this is the complete flow:

[image]

closed 2023-06-07 by dingo35