serkri/SmartEVSE-3 #151: Charging L2 L3 reporting false - 1.6.5

deqw, 2023-05-28:
While 3 phase charging (1.6.5)  the sensors Charging L2 and Charging L2 are reporting false and Charge in the web GUI reports 1 x 13.0 A. Attached some screenshots of the situation. Is this correct?

 
[image]
[image]
[image]

dingo35, 2023-05-28:
Is your car a Tesla by coincidence?

deqw, 2023-05-28:
No it's a BMW iX3.

dingo35, 2023-05-28:
Perhaps you can post a verbose telnet log? 
It could be a bug, but similar behaviour happened with a Tesla that started drawing current on a single phase and only after 11-25 seconds it started charging over the other phases...

deqw, 2023-05-28:
But at the moment of the screenshots it was already charging 3 phase for 21 minutes and drawing current on L1 L2 L3 from the start/1st second.

deqw, 2023-05-29:
I noticed this morning that without charging and with or without the cable connected to the car:

[image]

Then I did a reboot via the web interface (no cable connected) and it changed to:

[image]

[image]

Is this the correct reporting for L1, L2 and L3 in this situation?

dingo35, 2023-05-29:
Those booleans only have meaning during a charging session.

dingo35, 2023-06-30:
Committed in 14abe7cb0bf4

closed 2023-06-30 by dingo35