serkri/SmartEVSE-3 #154: Current limit with i4

deekoowee, 2023-06-03:
Smart EVSE current limit setting do not work with i4, car is overriding that.
The owner's handbook says, that car will set the charging current as charger has informed.
Seems, that car don't 'underestand' the charging current limitation data from Smarts EVSE.
Any ideas ?

dingo35, 2023-06-03:
Sounds like operator error, please be more clear on which settings give what results, preferably with telnet log.

deekoowee, 2023-06-04:
Here's short log
[evse.log](https://github.com/serkri/SmartEVSE-3/files/11645349/evse.log)
Lot of errors...

Changed the charging current at SmartEVSE webpage, doesn't change the charging real current at car.
First, I have 6A and set it to 10A, but car's dashboard still says 6A.

Found, that SmartEVSE do not actually change the charging current before I set it off, set new current and set it back on.
So, changing of the charge current is not possible 'on fly'

Another finding;
Car have set 'unlock charge port automatically, when charging has stopped'
However, that do not work. Seems, that car do not get unlocking info from SmartEVSE, when the charging has stopped.
I have to unlock the car's charging port manually.

dingo35, 2023-06-04:
You dont understand how charging works: the EVSE gives the Maximum current that the EV can draw, but the EV decides how much it is going to take.

It is very well possible that your car decides on another current when you restart the session.

Please read up on EV electric charging principles before posting bug reports.

closed 2023-06-04 by dingo35

deekoowee, 2023-06-04:
So, the errors in log are caused because I don't understand

dingo35, 2023-06-04:
No problem, we all have to learn. But if you see strange stuff, read up on it, dont immediately start posting. This is not facebook.... :-)

arpiecodes, 2023-07-28:
> So, the errors in log are caused because I don't understand

There are no errors, its displaying `Error: 0` as part of the debug output meaning there's in fact no error currently.