serkri/SmartEVSE-3 #165: Car connected wrong sensor value

negronoir, 2023-07-06:
Hello,

In v1.6.7 the 'sensor.smartevse_car_connected' doesn't show if car is connected. Always take the opposite value of 'sensor.smartevse_access' despite if car is connected or not.

If I set to true the access, the car connected sensor shows false.

If I set to false the access, the car connected sensor shows true.

In former versions including v1.6.6, this works fine. 

Thanks

dingo35, 2023-07-08:
Thanks for your bug report; this version should solve the problem, could you please test?
[nomodem.zip](https://github.com/serkri/SmartEVSE-3/files/11990383/nomodem.zip)

dingo35, 2023-07-09:
Fixed in v1.7.0.

closed 2023-07-09 by dingo35

negronoir, 2023-07-17:
Hi!

Sorry for the delay in the answer. I was on holidays out of home.

Checked and fixed.

Thanks!!