serkri/SmartEVSE-3 #166: MQTT setting fields are empty after page reload

hmmbob, 2023-07-12:
I've succesfully connected the SmartEVSE to my mqtt server (great!), but when I now reopen the mqtt settings tab all fields are empty:

[image]

I expected to see the current config here (maybe without the password or so).

hmmbob, 2023-07-12:
Is there any way to connect to the SmartEVSE and edit the index.html directly, or is there always a new build needed?

I have no experience with the latter (I always update with your provided binaries), so I haven't tested this locally. However, given that the key names were just not matching, I'm 99% sure this is fixed with the PR submitted.

dingo35, 2023-07-12:
You can test your local index.html easy by editing the line that has 10.0.0.36 in it; comment the line above it, change it to your own smartevse's ip address, and if you approach file:///path to your local index.html you can see immmediate results

hmmbob, 2023-07-12:
The "local override" (see edit above) worked - it's confirmed tested

dingo35, 2023-07-12:
Yup we were typing in parallel; hope to test this afternoon.

dingo35, 2023-07-12:
@hmmbob Thanks for your work!

closed 2023-07-12 by dingo35

deekoowee, 2023-07-16:
@hmmbob , how did you reach the mqtt config page ?
[image]
My Smart EVSE main page looks like this;
[image]

hmmbob, 2023-07-16:
Did you flash spiffs too? Or only firmware.bin?

deekoowee, 2023-07-16:
I have flashed only firmware.bin
Now I flashed spiffs too and config functionality appears on main screen.
Thanks for helping on this.

PR #167 (2023-07-12): Fix empty MQTT fields
This should fix #166, as the keys the JS was looking for are called differently in the settings JSON.

~~Haven't tested it locally though.~~