serkri/SmartEVSE-3 #168: Documentation for MQTT topics?

pnuding, 2023-07-13:
Excited about the new MQTT client in 1.7.0 i was trying today to move my communication of meter readings from http calls to mqtt. For battery current it worked instantly. However I haven't been able to get smartevse to accept my mains meter readings. From looking at the code I figured I'd need to send a JSON like {"L1":15,"L2":0,"L3":0} to the SmartEVSE-xxxxx/Set/MainsMeter topic
But it doesn't seem to accept it. 
Is there any documentation on this? 
Similarly i'm puzzled by the new "PVCurrent" topics exposed, where does that come from and what's the actual meaning?

stevoh6, 2023-07-13:
Hi, the documentation is mifssing, but you need to send this to that topic "15:0:0", just raw string, values separated with ":". 

Also this is diferrent from API, since home battery is set separatelly via new topic SmartEVSE-xxxxx/Set/HomeBatteryCurrent.

pnuding, 2023-07-13:
Gotcha, a bit odd but i got it working now 👍
I guess in terms of how most devices work with MQTT it'd be clearer to either supply the values in a JSON or ideally even asychronously on separate topics - same way that they're being output. I realise that's clunkier in terms of detecting communication errors, though

arpiecodes, 2023-07-13:
Separated topics are a can of worms you probably don't want to open, as the EVSE needs a stable and atomic view of all current values to make decisions. Allowing them to be set separately could introduce variance that may interfere with the balancing logic.

The reason it's not JSON is basically since it's a microcontroller, and any JSON parsing/stringifying costs resources. ESP32 is a pretty beefy chip, but it has a lot going on in case of the SmartEVSE. Using raw strings with delimiters and sprintf seemed the more natural choice, as you'd need to manually format the message on NodeRED/HA side anyways.

dingo35, 2023-07-14:
@pnuding glad you worked it out, I personally don't feed the API with current info so I hadn't run into this one.

The PV current is for a kWh meter that monitors your solar panels; it is hardly used in combination with SmartEVSE, not much practical use there since normally you want to monitor the feed / usage of the grid through MainsMeter.

Would you be so kind to document your MQTT-feeding, either in a PR or a text document?

closed 2023-08-04 by dingo35